betaflight/betaflight #11329: BF 4.3.0 on RC2 and Nightly #2627 - random Failsafe on GEP-F722_AIO

RobertS90, 2022-01-22:
### Describe the bug

On the GEP-F722-35A AIO Board i get random Failsafes. In the Blackbox I see that the input of the throttle suddenly goes for example from 1300 to 1000. Sometimes it happens right after a few seconds and sometimes it takes a little longer. The drone just falls down. Sometimes the beeper turns on (Failsafe) and sometimes i can just take off after falling down.
With 4.2.x I have no Problems with this Board.

I´ll attach the Blackbox Screenshot, maybe this helps:
[image]
.

### To Reproduce

Just take off and fly for a few seconds.

### Expected behavior

No Failsafe during flight.

### Flight controller configuration

```
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.11 Nov  9 2021 / 20:29:32 (948ba6339) MSP API: 1.43
# config: manufacturer_id: GEPR, board_name: GEPRC_F722_AIO, version: d3484d96, date: 2021-04-28T07:05:45Z

# start the command batch
batch start

board_name GEPRC_F722_AIO
manufacturer_id GEPR

# name: CineLog35

# resources
resource MOTOR 1 C07
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C08
resource SERIAL_TX 3 NONE
resource SERIAL_RX 3 NONE

# feature
feature -LED_STRIP
feature -AIRMODE

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 4 64 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 1700 2100 0 0
aux 2 2 2 1325 1700 0 0
aux 3 13 1 1700 2100 0 0
aux 4 28 3 1700 2100 0 0
aux 5 35 1 1300 1700 0 0

# master
set gyro_lowpass2_hz = 275
set dyn_lpf_gyro_min_hz = 220
set dyn_lpf_gyro_max_hz = 550
set acc_calibration = -51,3,36,1
set mag_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_auto_smoothness = 30
set serialrx_provider = CRSF
set dshot_idle_value = 500
set motor_pwm_protocol = DSHOT300
set align_board_roll = 180
set align_board_yaw = 0
set vbat_max_cell_voltage = 435
set vbat_min_cell_voltage = 340
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 2
set osd_warn_rssi = ON
set osd_vbat_pos = 2338
set osd_rssi_pos = 2104
set osd_link_quality_pos = 2136
set osd_flymode_pos = 2146
set osd_current_pos = 2402
set osd_mah_drawn_pos = 2434
set osd_craft_name_pos = 2062
set osd_pidrate_profile_pos = 2178
set osd_warnings_pos = 14761
set osd_avg_cell_voltage_pos = 2370
set osd_battery_usage_pos = 457
set osd_disarmed_pos = 2242
set osd_flip_arrow_pos = 238
set debug_mode = GYRO_SCALED
set gyro_1_sensor_align = CW180
set gyro_1_align_yaw = 1800
set name = CineLog35

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 77
set dyn_lpf_dterm_max_hz = 187
set dterm_lowpass_hz = 165
set dterm_lowpass2_hz = 165

rateprofile 0

# rateprofile 0
set thr_expo = 50
set roll_rc_rate = 90
set pitch_rc_rate = 90
set yaw_rc_rate = 90
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set pitch_srate = 65
set yaw_srate = 65

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: CAMERA_CONTROL
A01: FREE
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: GYRO_EXTI
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: GYRO_CS 1
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: FREE
B09: FLASH_CS
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: LED 1
C05: FREE
C06: MOTOR 2
C07: MOTOR 1
C08: MOTOR 4
C09: MOTOR 3
C10: FREE
C11: FREE
C12: SERIAL_TX 5
C13: FREE
C14: FREE
C15: BEEPER
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1: CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: MOTOR 2
    CH2: MOTOR 1
    CH3: MOTOR 4
    CH4: MOTOR 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

GepRC GEP-F722-35A AIO, Banggood

### Other components

- DJI Caddx Vista, Polar Pro Cam
- StanFPV 2203 3390 KV Motor
- TBS Crossfire Nano R

### How are the different components wired up

- Caddx Vista is directly connected with 6-Pin connector. - UART1
- Motos are wired to the AIO.
- TBS Crossfire Nano RX - UART5

### Add any other context about the problem that you think might be relevant here

I fly other drones with 4.3 and I have no Problem at all. I love it and would like to use it with this board. I also tested different Firmware Version on my ESC (BLHeli-S and Bluejay). The drone always drops out of the sky...
I testet 4KHz PID Loop with DShot 300 and 8KHz PID Loop with DShot 600. with and without Bidirectional DShot.

Thank you for your work, I really enjoy flying my drones with Betaflight :)

granolaFPV, 2022-01-22:
FYI the "diff all" you included is on 4.2.11, you may want to update with the diff from the version that's giving you issues (4.3 RC2 and/or nightly 2627). Same goes for "resource show all".

RobertS90, 2022-01-22:
> FYI the "diff all" you included is on 4.2.11, you may want to update with the diff from the version that's giving you issues (4.3 RC2 and/or nightly 2627). Same goes for "resource show all". It might also be helpful to share a link to your blackbox log.

You are right. My bad. I'll update it soon and also share the Blackbox file. 

Thank you for your answer 😊👍

daleckystepan, 2022-02-07:
RC2 has some issues with SPI RXes. Can you test it with RC3 or the latest nightly? Should be fixed.

closed 2022-03-18 by github-actions[bot]