betaflight/betaflight #11332: jhemcu f405pro/version f405 firmware

AlbanyFpv, 2022-01-23:
### Describe the bug

once flashed with target firmware [4.3 RC1] gyro was not showing as working, double checked correct firmware was flashed but not sure why wasnt working. 
so i decided to drop back to 4.3 rc1 to see if that would get my fc operational again and no luck on that front, after trying to revert back too 3.2.11 the fc became non responsive and is now for a lack of a better word....Bricked.

### To Reproduce

flash any version of bf 4.3 , target : jhef405

### Expected behavior

i expect my jhemcu 45am f405 aio not to be bricked...

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```

### Flight controller

jhemcu 45amp f405 aio 

### Other components

hglrc zeus nano vtx.
tbs crossfire 
runcam nano 2
rgb leds x8

### How are the different components wired up

Cam to fc passthrough to vtx with addition of irtramp protocal
tbs crossfire to uart2 
led to led/buzz tab

tri4ng1e, 2022-03-02:
I had the same issue with JHEMCU F405 PRO AIO 25A. Used auto-detect feature and it selected "JHEF405". It applied successfully, but the gyro was not present. Then I noticed that there is "JHEF405PRO" target. Obviously, this one worked like a charm, gyro is present and all seems to be OK.

So sum up:
1. [@AlbanyFpv](https://github.com/betaflight/betaflight/issues/11332#issue-1111893598) please try to flash it with JHEF405PRO target, if the FC is reachable in any way. If you are using Windows, try to use [ImpulseRC Driver Fixer](https://impulserc.blob.core.windows.net/utilities/ImpulseRC_Driver_Fixer.exe), it helps to recover the FC sometimes.
2. Is it possible to separate F405 and F405PRO for auto-detect feature, or those FCs are using the same IDs / whatever?

haslinghuis, 2022-03-02:
> 2. Is it possible to separate F405 and F405PRO for auto-detect feature, or those FCs are using the same IDs / whatever?

@tri4ng1e auto-detect just looks at the target present already on the board.

See also: https://github.com/betaflight/unified-targets/pull/511

Duplicate of https://github.com/betaflight/betaflight/issues/10825

closed 2022-03-02 by haslinghuis

Lennypassarelli, 2022-08-11:
ok so i am having same issue ......i watched a video from cyclone fpv ......he said try flywoo411    but i am still having issues with the 405 non pro ......they come with different firmware .....you can also try betaflight F4 .......mine worked but now the esc's are not working ........i am still trying to figure it out