betaflight/betaflight #11334: 'Reboot to bootloader' boots to wrong bootloader.

hydra, 2022-01-24:
### Describe the bug
Betaflight supports bootloaders, bootloaders are used to do the flashing of firmware.  Two examples are a) the original OpenPilot bootloader, and b) the SPRacing EXST bootloader.

When the user clicks 'reboot to bootloader' they expect the firmware to reboot to the bootloader that allows for flashing new firmware via the tool they are currently using.  i.e.  use 'reboot to bootloader' in the configurator to flash via the configurator.  Users don't do anything else other than flash firmware or unplug the FC from the configurator when the FC is in it's bootloader mode.  Users in the SPRacing discord server will attest to this situation.

They do NOT expect 'reboot to bootloader' that takes them to a bootloader (the built-in ST ROM-based DFU bootloader) that can potentially require reflashing/recovery of the pre-installed bootloader.

The usual 'Flash Firmware' button reboots to the correct bootloader for flashing.

### To Reproduce
Click 'Reboot to bootloader', flash wrong firmware to FC that has a bootloader.

### Expected behavior
1) The firmware should reboot to the firmware used for flashing BY DEFAULT when asked to go to the 'bootloader' (both via CLI 'bl' command and MSP).
2) The configurator should reboot to the firmware used for flashing when 'reboot to bootloader' is pressed.
3) The firmware and configurator should NEVER boot to the ROM bootloader unless specifically requested, as in the case of an FC with a bootloader that is for developers and manufacturers only.  e.g.  they should use 'bl rom' in the CLI or jump the boot pads if they *really* need to, e.g. for connecting a debugger or flashing bootloader firmware, not BF firmware.

To summarize:
cli: `bl` should reboot to the 'default' bootloader.  i.e. flash-based bootloader, if one exists.
cli: `bl flash` should reboot to flash-based bootloader, if one exists.
cli: `bl rom` should reboot to ROM-based bootloader.
gui: 'reboot to bootloader' should reboot to the 'default' bootloader, i.e. flash-based bootloader, if one exists.
gui: 'flash firmware' should reboot to the 'default' bootloader, i.e. flash-based bootloader, if one exists.

gui: additionally, if there were 'reboot' and 'reboot to ROM bootloader' expert-user-only buttons in the GUI that would be helpful too.

### Setup
- Configurator version: nightly

```
N/A
```

### Additional information

Way back when the EXST bootloader existed the original behavior was to reboot to the DEFAULT bootloader, the target knows what it's default is.  In the case of normal targets it was ROM, in the case of EXST targets it was FLASH.

Then this PR was created 'Add booting into the flash boot loader as an option' which changed the default behavior to always be the ROM bootloader.  I disagreed strongly via slack at the time that this would cause support issues, additional documentation and a bad user experience.
https://github.com/betaflight/betaflight/pull/8429

Due to that change more code had to be added to the configurator to allow it to reboot to the correct bootloader when flashing - previously it 'just worked' by rebooting to the 'default' bootloader, i.e. the one that allows flashing.
https://github.com/betaflight/betaflight/pull/8680

hydra, 2022-01-26:
I don't think any PR's will revert cleanly now, especially because more code was added which is now used by the configurator and is now required to be kept until the configurator drops support for the oldest firmware that needed the MSP support.  

A new one should be created that fixes the issues mentioned, then later we could schedule the changes to MSP for deprecation and finally another PR to delete them if you want to reclaim the previous MSP behavior and reduce the code-size and complexity of the firmware.

ctzsnooze, 2022-04-15:
@hydra please confirm if this issue is still active in current master?  
If not, maybe make a PR to address it?

hydra, 2022-04-15:
@ctzsnooze confirmed it's still an issue in master.  Yes, I can make a PR to address it. I'm currently unwell and haven't been able to work on any of my PRs recently.

klutvott123, 2022-04-23:
If the options to reboot to bootloader in flash or rom had been added in addition to just regular reboot to default bootloader this wouldn't have been an issue. 
The simple solution to  would be to add the missing code in the configurator to make the button reboot to bootloader in flash if one exists, and change the default behaviour of `bl` in cli

klutvott123, 2022-04-23:
I made a PR for discussion https://github.com/betaflight/betaflight-configurator/pull/2904

klutvott123, 2022-04-23:
Logic in cli looks sound to me. It will use the bootloader in flash if available and no arguments are given https://github.com/betaflight/betaflight/blob/a0ed6b0f1d1c8eb022d4c789089e0fc5ff6c6a5f/src/main/cli/cli.c#L3568-L3592

closed 2022-04-24 by haslinghuis

hydra, 2022-05-02:
@klutvott123 confirmed that the firmware logic does appear to be correct.  I just tested `bl` with no arguments on the H7RF and it reboots to the SP Racing bootloader.  Not sure exactly when that changed.  I'll check out the configurator side of things and comment/update as appropriate.