betaflight/betaflight #11336: Increase number of VTX control lines from 10 to a more useable number

dabit20, 2022-01-24:
### Is your feature request related to a problem? Please describe

Basically the same 'issue' as described earlier here: https://github.com/betaflight/betaflight/issues/10572
The number of 'vtx' lines allowed is limited to 10. Now, in the quoted issue above there is a good explanation why the limit is 10, but at the same time that limit of 10 seems like an arbitrary number to me. 

In my case, 10 lines is too limited. And given the very few people using this feature it probably is for many other folks too. 
I have repurposed a trim switch on my TX16S radio as channel selector using GV adjustments and logical switches, and the 6-position switch as power/pitmode/off selector. Being able to increase power beyond 25mW when searching for a crashed quad or killing power when crashing close to the pilot area is valuable, as is quickly selecting another channel when people join or leave the event. Sure, it can be done from the BF LUA script or ExpressLRS LUA script also, but buttons on the radio are so much faster.

Also, I can change the VTX settings on the radio before powering up the quad, and when powering on the quad it is very quick to take the new setting, minimizing the chance of disturbing other feeds. So all in all, VTX control on radio switches is a very useful and IMHO underutilised feature.

### Describe the solution you'd like

I would like to have the number of lines allowed increased to a number that actually matches a broader range of use cases. I understand that it is impossible to cover all possible use cases, but a little more makes the feature far more useable. 

I think that an increase from 10 to 15 would work. The reasoning behind that is that not many pilots are using more than 8 channels and 2 bands (e.g. Raceband and IMD6 or Fatshark). VTXes with more than 4 power levels are quite rare and even when offered more it hardly makes sense to be able to select all of them. 
When adding it up, this yields 14 lines. Add a single 'to be defined' line, and we are at 15. 

### Describe alternatives you've considered

I currently run a self-compiled HEX file with MAX_CHANNEL_ACTIVATION_CONDITION_COUNT increased to 15 in vtx_control.h. I have done this for development BF4.3, 4.3RC1 and 4.3RC2. Increasing MAX_CHANNEL_ACTIVATION_CONDITION_COUNT to 15 fully solves my 'issue'. 

I have tested this on various quads with F722-based flight controllers and a single one with F405 based FC, no issues seen on any of them. 

Regarding alternative solutions: Using the OSD or HDZero VTX menu is not a substitution since you have to be transmitting video in order to use those.  I prefer to setup these things while not transmitting video as a courtesy to other pilots.

Using the BF LUA is a better solution, but the FC and RX needs power to be able to do so, which is not the case when pre-selecting the required VTX settings on the transmitter. 

### Other information

This feature request was initially posted here: https://github.com/betaflight/betaflight-configurator/issues/2738
I accidentally posted this in the wrong place, and I sincerely apologize for that.

haslinghuis, 2022-01-24:
Duplicate of #11034

dabit20, 2022-01-24:
Addition to the opening post: I was not aware of the stick commands to change band/channel; I learned that in #11034. However, in my humble opinion that method is even worse than using one of the various LUA scripts with VTX control functionality or using the OSD.

closed 2022-03-03 by github-actions[bot]

clement-nardi, 2022-05-21:
Hi @dabit20 ,

Thank you for your work on this issue, I just stumbled on this limitation to the number of vtx lines after spending hours setting up my transmitter to fully control bands, channels and power levels... quite frustrating.
Maybe you could open a pull-request yourself? Given the fact that you've already successfully tested the modified code.
I'd personally prefer to set the maximum to at least 17 (8 channels + 5 bands + 4 power levels), 20 seems like a reasonable value, I don't know if there can be negative consequences to increasing this value?

Anyway, thank you again!

Best regards,
Clement Nardi

crookz, 2026-09-22:
I second this for the same reason as listed above. Thanks

dabit20, 2026-09-23:
a PR probably won't get accepted for reasons mentioned in etracer's post, so I will not waste anyones time by sending one in: https://github.com/betaflight/betaflight/issues/11034#issuecomment-947840898 

However, it is a one-number change in one file. Go ahead and make the change.