betaflight/betaflight #11337: Wired lound sound and jolting

bakwc, 2022-01-24:
### Describe the bug

I build firmware from master branch and got wired loud motors sound and jolting. Also drone hardly response to any control. In betaflight 4.2 everything works fine.

### To Reproduce

1) Build firmware from **master** branch
2) Flash it, use default settings
3) Try to fly

### Expected behavior

Smooth flying

### Flight controller configuration

```

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F405 (S405) 4.3.0 Jan 24 2022 / 18:17:30 (f2559fbfd) MSP API: 1.44
# config: manufacturer_id: IFRC, board_name: IFLIGHT_SUCCEX_E_F4, version: 714fcb3f, date: 2021-12-07T06:25:44Z

# start the command batch
batch start

board_name IFLIGHT_SUCCEX_E_F4
manufacturer_id IFRC

# name: -3 0 39

# feature
feature GPS

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 3 2 115200 19200 0 115200
serial 5 64 115200 57600 0 115200

# aux
aux 0 0 0 900 1250 0 0
aux 1 1 2 1750 2100 0 0
aux 2 46 3 1700 2100 0 0
aux 3 13 1 1700 2100 0 0
aux 4 28 2 1300 1700 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 BAND_D   D CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set acc_calibration = -47,56,114,1
set mag_hardware = NONE
set serialrx_provider = CRSF
set failsafe_procedure = GPS-RESCUE
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_angle = 35
set gps_rescue_initial_alt = 100
set gps_rescue_descent_dist = 100
set gps_rescue_descend_rate = 500
set gps_rescue_min_sats = 6
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set osd_craft_name_pos = 14538
set osd_gps_sats_pos = 14471
set osd_altitude_pos = 14475
set gyro_1_align_yaw = 1800
set name = -3 0 39

profile 0

# profile 0
set throttle_boost = 0
set f_pitch = 0
set f_roll = 0
set f_yaw = 0

rateprofile 0

# rateprofile 0
set yaw_rc_rate = 200
set yaw_srate = 0

# end the command batch
batch end

```
```

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: BEEPER
B05: LED 1
B06: LED_STRIP
B07: FREE
B08: FREE
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: MOTOR 4
C09: MOTOR 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: INVERTER 2
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH3 : MOTOR 1
    CH4 : MOTOR 2
TIM4:
    CH1 : LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : MOTOR 4
    CH4 : MOTOR 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```

### Flight controller

IFLIGHT SUCCEX-E F4

### Add any other context about the problem that you think might be relevant here

Blackbox logs: https://drive.google.com/file/d/1oNi13EjJA0BuTxL4C2nd0T1xbXvAZ-Bk/view?usp=sharing
See the second part.
I armed, increased throttle, fly on a few centimeters and landed.
Sound started as I slightly increased throttle.

bakwc, 2022-01-24:
Possible duplicate of #11314

JoeSa3rd, 2022-01-24:
I'm just curious...
Any video of this issue?
Are using OpenTX or EdgeTX if so which FW version and do you have ADC filter disabled, (Recommended with BF v4.3)?
Are you also using TBS Tracer/Crossfire, or ExpressLRS? If so which FW version?
When setting up BF v4.3 did you happen to configure any RC_LINK presets?

bakwc, 2022-01-24:
OpenTX + Crossfire.
According to blackbox - there are some PID issues - the begin to oscillate more and more and don't stabilizing.

klutvott123, 2022-01-24:
Did you use default pids on 4.2? Default D gains are higher and the gyro lpf cutoffs are a little higher in 4.3.

bakwc, 2022-01-24:
I'll try to reduce everything and check if it helps.
On 4.2 default PIDs worked good on my quad.

bakwc, 2022-01-24:
Reduced to 0.4, it helped but quad control is awful. Tried to copy all the exact values from 4.2 (PIDs, filters, gyro cutoffs) - jittering still present.

klutvott123, 2022-01-24:
There's some noise in the 100-150Hz range present in both logs. Could be resonance at 120Hz Maybe try lowering the dyn notch min hz and see if it can deal with that? Maybe decrease Q a little too

bakwc, 2022-01-24:
It helped, thank you! Blackbox with fix: https://drive.google.com/file/d/1PcbQV2Yz9VrMfxUpHoydHh2DX5yfpKZA/view?usp=sharing - no more oscillations.
Still strange that in 4.2 was flying ok. Is there any changes related to filtering?

haslinghuis, 2022-01-24:
Yes new defaults:

```
            versionFilterDefaults.dyn_notch_q_rpm = 500;
            versionFilterDefaults.dyn_notch_q = 300;
            versionFilterDefaults.gyro_lowpass_hz = 250;
            versionFilterDefaults.gyro_lowpass_dyn_min_hz = 250;
            versionFilterDefaults.gyro_lowpass2_hz = 500;
            versionFilterDefaults.dterm_lowpass_hz = 75;
            versionFilterDefaults.dterm_lowpass_dyn_min_hz = 75;
            versionFilterDefaults.dterm_lowpass_dyn_max_hz = 150;
```

PID:
```
            versionPidDefaults = [
                45, 80, 40, 30, 120,
                47, 84, 46, 34, 125,
                45, 80,  0,  0, 120,
            ];
```

bakwc, 2022-01-24:
Seems like new defaults not always good for all setups (probably they don't have to). My issue is solved, if someone tweaks the defaults - would be glad to test new one. Currently I set Q factor to 80 and min frequency to 100.

closed 2022-01-24 by bakwc

klutvott123, 2022-01-24:
@bakwc Looks nice and clean now! :thumbsup: Defaults will never be perfect for all builds, but as you see it doesn't take much tuning to make it work 🙂

bakwc, 2022-01-26:
I thought a little bit on this issue. New defaults does not explain the situation.
We can see noise in 4.2 and 4.3 both. Filters are just workaround to reduce the noise.
In 4.2 we had a stable flight with noise.
In 4.3 with the same noise (and same PIDs) we have an issue.
Why? May be there were some other changes? Looks like we need some investigation here (may be binary search commits to see what was the change responsible for it).