betaflight/betaflight #11345: Strange throttle behaviour when upgrading from 4.2 to 4.3 RC2

xoga74, 2022-01-25:
### Describe the bug

Hello, 

when upgrading from 4.2.11 to 4.3 RC2 on OmnibusF4SD Board  motors arre not spinning when arming and throttle at idle postition.  

When i put all PID's at 0 to bypass the PID loops, motors only start to spin when throttle is at 50%.

In 4.2.11 motors are working fine.

Dumps of both 4.2.11 and 4.3 RC2 are attached
[BTFL_cli_X09A74_N06_20220125_210310.txt] (https://github.com/betaflight/betaflight/files/7937044/BTFL_cli_X09A74_N06_20220125_210310.txt)

### To Reproduce

Flash firmware 4.3 RC2 whith same connections and settings as 4.2.11.

Arming with throttle stick in idle position.  Motors are not spinning on set idle rate.

When reflashing 4.2.11, everything works fine again.  Motors spinning  in idle speed at throttle on idle position

### Expected behavior

When arming in 4.3 RC2, motors should spin at idle throttle stick (low throttle)

### Flight controller configuration

```

# version
# Betaflight / STM32F405 (S405) 4.3.0 Jan  9 2022 / 11:40:24 (ee8560af4) MSP API: 1.44
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4SD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id AIRB

# name: X09A74 N06

# feature
feature -RX_SERIAL
feature -AIRMODE
feature RX_PPM

# map
map RETA1234

# aux
aux 0 0 0 1300 1700 0 0

# master
set acc_calibration = -219,180,102,1
set mag_hardware = NONE
set min_throttle = 1020
set small_angle = 180
set pid_process_denom = 4
set gyro_1_sensor_align = CW90
set gyro_1_align_yaw = 900
set name = X09A74 N06

profile 0

rateprofile 0

# end the command batch
batch end
```
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: BARO_CS
B04: BEEPER
B05: LED 1
B06: FREE
B07: SDCARD_DETECT
B08: PPM
B09: FREE
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 4
    CH4 : MOTOR 3
TIM3:
    CH3 : MOTOR 1
    CH4 : MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10:
    CH1 : PPM
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: ADC 2
DMA2 Stream 4: FREE
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```

---
# version
# Betaflight / STM32F405 (S405) 4.2.11 Nov  9 2021 / 20:27:49 (948ba6339) MSP API: 1.43
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4SD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id AIRB

# name: X09A74 N06

# feature
feature -RX_SERIAL
feature -AIRMODE
feature RX_PPM

# map
map RETA1234

# aux
aux 0 0 0 1300 1700 0 0

# master
set acc_calibration = -217,194,109,1
set mag_hardware = NONE
set min_throttle = 1020
set small_angle = 180
set pid_process_denom = 4
set gyro_1_sensor_align = CW90
set gyro_1_align_yaw = 900
set name = X09A74 N06

profile 0

rateprofile 0

# end the command batch
batch end
----

----
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: BARO_CS
B04: BEEPER
B05: LED 1
B06: FREE
B07: SDCARD_DETECT
B08: PPM
B09: FREE
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 4
    CH4: MOTOR 3
TIM3:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10:
    CH1: PPM
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: ADC 2
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
---------

### Flight controller

OMNIBUS F4 SD

xoga74, 2022-01-25:
Tested also with RC1 and got same wrong behavour

daleckystepan, 2022-02-07:
Please retest with the latest nightly or RC3.

francoisvoyer, 2022-03-02:
I observe the same behaviour with an Openpilot revolution board in RC1/RC2/RC3, with PWM ESC.
When Throttle is low, it is like the command is under the minimum accepted by ESC, and ESC start to beep.
When Throttle is upper, the ESC stop to beep as command is in the expected range.
When Throttle is at mid stick, motor start to run.

klutvott123, 2022-03-02:
@francoisvoyer You are correct. It's fixed in https://github.com/betaflight/betaflight/pull/11429

francoisvoyer, 2022-03-02:
Thank you! Good news.

closed 2022-04-02 by haslinghuis