betaflight/betaflight #11347: OSD stats display shifted vertical moving the title off the top of the display

SteveCEvans, 2022-01-26:
### Describe the bug

When disarming the stats display title isn't always visible.

[image]

Discovered at https://github.com/betaflight/betaflight/pull/11330#issuecomment-1020692561 although not related to that PR is transpires, hence the creation of this issue.

### To Reproduce

Use the stats selection as per the below.

[image]

### Expected behavior

Stats display should have a title as per the example below.

[image]

### Flight controller configuration

```
# diff all

# version
# Betaflight / STM32H743 (SH74) 4.3.0 Jan 26 2022 / 22:17:30 (f2559fb) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKH743, version: e9841341, date: 2022-01-17T15:06:19Z
# board: manufacturer_id: MTKS, board_name: MATEKH743

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKH743
manufacturer_id MTKS
mcu_id 0033002a3130511333343837
signature 

# feature
feature -RX_SERIAL
feature RX_MSP
feature ESC_SENSOR

# serial
serial 7 1024 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0

# master
set acc_calibration = -97,-13,-3986,1
set motor_pwm_protocol = DSHOT600
set battery_meter = ESC
set osd_warn_rc_smoothing = OFF
set osd_logo_on_arming = ON
set osd_tim1 = 2592
set osd_vbat_pos = 2488
set osd_link_quality_pos = 2433
set osd_tim_1_pos = 6604
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2519
set osd_mah_drawn_pos = 2553
set osd_motor_diag_pos = 2465
set osd_display_name_pos = 2542
set osd_gps_lon_pos = 0
set osd_gps_lat_pos = 17
set osd_gps_sats_pos = 417
set osd_compass_bar_pos = 2090
set osd_debug_pos = 2402
set osd_power_pos = 2497
set osd_pidrate_profile_pos = 2054
set osd_avg_cell_voltage_pos = 2456
set osd_pit_ang_pos = 2199
set osd_rol_ang_pos = 2231
set osd_battery_usage_pos = 2528
set osd_nheading_pos = 2101
set osd_rtc_date_time_pos = 357
set osd_rate_profile_name_pos = 2064
set osd_pid_profile_name_pos = 2058
set osd_profile_name_pos = 2048
set osd_rcchannels_pos = 2112
set osd_efficiency_pos = 449
set osd_stat_rtc_date_time = ON
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_fft = ON
set osd_stat_total_flights = ON
set osd_stat_total_time = ON
set vcd_video_system = PAL
set displayport_max7456_wht = 1

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save 
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 3
A01: MOTOR 4
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: FREE
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: BEEPER
B00: MOTOR 1
B01: MOTOR 2
B02: GYRO_EXTI
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: SDIO_D0
C09: SDIO_D1
C10: SDIO_D2
C11: SDIO_D3
C12: SDIO_CK
C13: FREE
C14: FREE
C15: GYRO_CS 1
D00: FREE
D01: FREE
D02: SDIO_CMD
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: SPI_MOSI 1
D08: FREE
D09: FREE
D10: PINIO 1
D11: PINIO 2
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: SERIAL_RX 8
E01: FREE
E02: FREE
E03: LED 1
E04: LED 2
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: GYRO_CS 2
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1 : BEEPER
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
    CH2 : DSHOT_BITBANG 1
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: DSHOT_BITBANG 2
DMA1 Stream 1: DSHOT_BITBANG 1
DMA1 Stream 2: SPI_MOSI 1
DMA1 Stream 3: SPI_MISO 1
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MISO 2
DMA1 Stream 6: SPI_MOSI 4
DMA1 Stream 7: SPI_MISO 4
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

MATEKH743

### Other components

None

### How are the different components wired up

Video output connected to a USB 2861 PAL capture device.

ryanconj, 2022-02-01:
The OSD will intentionally drop the "--- STATS ---" if it's not going to fit on display. 
 
https://github.com/betaflight/betaflight/blob/18ee33e7a80904a8ddd11b8e13609068d8643d08/src/main/osd/osd.c#L864L892

I wasn't able to figure out what was going on. 
You have 14 options enabled, and your display only shows 13 without the Stats line. As you have the display set to PAL I think it should be able to output 16.  (VIDEO_LINES_PAL in max7456.h)

hydra, 2022-02-07:
Yeah, it looks like the calculation for display lines is not correct somewhere.

@SteveCEvans are you able to position OSD elements on lines 14-16?

SteveCEvans, 2022-03-10:
Yes

[image]

SteveCEvans, 2023-03-22:
Tested and this is now working. Stats display including heading is centred vertically and will display as many of the selected stats as will fit.

closed 2023-03-22 by SteveCEvans