betaflight/betaflight #11348: Betaflight v4.3 D-Term Filter settings not showing correctly on OSD

Zuldan, 2022-01-27:
### Describe the bug

In the Filter PP menu, the settings for D Term LPF1 "Dynamic" mode are not showing which make it difficult to tune filter settings on the field. Since D Term LPF1 "Dynamic" is the Betaflight 4.3 default, it should be at least be configurable via OSD.

### To Reproduce

N/A

### Expected behavior

D Term LPF1 "Dynamic" settings should be displayed in the OSD when active and D Term LPF1 Static settings hidden. Visa versa when Static is used.

### Flight controller configuration

N/A

### Flight controller

N/A

Zuldan, 2022-01-27:
[image]

Example of the bug ;-)

klutvott123, 2022-01-28:
I just checked this with the betaflight CMS script and the changes show up. Keep in mind that if you have the cms menu open while you're changing values in the configurator, you need to exit and enter cms again for the changes to whow up

Zuldan, 2022-01-30:
@klutvott123 thanks for checking your side. I still can't get the correct values to show in the OSD even after a full power cycle of the quad. It is also not connected to Betaflight.

closed 2022-03-08 by github-actions[bot]