betaflight/betaflight #11352: 4.3.0RC2 Motors Tab "Mixer Mode Problem Detected" BF thinks motors not tied to resources, when they are. FC is FLYWOOF745HEX

MrMPod, 2022-01-27:
### Describe the bug

Freshly upload new firmware 4.3 rc2
accept defaults;
go to motors tab; 
set Mixer: Hex X
set ESC/motor protocol: "DSHOT600",

and message showed:
[image]

### To Reproduce

Freshly upload new firmware 4.3 rc2
accept defaults;
go to motors tab; 
set Mixer: Hex X
set ESC/motor protocol: "DSHOT600",

### Expected behavior

be able to test motor configuration and have a Hex X working fine and arm the hexacopter (now after arming motors are not spining)spinning

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
# version
# Betaflight / STM32F745 (S745) 4.3.0 Jan  9 2022 / 11:51:13 (ee8560af4) MSP API: 1.44
# config: manufacturer_id: FLWO, board_name: FLYWOOF745, version: fda05330, date: 2020-11-21T10:29:52Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FLYWOOF745
manufacturer_id FLWO
mcu_id 001e00213135510b31303738
signature 

# name: MrMPod's LR AN

# feature
feature -RX_PARALLEL_PWM
feature -LED_STRIP
feature GPS
feature TELEMETRY

# serial
serial 1 2 115200 57600 0 115200
serial 2 64 115200 57600 0 115200

# mixer
mixer HEX6X
mmix 0  1.000 -0.500  0.866  1.000
mmix 1  1.000 -0.500 -0.866  1.000
mmix 2  1.000  0.500  0.866 -1.000
mmix 3  1.000  0.500 -0.866 -1.000
mmix 4  1.000 -1.000  0.000 -1.000
mmix 5  1.000  1.000  0.000  1.000

# map
map TAER1234

# aux
aux 0 0 1 1300 2100 0 0
aux 1 1 3 1700 2100 0 0
aux 2 2 3 1300 1700 0 0
aux 3 13 0 1700 2100 0 0

# master
set gyro_lpf1_static_hz = 325
set gyro_lpf2_static_hz = 650
set gyro_lpf1_dyn_min_hz = 325
set gyro_lpf1_dyn_max_hz = 650
set acc_calibration = -51,24,50,1
set mag_hardware = NONE
set baro_hardware = AUTO
set min_check = 1010
set fpv_mix_degrees = 25
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set failsafe_delay = 11
set failsafe_procedure = GPS-RESCUE
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 280
set vbat_warning_cell_voltage = 330
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_descent_dist = 30
set gps_rescue_min_sats = 5
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set simplified_gyro_filter_multiplier = 130
set osd_vbat_pos = 256
set osd_rssi_pos = 160
set osd_rssi_dbm_pos = 128
set osd_tim_1_pos = 311
set osd_tim_2_pos = 2487
set osd_flymode_pos = 224
set osd_vtx_channel_pos = 277
set osd_craft_name_pos = 2504
set osd_gps_speed_pos = 2231
set osd_gps_lon_pos = 2096
set osd_gps_lat_pos = 2128
set osd_gps_sats_pos = 2091
set osd_home_dir_pos = 2089
set osd_home_dist_pos = 2263
set osd_flight_dist_pos = 2199
set osd_altitude_pos = 2166
set osd_warnings_pos = 14729
set osd_avg_cell_voltage_pos = 2121
set osd_disarmed_pos = 363
set osd_flip_arrow_pos = 64
set osd_core_temp_pos = 2295
set osd_log_status_pos = 96
set gyro_1_align_yaw = 2700
set name = MrMPod's LR AN

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 90
set dterm_lpf1_dyn_max_hz = 180
set dterm_lpf1_static_hz = 90
set dterm_lpf2_static_hz = 180
set p_pitch = 39
set i_pitch = 70
set d_pitch = 32
set f_pitch = 104
set p_roll = 37
set i_roll = 67
set d_roll = 28
set f_roll = 100
set p_yaw = 37
set i_yaw = 67
set f_yaw = 100
set d_min_roll = 21
set d_min_pitch = 23
set simplified_master_multiplier = 70
set simplified_pi_gain = 120
set simplified_feedforward_gain = 120
set simplified_dterm_filter_multiplier = 120

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 1

# save configuration
save

  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: LED 1
A03: MOTOR 6
A04: FLASH_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: PINIO 1
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: MOTOR 5
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: SERIAL_TX 2
D06: SERIAL_RX 2
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH2 : DSHOT_BITBANG 3
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
    CH4 : DSHOT_BITBANG 5
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 5

### Flight controller

FLYWOOF745 - from flywoo.net

haslinghuis, 2022-01-27:
Please try `set dhot_bitbang = off`, and report `dma show` and `timer show`

MrMPod, 2022-01-27:
# set dshot_bitbang = OFF
dshot_bitbang set to OFF
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 5

# timer show

Currently active Timers:
-----------------------
TIM1:
    CH2 : DSHOT_BITBANG 3
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
    CH4 : DSHOT_BITBANG 5
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

MrMPod, 2022-01-27:
now I see all 6 motors
[image]

MrMPod, 2022-01-27:
> Please try `set dshot_bitbang = off`, and report `dma show` and `timer show`

the hex is arming and it is fly all is looking great :)
thank you very much Sir :)

closed 2022-01-27 by MrMPod