betaflight/betaflight #11356: HGLRC Zeus 5 - No more reboot after triming acceletometer

LoakAudio, 2022-01-28:
### Describe the bug

Hi all,
With previous Betaflight versions, when I trimmed the accelerometer with the sticks the quad was rebooting, triggering "Telemetry lost" and "Telemetry recovered".
I've update my Zeus 5 with Betaflight 4.3 last nighty build and no more reboot, no more 'Telemtry ...". I've found that I need to unplug and re-plug the battery to take the change into account.
My other quad, a Crux3nlr with the same Betaflight version is working as expected (telemetry lost, telemetry recovered and change taken into account).

### To Reproduce

Try this with an updated Zeus 5 and a HM EP1 RX on UART 2.

### Expected behavior

Change to be taken into account with or without reboot but without the need of unplugging re-plugging the battery.

### Flight controller configuration

```
# diff

# version
# Betaflight / STM32F411 (S411) 4.3.0 Jan 26 2022 / 13:08:18 (b6bc232a1) MSP API: 1.44
# config: manufacturer_id: HGLR, board_name: HGLRCF411, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

board_name HGLRCF411
manufacturer_id HGLR

# name: Nano LR

# resources
resource MOTOR 5 NONE
resource SERIAL_TX 11 B03
resource PINIO 1 B10

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY

# serial
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# beeper
beeper -BAT_LOW
beeper -ON_USB
beeper -BLACKBOX_ERASE

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 1300 2100 0 0
aux 2 27 2 1700 2100 0 0
aux 3 13 1 900 1300 0 0
aux 4 35 1 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905 5925    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 20 23 25
vtxtable powerlabels 25 100 200 MAX

# master
set dyn_notch_count = 5
set acc_trim_pitch = 4
set acc_trim_roll = 22
set acc_calibration = -23,27,-7,1
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT300
set motor_output_reordering = 2,3,0,1,4,5,6,7
set failsafe_off_delay = 50
set failsafe_throttle = 1300
set failsafe_switch_mode = STAGE2
set failsafe_procedure = AUTO-LAND
set align_board_roll = 180
set bat_capacity = 3000
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 290
set vbat_warning_cell_voltage = 310
set current_meter = NONE
set yaw_motors_reversed = ON
set small_angle = 180
set osd_warn_launch_control = OFF
set osd_rssi_dbm_alarm = -107
set osd_link_quality_pos = 2379
set osd_link_tx_power_pos = 2411
set osd_rssi_dbm_pos = 2348
set osd_tim_2_pos = 14710
set osd_flymode_pos = 14679
set osd_throttle_pos = 2369
set osd_crosshairs_pos = 2253
set osd_warnings_pos = 14601
set osd_avg_cell_voltage_pos = 14689
set osd_disarmed_pos = 14474
set osd_flip_arrow_pos = 14510
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_min_rssi_dbm = ON
set cpu_overclock = 120MHZ
set vtx_band = 1
set vtx_channel = 8
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5725
set vcd_video_system = NTSC
set pinio_config = 129,1,1,1
set pinio_box = 0,255,255,255
set gyro_1_align_yaw = 1800
set name = Nano LR

profile 0

# profile 0
set vbat_sag_compensation = 100
set p_pitch = 62
set i_pitch = 110
set d_pitch = 57
set f_pitch = 158
set p_roll = 59
set i_roll = 105
set d_roll = 50
set f_roll = 151
set p_yaw = 59
set i_yaw = 105
set f_yaw = 151
set d_min_roll = 39
set d_min_pitch = 44
set simplified_master_multiplier = 110
set simplified_d_gain = 120
set simplified_pi_gain = 120
set simplified_dmax_gain = 80
set simplified_feedforward_gain = 115

rateprofile 0

# rateprofile 0
set thr_expo = 50
set roll_srate = 36
set pitch_srate = 36
set yaw_srate = 36

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: ADC_BATT
B01: FREE
B02: BEEPER
B03: SERIAL_TX 11
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: PINIO 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2 : SERIAL_TX 11
TIM3:
    CH1 : MOTOR 1
    CH2 : MOTOR 2
TIM4:
    CH1 : MOTOR 3
    CH2 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: MOTOR 1
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```

### Flight controller

HGLRC Zeus 5

### Other components

HappyModel EP1
Rush Tiny Tank

### How are the different components wired up

HM EP1 on UART 2
Rush Tiny Tank on SoftSerial 1

LoakAudio, 2022-02-27:
Probably same as #11438

haslinghuis, 2022-04-02:
Duplicate of https://github.com/betaflight/betaflight/issues/11438

closed 2022-04-02 by haslinghuis

TrailBlazerCIED, 2022-06-21:
Out of nowhere sorry. How does one boot the Zeus5 AIO in DFU? Trying to Update from BF 4.29 to 4.3 and CLI>BL command is not working. I can seem to find a boot button on the board.