betaflight/betaflight #11360: GPS Rescue, possibility to use MAGNETOMETER for direction, instead of GPS

maRCnh33, 2022-01-28:
### Is your feature request related to a problem? Please describe

GPS is used for heading while using GPS RESCUE. So it can be used with boards that have no compass.

But i have a compass onboard and i like to use that for heading.

### Describe the solution you'd like

Should be nice if you can add an option like :

"Use magnetometer for GPS RESCUE for heading, instead of GPS"

### Describe alternatives you've considered

To use GPS wich i do not prefer...

etracer65, 2022-01-28:
Compass is already used if available.

https://github.com/betaflight/betaflight/wiki/GPS-rescue-mode#requirements

closed 2022-01-28 by etracer65

maRCnh33, 2022-01-28:
i like to set the minimal distance some closer to me, like 50meters.. 

200 is a bit too much. I can understand why it is 200meter when using only gps. 

with compass onboard i can set it to 50meter?