betaflight/betaflight #11363: Mutlishot doesn"t arm

Alveran81, 2022-01-29:
### Describe the bug

Motorprotokoll Multishot, Betaflight 4.3 RC2
When i want to arm nothing Happens and when i disarm FC is rebooting. 

### To Reproduce

Evrytime IT happens.

### Expected behavior

Motors dont start Spinning 

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```

### Flight controller

OmnibusF4SD

Quick-Flash, 2022-01-31:
what software is on your esc? is it bluejay or Jesc? if so it doesn't support multishot.

Alveran81, 2022-01-31:
Its latest Blheli Revision 14.9
The DYS XMS30A are very old ESCs and only support Oneshot 125 and Multishot.

haslinghuis, 2022-01-31:
Latest firmware seems to be: 16.7

Did you calibrate the ESC's: https://bluerobotics.com/wp-content/uploads/2018/10/BLHeli_S-manual-SiLabs-Rev16.x.pdf

Asizon, 2022-01-31:
@haslinghuis this is Blheli and not blheli S

Asizon, 2022-01-31:
https://github.com/bitdump/BLHeli/tree/master/Atmel/Hex%20files

Asizon, 2022-01-31:
Latest supported version for atmel blheli is 14.9

Alveran81, 2022-01-31:
Yes 14.9 is latest. I flashed back Betaflight 4.2.5 to try out it works fine and copter is flying after that if flashed 4.3.0 RC2 again. 
Same Problem. When i arm nothing happens and when i disarm fc is rebooting.
I think its a bug im implementation of Multishot in 4.3.0

Alveran81, 2022-02-01:
Same Problem in RC3 but i can Spin Motors in Betaflight Configurator im Motor Tab, this was not possible in RC2

haslinghuis, 2022-02-01:
Still think you need to calibrate your ESC

Alveran81, 2022-02-02:
With betaflight 4.2.x it works fine. With Betaflight 4.3.0 nothing happens when i arm and when i disarm flight Controller is rebooting. This has nothing to do with ESC calibration.

haslinghuis, 2022-02-08:
`resource show all` is missing `timer` and `dma` allocation. Please include this information.
You could also use `dma show` and `timer show`.

Also like to know if oneshot 42 or 125 is still working with RC3?

Alveran81, 2022-02-09:
Hello i kicked Out this old ESCs and built in newer with dshot. 
If you want it for develeopent i can attache one esc to another FC on a table youst for testing.

PR #11429 (2022-02-26): Fix non DSHOT protocols
Non DSHOT protocols are currently broken. At least multishot and oneshot.
Tested and confirmed working with all protocols except brushed and pwm, but that probably works too.

Fixes #11413, fixes #11363