betaflight/betaflight #11367: READY_BEEP

GregorPotrebujes, 2022-01-29:
### Describe the bug

Hi there,

can someone please check the functioning of the READY_BEEP switch, located in the Configuration tab under "Beeper configuration".
If I understand this correctly, there should be some kind of a beep when the GPS gets a fix (if there is a GPS attached to the FC and it's properly configured in the Betaflight)... Well, I have never ever heard that beep on any of my GPS drones... but I would really appreciate that function!

I know there is some option to count the satellites, but that's annoying and not very useful for me. So I would really love to see that READY_BEEP working sometime in the future.

### To Reproduce

Enable READY_BEEP in the Configuration tab under Beeper configuration. 
Properly attach GPS to the FC, configure it in the Ports tab and then also in the Failsafe tab, where you set GPS rescue as a stage 2 failsafe. Select your desired number of satellites for "minimum satellites". 
In the OSD tab place your "GPS sats" icon somewhere onto the screen.

### Expected behavior

When the "GPS sats" icon stops blinking (meaning the GPS position is locked) in the OSD and if there is >= minimum sattelites, some sort of a beep should be triggered, so that we know we are free to arm and fly out.

### Flight controller configuration

N/A

### Flight controller

N/A

### Other components

N/A

### How are the different components wired up

N/A

haslinghuis, 2022-01-30:
https://github.com/betaflight/betaflight/blob/master/docs/Buzzer.md

GregorPotrebujes, 2022-03-01:
So, have anyone figured the function of the READY_BEEP switch yet? :)

SteveCEvans, 2022-03-01:
> Just for reference: https://github.com/betaflight/betaflight/blob/master/docs/Buzzer.md

Hmmmm. Explains how to connect a buzzer to your Naze32 or CC3D…

haslinghuis, 2022-03-01:
https://intofpv.com/t-betaflight-cli-set-beeper-frequency
https://reefwing.medium.com/creating-a-new-target-in-betaflight-c0d879d42dbd

GregorPotrebujes, 2022-03-01:
@haslinghuis I appreciate your help, but I really don't see how these two links are in any way related to the issue... my beeper is connected properly and the frequency is set correctly. The beeper works well, but I just can't figure out how the READY_BEEP switch functions, or - to be more precise - how it shoud function.

GregorPotrebujes, 2022-03-13:
What I'm asking for here is an explanation/definiton of this READY_BEEP switch. Does anyone know where to find it?

GregorPotrebujes, 2022-04-13:
Just posting this message in order to keep the thread alive... 
So have anyone experimented with this flag?
[image]

klutvott123, 2022-04-13:
Please provide diff and the other requested information in the first post. Which BF version is this? Does it work with other versions?
Is it just the beep that doesn't occur? Does the GPS get a fix?

klutvott123, 2022-04-13:
Looks like that READY_BEEP hasn't been implemented in the gps code 🙂

klutvott123, 2022-04-13:
@GregorPotrebujes Could you please try this? https://github.com/betaflight/betaflight/pull/11521
It will play the ready beep when the gps gets a fix.

Relabeling this as a BUG

closed 2022-04-16 by haslinghuis