betaflight/betaflight #11368: Robust altitude estimation

bakwc, 2022-01-29:
### Is your feature request related to a problem? Please describe

Currently altitude estimation is not very robust, eg. it doesn't handle baro errors, don't perform sensor fusion and doesn't do a lot of other things that good altitude estimation should do.

### Describe the solution you'd like

High quality robust altitude estimation.

### Describe alternatives you've considered

Live with an old altitude estimation. It's sad.

### Other information

Here #11312

Quick-Flash, 2022-01-31:
The code currently does do sensor fusion, sure a rudimentary version, but the code does do sensor fusion. Also, barometers at low altitude tend to be inaccurate especially due to how the props mess with airflow and pressure. Unless the baro is handled correctly it is easily prone to a lot of bias.

bakwc, 2022-01-31:
> The code currently does do sensor fusion, sure a rudimentary version, but the code does do sensor fusion. Also, barometers at low altitude tend to be inaccurate especially due to how the props mess with airflow and pressure. Unless the baro is handled correctly it is easily prone to a lot of bias.

IMO current sensor fusion is not the way how it should be done.
1) It should use accelerometer
2) It should not be just some sort of averaging, it should use the fact that GPS based altitude doesn't drift at all (in time), but has large jumps. Baro usually has less jumps but it can drift. Acceleration has high accuracy on a short distance but the highest drift from then all.
3) It should check for all sensors errors, eg. when accelerometer shows that altitude doesnot change in a last few milliseconds and baro shows a 1/2 meter jump - it should ignore this baro measurement.

ledvinap, 2022-01-31:
@bakwc : 3. eg. part is bad idea. It will most probably end up as total headache, with tons of heuristics and accompanying edge cases. Keeping the algorithm is probably better long-term.

bakwc, 2022-02-04:
Made some tests.

### Baro behavior

1) No long-term drift (10 minutes flight, 20cm drift).
2) Medium occasionly short-term ocsillations (+-1 meter for 1 seconds)

### Gps behavior (tested on 10 sats and on 16 sats)

1) High mid-term and long-term drift (+-10 meters)
2) High occasionally short-term ocsillations (+-10 meters for 10 seconds, +-5 meters for 5 seconds)
3) +-10 meters drift observed even on 16 sats.

### Latency
Sometimes GPS reacts faster than baro (0.5 seconds), sometimes they have the same reaction.

Quick-Flash, 2022-02-08:
Baro isn't as accurate low to the ground, and weather and other factors will cause it to drift. GPS is noisy and accurate in the long run. Baro is typically better at reading changes in altitude while your GPS is better for long term readings. This means that by fusing them together with some form of complimentary filter you can get better results from them. You could have a look at what Inav is doing if you are curious. Inav won't listen to the baro when your low to the ground and will use acc readings along with baro+GPS to get position estimates.

Quick-Flash, 2022-02-08:
> In my tests - GPS altitude is completely unreliable. I will look at inav and ardupilot to check how they handle it. On 16 sats GPS I receive +-10 meters altitude changes, just staying on the ground without any actual movements. If someone want to help - I can prepare a firmware and instruction to reproduce tests, and post data here.

yes, +-10 meters is pretty normal for gps. You need to use it as something that is noisy, but is accurate in the long run. 

https://www.youtube.com/watch?v=whSw42XddsU
This video is a great example of complimentary filters in combining acc and gyro data, but you can think of it as baro and gps.

bakwc, 2022-02-08:
Inav doesn't use GPS for altitude estimation. It use it only for velocity estimations, and performs error-check before usage.
Source: https://github.com/iNavFlight/inav/blob/master/src/main/navigation/navigation_pos_estimator.c
(see how `ctx->estPosCorr.z` calculated and later applied to `posEstimator.est.pos`)

Quick-Flash, 2022-02-08:
> Inav doesn't use GPS for altitude estimation. It use it only for velocity estimations, and performs error-check before usage. Source: [...]

interesting, isn't that velocity estimate used for altitude estimates?

bakwc, 2022-02-08:
Don't know how it used later, probably velocity is used alone in some PID loops. Not for altitude estimation, altitude estimation calculated without velocity usage.

bakwc, 2022-02-12:
ArduPilot also use baro as a primary altitude source by default:
https://ardupilot.org/copter/docs/common-apm-navigation-extended-kalman-filter-overview.html#commonly-modified-parameters

Quick-Flash, 2022-02-12:
Seems like barometer isn't the best way to do it. A fusion of baro and gps should be better. Baro does drift... so using gps to correct for baro drift is likely the best way to do it.
[image]

bakwc, 2022-02-23:
We figured out that there is different barometers with different accuracy. Eg:
DSP310 - accuracy is +- 8 meters
BMP280 - accuracy is +- 1 meter
So if you have BMP280 - you will usually prefer it to a GPS, and for DSP310 a fusion will give better results.

tavdog, 2022-06-24:
Seems like a lot of work for something very few pilots care about.

closed 2022-07-01 by bakwc