betaflight/betaflight #11369: DJI with LQ & RSSI & warnings for Version 4.2.11

Alveran81, 2022-01-30:
### Is your feature request related to a problem? Please describe

Hello Mr. Howels,

I do not know how else to contact you. Can you please do that for Betaflight Version 4.2.11

All my three Copter doenst fly with 4.3.0 but your feature is very great. Thank you so much.
Sorry for this Issue. I know this is no Issue for Betafligth 4.3.0

https://github.com/howels/betaflight/releases/tag/4.3.0-RC2-DJI_OSD_LQ_warning

### Describe the solution you'd like

Build for 4.2.11 Plz

### Describe alternatives you've considered

Doesnt find it elsewhere

Asizon, 2022-01-30:
@Alveran81 you need to contact and DJI for that warnings, theh should add compatibility to their firmware.

Asizon, 2022-01-30:
PD Howel solution is not allowed hack here.

Alveran81, 2022-01-30:
Yes i know. Its a DJI problem and not a Betaflight problem. But his solution is working fine but i doesnt know how to contact him.

Asizon, 2022-01-30:
Yes, works but hacking other things, this adition is refused some tine ago. Also betaflight is ready to allow dji warnings, please contact dji and tell him, maybe he can add now.

howels, 2022-01-30:
Hi @Alveran81 as we are nearing the release of 4.3 I am focusing on supporting 4.3 with the patch and not 4.2.
I have been flying 4.3 builds since July 2021, it works great!

As @Asizon points out, the DJI OSD hack was discussed at length with betaflight devs and they do not wish to pursue workarounds or hacks for specific vendors. 

Please try the 4.3 builds and let me know how you get on.

closed 2022-01-30 by etracer65