betaflight/betaflight #11371: F722: High current values while idle

FireflySoul, 2022-01-31:
### Describe the bug

The current values displayed while idle are too high: 4-6A and varies over time. 

### To Reproduce

1) Flash latest version.
2) Apply defaults settings for iFlight F722 TWING board
3) Set current sensor scale value
4) Apply and check current value with battery plugged-in. 

### Expected behavior

Current values should be under 1A on idle (measured values are between 0.4 .. 0.6A).

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```

### Flight controller

DFR F722 Dual HD bought from https://www.drone-fpv-racer.com in the summer of 2021. Recommended target by manufacturer is  IFLIGHT_F722_TWING(IFRC).

### How are the different components wired up

The FC is wired to the  current sensor on a Matek FCHUB-6S PDB with modified scale of 375 (resistor replaced for a range of 90A).

### Add any other context about the problem that you think might be relevant here

To rule out hardware issues I've flashed iNav 4.0.0 (IFLIGHTF7 TWING) and got correct values(0.6A).

I've also checked voltage on the current pin of the FC and measured 210mV while FC was flashed with BF and 20mV when flashed with iNav. (maybe a wrong I/O config of the pin)

McGiverGim, 2022-02-03:
Have you selected the origin of your current sensor? I suspect it is using the "onboard" sensor, that is expecting a scale of 100.

haslinghuis, 2022-02-03:
Can you please try upgrading configurator 10.8RC3 and 4,3RC3.

FireflySoul, 2022-02-03:
Thanks for your replies!
Onboard ADC is the source for the current data.
Tried with configurator 10.8 RC3 and BF4.3 RC3 with same results - current shown 5.2A.

haslinghuis, 2022-02-03:
Was expecting a scale of 179. Voltage of the pin is 0-3.3V according to http://www.mateksys.com/?portfolio=fchub-6s

FireflySoul, 2022-02-03:
I have replaced the resistor on the PDB board to have a lower range (0-90A), that is why I use 375 for scale.

FireflySoul, 2022-02-03:
While looking at the unified target config file, trying to figure out what might be the problem, I've notice that there's a specific file for the DFR-F722 DUAL HD flight controller that I'm using.
After flashing with this target the ADC current readings are correct.

So in conclusion for anyone using the  DFR-F722 DUAL HD you should not use the recommended target from the manufacturer diagram(IFLIGHT_F722_TWING) and instead search for and use DFR_F722_DUAL_HD.

I'm sorry for not figuring this out before opening this issue, will try harder next time!

closed 2022-02-03 by FireflySoul

FireflySoul, 2022-02-04:
The issue reappeared again while setting up the quad in the configurator, the triggering setting was turning on d-shot.

It looks like this FC has a badly design pcb which results in crosstalk between the motors outputs and ADC inputs.

Tried reassigning the RSSI input as Current input but still no go because that input seems to also be affected.
Then used a capacitor connected at the current input pad and it lowered the noise enough to have stable and correct values registered by the ADC.

So entirely hardware issue. 🤦‍♂️