betaflight/betaflight #11375: problem with allocations of the DMA streams for BETAFPVF4SX1280 target - BF4.3.0 - RC3

Martivip, 2022-02-01:
### Describe the bug

The DMA streams are not allocated properly and that cause status ```gyro 1 locked``` instead of ```gyro 1 locked dma``` status.
Also warning message "Mixer mode problem detected" popup on the Motors configuration tab.
@SteveCEvans Please take a look on this bug.

**Status**
```
# Status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.28V, Core temp=51degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3773, max available: 16384
Devices detected: SPI:1
Gyros detected: gyro 1 locked
GYRO=MPU6000, ACC=MPU6000
OSD: MAX7456
System Uptime: 202 seconds, Current Time: 2022-02-01T13:59:35.854+00:00
CPU:28%, cycle time: 126, GYRO rate: 7936, RX rate: 15, System rate: 9
Voltage: 3 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: RXLOSS CLI MSP RPMFILTER DSHOT_BBANG

# 
```
**DMA**
```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### To Reproduce

Install the BF4.3.0-RC3 firmware for BETAFPVF4SX1280 target, enable Bidirectional DShot

### Expected behavior

DMA Streams should be allocated properly and the Gyro status should be ```gyro 1 locked dma```

### Flight controller configuration

```
 # 

# diff

# version
# Betaflight / BETAFPVF4SX1280 (BHER) 4.3.0 Feb  1 2022 / 20:04:33 (3267f0417) MSP API: 1.44

# start the command batch
batch start

board_name BETAFPVF4SX1280
manufacturer_id BEFH

# name: Meteor65-ELRS

# feature
feature -LED_STRIP
feature -ANTI_GRAVITY

# serial
serial 0 1 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# aux
aux 0 0 0 1950 2050 0 0
aux 1 1 1 1425 1575 0 0
aux 2 2 1 1925 2075 0 0
aux 3 35 2 1850 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 100 200 350

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 1000
set dyn_notch_count = 2
set dyn_notch_q = 450
set dyn_notch_min_hz = 100
set gyro_lpf1_dyn_min_hz = 0
set acc_trim_roll = 6
set acc_calibration = -10,-99,-110,1
set max_check = 2000
set rc_smoothing_auto_factor = 25
set rc_smoothing_auto_factor_throttle = 25
set dshot_idle_value = 600
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set bat_capacity = 300
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 300
set vbat_warning_cell_voltage = 334
set current_meter = NONE
set ibata_scale = 179
set small_angle = 180
set pid_process_denom = 4
set simplified_gyro_filter_multiplier = 200
set osd_warn_batt_not_full = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_link_quality = ON
set osd_warn_rssi_dbm = ON
set osd_link_quality_alarm = 30
set osd_rssi_dbm_alarm = -100
set osd_cap_alarm = 300
set osd_tim1 = 0
set osd_tim2 = 1
set osd_vbat_pos = 2401
set osd_rssi_pos = 193
set osd_link_quality_pos = 2263
set osd_link_tx_power_pos = 396
set osd_rssi_dbm_pos = 2232
set osd_tim_1_pos = 386
set osd_tim_2_pos = 2455
set osd_remaining_time_estimate_pos = 33
set osd_flymode_pos = 2442
set osd_throttle_pos = 2433
set osd_vtx_channel_pos = 2101
set osd_current_pos = 43
set osd_craft_name_pos = 393
set osd_home_dist_pos = 98
set osd_flight_dist_pos = 130
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 289
set osd_disarmed_pos = 2347
set osd_esc_tmp_pos = 163
set osd_esc_rpm_pos = 150
set osd_flip_arrow_pos = 2438
set osd_core_temp_pos = 2448
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_total_time = ON
set osd_stat_min_rssi_dbm = ON
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5658
set vcd_video_system = NTSC
set expresslrs_uid = 179,193,51,58,47,191
set expresslrs_rate_index = 1
set expresslrs_switch_mode = WIDE
set name = Meteor65-ELRS

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 150
set dterm_lpf1_dyn_max_hz = 300
set dterm_lpf1_static_hz = 150
set dterm_lpf2_static_hz = 300
set vbat_sag_compensation = 100
set anti_gravity_gain = 4500
set iterm_relax_cutoff = 20
set p_pitch = 71
set i_pitch = 63
set d_pitch = 67
set f_pitch = 249
set p_roll = 68
set i_roll = 60
set d_roll = 62
set f_roll = 239
set p_yaw = 68
set i_yaw = 60
set f_yaw = 239
set level_limit = 65
set d_min_roll = 62
set d_min_pitch = 67
set d_max_advance = 0
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 40
set feedforward_jitter_factor = 5
set feedforward_boost = 18
set feedforward_max_rate_limit = 95
set simplified_master_multiplier = 105
set simplified_i_gain = 50
set simplified_d_gain = 200
set simplified_pi_gain = 145
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 190
set simplified_pitch_d_gain = 95
set simplified_dterm_filter_multiplier = 200

rateprofile 0

# rateprofile 0
set roll_rc_rate = 10
set pitch_rc_rate = 10
set yaw_rc_rate = 10
set roll_expo = 54
set pitch_expo = 54
set yaw_expo = 54
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 70
set tpa_breakpoint = 1250

# end the command batch
batch end

# 
```
```
 # 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 2
A01: ADC_BATT
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: SERIAL_TX 11
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: RX_SPI_EXPRESSLRS_BUSY
A14: BEEPER
A15: RX_SPI_CS
B00: FREE
B01: FREE
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: GYRO_EXTI
B07: MOTOR 4
B08: MOTOR 1
B09: RX_SPI_EXPRESSLRS_RESET
B10: MOTOR 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: RX_SPI_EXTI
C14: LED 1
C15: LED
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : SERIAL_TX 11
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Flight controller

BETA FPV 5A AIO ELRS

### Other components

Meteor65Pro

EastCoaster, 2022-02-03:
Issue with Mixer mode. Opened a ticket with details. https://github.com/betaflight/betaflight/issues/11382

haslinghuis, 2022-02-04:
@Martivip please try disabling `dshot_bitbang` and report back with:
```
dma show
timer show
```

SteveCEvans, 2022-02-04:
Turn off soft serial to fix this. It’s using TIM1 which bitbanged DSHOT requires.

If you need to use smart audio then connect that to TX1 instead.

SteveCEvans, 2022-02-04:
Note that status `gyro 1 locked` instead of `gyro 1 locked DMA` is due to `dshot_bitbang` being on. Try setting that to off if you want to use soft serial. Report back with `dma show` and `timer show` outputs with it on and off.

Martivip, 2022-02-04:
@haslinghuis  ok ```dshot_bitbang``` set to OFF

```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: MOTOR 3
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```
```
# timer show

Currently active Timers:
-----------------------
TIM1:
    CH1 : SERIAL_TX 11
TIM2:
    CH1 : MOTOR 2
    CH3 : MOTOR 3
TIM3: FREE
TIM4:
    CH2 : MOTOR 4
    CH3 : MOTOR 1
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
```

Martivip, 2022-02-04:
@haslinghuis @SteveCEvans and with ```dshot_bitbang``` set to ON

```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```
```
# timer show

Currently active Timers:
-----------------------
TIM1:
    CH1 : SERIAL_TX 11
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
```

Martivip, 2022-02-04:
@SteveCEvans ok then will use TX1 instead soft serial for my Smart Audio purposes, thanks for the info.

SteveCEvans, 2022-02-04:
So the issue here is that `SPI_MOSI 3` couldn’t be allocated so no DMA available for the SPI RX. Not the end of the world. You have `SPI_MOSI 2`, but no `SPI_MISO 2`, but that’s fine for the OSD.

if you disable SOFTSERIAL then `dshot_bitbang=ON` will work and you’ll get `locked dma`.

To use soft serial, use `dshot_bitbang=OFF`.

Either should work fine.

SteveCEvans, 2022-02-04:
> @haslinghuis @SteveCEvans and with `dshot_bitbang` set to ON
>
> ```
> # dma show
>
> Currently active DMA:
> --------------------
> DMA1 Stream 0: SPI_MISO 3
> DMA1 Stream 1: FREE
> DMA1 Stream 2: FREE
> DMA1 Stream [...]

Can you please repeat with SOFTSERIAL disabled.

Martivip, 2022-02-04:
@SteveCEvans ok ```dshot_bitbang``` set to ON and soft serial disabled

```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```
```
# timer show

Currently active Timers:
-----------------------
TIM1:
    CH1 : DSHOT_BITBANG 2
    CH2 : DSHOT_BITBANG 1
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
```

SteveCEvans, 2022-02-04:
@Martivip Apologies for all the requests for CLI info, but I’m away from home at the moment, so can’t check myself, and want to be sure I’m giving you good information!

SteveCEvans, 2022-02-04:
That looks OK now. I’ll be interested to hear if you think it flies any different with bitbanged DSHOT on/off.

SteveCEvans, 2022-02-04:
Also please report back if you get any failsafes.

Martivip, 2022-02-04:
@SteveCEvans sure Thank You and I I will try to fly in both scenarios and will see, but for all stuff goodness in BF 4.3 I will disable soft serial and I reorganize my SA connection to TX1.

I'm just curious that issue is related to some F4 limitations or code improvements ?

SteveCEvans, 2022-02-04:
Bitbanged DSHOT requires TIM1 or TIM8, and the F411 doesn’t have TIM8. Unfortunately the choice of pin for SOFTSERIAL on this FC means it has to use TIM1 too, so we have to choose one or the other.

4.3 introduces DMA on SPI which is particularly useful for FLASH (black box) access, OSD, and potentially SPI RX as well as the gyro. 

Unfortunately there are limited resources available for DMA and if timer based DSHOT is used then four of the sixteen DMA streams get used for MOTOR 1-4. These typically clash with SPI_MISO/SPI_MOSI streams needed for the above DMA. A better option is to use bitbanged DSHOT as this only uses one stream per port (so if your motor pins are on ports A and B, for example you’ll need two streams) and these streams will be on DMA2 which won’t conflict with the SPI DMA streams for SPI 2 and SPI 3.

Alas there’s a bug in F4’s which means that we can’t use bitbanged access on DMA2 at the same time as SPI DMA on DMA2, and typically gyros are on SPI 1 which is on DMA2.

The upshot of this is that in order to get DMA access to OSD/FLASH/SPI RX, on most FCs we have to enable bitbanged DSHOT and lose gyro DMA as a consequence. We also lose anything else that uses TIM1, which may impact SOFTSERIAL or LEDSTRIP.

Careful selection of the pins used for the motors can allow for four MOTOR resources and all the SPI DMA resources at the same time. Then we get the best of everything. The FURYF4OSD got this right!

Martivip, 2022-02-04:
@SteveCEvans always top explanations thanks.
right noe I sitting on my bench and unfortunately when I only disable soft serial on ports tab and I leave the ```dshot_bitbang``` set to ON the status show only ```gyro 1 locked``` not DMA

SteveCEvans, 2022-02-04:
Correct, because enabling bitbanged DSHOT means we can’t do SPI DMA on SPI 1. If you disable it then you’ll get `DMA`. Both will work fine, but I’m curious to know if you think either flies better than the other.

To be honest the differences are very small and are currently completely masked by ELRS timing issues which I’m working on using that FC for development. The next release candidate should see a big improvement on this FC.

Martivip, 2022-02-04:
@SteveCEvans ok then I will wait for the RC4, because I've rewired my SA control to TX1 and VTX control don't work properly or at all :( channel is changing but for a wile it reverting to previously set.

But I see improvement in RC3 that I'm capable to use with Bidirectional Dshot 4K pid loop due to less CPU usage 👍 , on RC2 max 2K works OK

SteveCEvans, 2022-02-04:
Are you sure you’ve configured UART1 correctly? SA works for me.

Martivip, 2022-02-04:
@SteveCEvans yes everything is weird and set correctly because I see that connection is ON, this issue is related rather to ELSR code because this target BETAFPVF4SX1280 has SPI ELRS RX. On my F745 target with external RX everything works OK

closed 2022-03-12 by Martivip