betaflight/betaflight #11376: H743 AIO - bidirectional dshot performing erratically - motors 3/4 sometimes show no RPM data

howels, 2022-02-01:
### Describe the bug

I have an H743 AIO. The board is the JHEH743. I am aware of the lack of software support (I am not the manufacturer or associated with them) but I intend to use the board and attempt to create a unified-target for it.

In testing the board through the original firmware, RC2, RC3 and other builds I noticed erratic behaviour with bidirectional dshot. The board is using bitbang but sometimes I would see motors fail to initialise and get no RPM data in configurator:

Sometimes it works perfectly:
[image]

But sometimes motors 3/4 just don't show up. I tried disabling bitbang but the results are worse:
[image]

One thing I have found is that if I build betaflight with `DEBUG=GDB`as mentioned in https://github.com/betaflight/betaflight/pull/10383#issuecomment-932937466 then RPM data is working. Without `DEBUG=GDB`the RPM data fails 99% of the time and only very occasionally when I start the board do I get RPM data.

As this build option changes optimisations from `-02` to `-0g` it appears that this differing optimisation is changing the ESC initialisation for H743. I am currently seeing https://github.com/betaflight/betaflight/commit/074172b4c94206e33a42d3aa5d8e4da2bd1a3e6d built with `DEBUG=GDB` work whereas the final RC3 build causes motors 3/4 to return no data.

ESCs have been tested using BlueJay 0.14, 0.16 (same problem on both). Testing on BLEHLI-S 16.71 showed erroneous RPM data of 666666 RPM on one motor and appeared to behave worse than BlueJay which would either show correct data or 0 RPM.

### To Reproduce

Flash RC3 for H743

### Expected behavior

bidirectional dshot should work and return RPM data from all 4 ESC.

### Flight controller configuration

```
 # diff

# version
# Betaflight / IFLIGHT_H743_AIO (IF7A) 4.3.0 Jan 31 2022 / 12:44:51 (074172b4c) MSP API: 1.44

# start the command batch
batch start

board_name IFLIGHT_H743_AIO_V2

# name: whatever

# resources
resource MOTOR 1 E09
resource MOTOR 2 B00
resource MOTOR 3 E11
resource MOTOR 4 B01

# timer
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)
timer D13 AF2
# pin D13: TIM4 CH2 (AF2)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer A00 AF2
# pin A00: TIM5 CH1 (AF2)
timer A01 AF2
# pin A01: TIM5 CH2 (AF2)
timer A02 AF2
# pin A02: TIM5 CH3 (AF2)
timer B10 AF1
# pin B10: TIM2 CH3 (AF1)
timer B11 AF1
# pin B11: TIM2 CH4 (AF1)

# dma
dma pin A08 10
# pin A08: DMA2 Stream 2 Request 11
dma pin B03 0
# pin B03: DMA1 Stream 0 Request 19
dma pin D13 5
# pin D13: DMA1 Stream 5 Request 30
dma pin C09 7
# pin C09: DMA1 Stream 7 Request 50
dma pin A00 0
# pin A00: DMA1 Stream 0 Request 55
dma pin A01 0
# pin A01: DMA1 Stream 0 Request 56
dma pin A02 0
# pin A02: DMA1 Stream 0 Request 57
dma pin B10 0
# pin B10: DMA1 Stream 0 Request 20
dma pin B11 0
# pin B11: DMA1 Stream 0 Request 21

# feature
feature -RSSI_ADC
feature -LED_STRIP
feature -AIRMODE
feature GPS

# serial
serial 0 64 115200 57600 0 115200
serial 2 2 115200 57600 0 115200
serial 5 1 115200 57600 0 115200

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 900 1300 0 0
aux 2 2 2 1300 1700 0 0
aux 3 46 1 1700 2100 0 0
aux 4 13 5 1700 2100 0 0
aux 5 35 4 1700 2100 0 0
aux 6 37 1 1300 1700 0 0

# rxrange
rxrange 0 988 2011
rxrange 1 988 2011
rxrange 2 988 2011
rxrange 3 988 2011

# rxfail
rxfail 11 s 1000

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 0
set gyro_to_use = FIRST
set dyn_notch_count = 1
set dyn_notch_q = 500
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 400
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_calibration = 68,-14,1,1
set mag_hardware = NONE
set rssi_channel = 12
set serialrx_provider = CRSF
set blackbox_sample_rate = 1/8
set dshot_idle_value = 1000
set dshot_bidir = ON
set dshot_bitbang = ON
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 10
set failsafe_procedure = GPS-RESCUE
set align_board_yaw = -90
set vbat_max_cell_voltage = 439
set ibata_scale = 120
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set gps_auto_baud = ON
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 2
set simplified_gyro_filter = OFF
set simplified_gyro_filter_multiplier = 0
set osd_rssi_pos = 2497
set osd_link_quality_pos = 2465
set osd_flymode_pos = 2520
set osd_throttle_pos = 2501
set osd_crosshairs_pos = 2286
set osd_current_pos = 2483
set osd_mah_drawn_pos = 2489
set osd_gps_speed_pos = 2080
set osd_gps_lon_pos = 2065
set osd_gps_lat_pos = 2048
set osd_gps_sats_pos = 2065
set osd_home_dir_pos = 2063
set osd_altitude_pos = 2103
set osd_warnings_pos = 14825
set osd_avg_cell_voltage_pos = 2516
set osd_disarmed_pos = 2506
set osd_flip_arrow_pos = 2469
set osd_displayport_device = MAX7456
set debug_mode = GYRO_SCALED
set gyro_1_sensor_align = DEFAULT
set gyro_1_align_yaw = 1800
set name = 2Fiddy

profile 2

# profile 2
set dterm_lpf1_type = BIQUAD
set dterm_lpf1_static_hz = 0
set dterm_lpf2_static_hz = 0
set vbat_sag_compensation = 100
set p_pitch = 82
set i_pitch = 147
set d_pitch = 92
set f_pitch = 218
set p_roll = 78
set i_roll = 140
set d_roll = 80
set f_roll = 210
set p_yaw = 78
set i_yaw = 140
set f_yaw = 210
set d_min_roll = 60
set d_min_pitch = 68
set motor_output_limit = 95
set auto_profile_cell_count = 3
set thrust_linear = 25
set dyn_idle_min_rpm = 13
set simplified_master_multiplier = 175
set simplified_d_gain = 115
set simplified_dterm_filter = OFF
set simplified_dterm_filter_multiplier = 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 1
set roll_expo = 54
set pitch_expo = 54
set yaw_expo = 54
set roll_srate = 100
set pitch_srate = 100
set yaw_srate = 94

# end the command batch
batch end

```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: MOTOR 2
B01: MOTOR 4
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: PREINIT 2
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_EXT
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: CAMERA_CONTROL
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: LED 1
C14: FREE
C15: FREE
D00: GYRO_EXTI
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: FREE
E02: SPI_SCK 4
E03: FREE
E04: OSD_CS
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: MOTOR 1
E10: FREE
E11: MOTOR 3
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : DSHOT_BITBANG 5
    CH2 : DSHOT_BITBANG 2
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: DSHOT_BITBANG 5
DMA1 Stream 1: DSHOT_BITBANG 2
DMA1 Stream 2: SPI_MOSI 1
DMA1 Stream 3: SPI_MISO 1
DMA1 Stream 4: SPI_MOSI 3
DMA1 Stream 5: SPI_MISO 3
DMA1 Stream 6: SPI_MOSI 4
DMA1 Stream 7: SPI_MISO 4
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```

### Flight controller

JHEMCU JHEH743 

### Other components

only motors are connected. ESCs are part of the AIO board.

### How are the different components wired up

Motors are wired to the ESCs (they spin without issues when ESCs initialise correctly).

daleckystepan, 2022-02-01:
Thank you for reporting this however we are aware of it https://github.com/betaflight/betaflight/issues/11155 . Solving this type of issue takes some time to be able to properly debug it and fix it. So no need to create another one - we are on it.

howels, 2022-02-01:
> Thank you for reporting this in details however we are aware of it #11155 . Solving this type of issue takes some time to be able to properly debug it and fix it. So no need to create another one - we are on it.

Thanks, just wanted to post more details in case it would help.

daleckystepan, 2022-02-01:
@howels If I understand it correctly it sometimes works and sometimes does not. So it probably depends on initialization of FC. Can you boot your FC few times in a row and send us output of `resource show all`?  In the best case send one when ESC/motors working correctly and one when ESC/motos don't work. Please don't change any settings during this test. Thank you.

ot0tot, 2022-02-01:
I think this is an issue:
```# dma
dma pin A08 10
# pin A08: DMA2 Stream 2 Request 11
dma pin B03 0
# pin B03: DMA1 Stream 0 Request 19
dma pin D13 5
# pin D13: DMA1 Stream 5 Request 30
dma pin C09 7
# pin C09: DMA1 Stream 7 Request 50
dma pin A00 0
# pin A00: DMA1 Stream 0 Request 55
dma pin A01 0
# pin A01: DMA1 Stream 0 Request 56
dma pin A02 0
# pin A02: DMA1 Stream 0 Request 57
dma pin B10 0
# pin B10: DMA1 Stream 0 Request 20
dma pin B11 0
# pin B11: DMA1 Stream 0 Request 21```

On H7, `dma pin Xxx 0` tells the DMAMUX1 to assign that to MUX1 Ch0, which is DMA1 Stream 0. This is in contrast to F4/F7 boards that use `dma pin Xxx 0` to mean the primary DMA assignment for that pin. If DMA is not being used for these pins (which it doesn't look like they are assigned to resources), I believe it they should be `dma pin Xxx NONE`.

howels, 2022-02-01:
Thanks for the responses everyone. I'll check tomorrow when I can do lots of beeping!

howels, 2022-02-02:
> @howels If I understand it correctly it sometimes works and sometimes does not. So it probably depends on initialization of FC. Can you boot your FC few times in a row and send us output of `resource show all` and `dma`? In the best case send one when ESC/motors are working correctly and one when ESC/motors don't work. Please don't change any settings during this test. Thank you.

*MOTOR 3 NOT WORKING*

Motor 3 not working but RPM readings working on motors 1,2,4

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: MOTOR 2
B01: MOTOR 4
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: PREINIT 2
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_EXT
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: CAMERA_CONTROL
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: LED 1
C14: FREE
C15: FREE
D00: GYRO_EXTI
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
[... 556 lines cut ...]
dma pin B05 5
# pin B05: DMA1 Stream 5 Request 24
dma pin E13 6
# pin E13: DMA1 Stream 6 Request 13
dma pin E14 7
# pin E14: DMA1 Stream 7 Request 14
dma pin A08 10
# pin A08: DMA2 Stream 2 Request 11
dma pin B03 0
# pin B03: DMA1 Stream 0 Request 19
dma pin D13 5
# pin D13: DMA1 Stream 5 Request 30
dma pin C09 7
# pin C09: DMA1 Stream 7 Request 50
dma pin A00 0
# pin A00: DMA1 Stream 0 Request 55
dma pin A01 0
# pin A01: DMA1 Stream 0 Request 56
dma pin A02 0
# pin A02: DMA1 Stream 0 Request 57
dma pin B10 0
# pin B10: DMA1 Stream 0 Request 20
dma pin B11 0
# pin B11: DMA1 Stream 0 Request 21
```

daleckystepan, 2022-02-02:
Cannot see any difference between working and non-working dump.

howels, 2022-02-03:
> Cannot see any difference between working and non-working dump.

Yup. There's no difference with bitbang. 2 motors per channel and only one motor is offline so there are still the same bitbang DMA and timers  in use

howels, 2022-02-18:
> 
RC3 with this fix seems good - latest master has developed some issues connecting to USB where device-descriptors aren't being sent but that doesn't seem related to this issue.
All 4 motors report RPM data and they seem to do so reliably across 10+ FC boots! Huge thanks to you @hydra for your expertise here. Do we incur a noticeable performance penalty disabling the data cache?

Here's the lack of USB descriptors after flashing latest master.

```
[44539.525801] usb 7-2.4.1: new full-speed USB device number 41 using xhci_hcd
[44552.881729] usb 7-2.4.1: device descriptor read/64, error -32
[44561.273671] usb 7-2.4.3: new full-speed USB device number 42 using xhci_hcd
[44576.701571] usb 7-2.4.3: device descriptor read/64, error -110
[44592.305468] usb 7-2.4.3: device descriptor read/64, error -110
[44592.497459] usb 7-2.4.3: new full-speed USB device number 43 using xhci_hcd
[44607.925381] usb 7-2.4.3: device descriptor read/64, error -110
[44623.545267] usb 7-2.4.3: device descriptor read/64, error -110
[44623.657333] usb 7-2.4-port3: attempt power cycle
[44624.265252] usb 7-2.4.3: new full-speed USB device number 44 using xhci_hcd
[44629.345260] xhci_hcd 0000:0b:00.3: Timeout while waiting for setup device command
[44634.721227] xhci_hcd 0000:0b:00.3: Timeout while waiting for setup device command
[44634.929191] usb 7-2.4.3: device not accepting address 44, error -62
[44635.009192] usb 7-2.4.3: new full-speed USB device number 45 using xhci_hcd
[44640.101175] xhci_hcd 0000:0b:00.3: Timeout while waiting for setup device command
[44645.473131] xhci_hcd 0000:0b:00.3: Timeout while waiting for setup device command
[44645.681113] usb 7-2.4.3: device not accepting address 45, error -62
[44645.681258] usb 7-2.4-port3: unable to enumerate USB device

```

Normal connections look like:

```
[44484.734161] usb 7-2.4.3: new full-speed USB device number 40 using xhci_hcd
[44484.854157] usb 7-2.4.3: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[44484.854168] usb 7-2.4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[44484.854172] usb 7-2.4.3: Product: IFLIGHT_H743_AIO_V2
[44484.854174] usb 7-2.4.3: Manufacturer: Betaflight
[44484.854176] usb 7-2.4.3: SerialNumber: 387E347B3330
[44484.891202] cdc_acm 7-2.4.3:1.0: ttyACM0: USB ACM device
```

haslinghuis, 2022-02-18:
@howels thanks for pointing out the USB descriptor problem! This issue is also observed with F4 / F7 MCU's.

hydra, 2022-02-18:
@howels As I understand your comment, you saying that when you disabled the cpu data caches you didn't have any problems.  please confirm.

Yes, there is a huge performance drop when disabling the data cache and it is not a long-term solution.

If it works fine with the cpu data cache disabled then there is a cache management issue in the dshot buffers that needs to be fixed.  IIRC @SteveCEvans  might know something about this?  I recall reading something related the dshot and dma recently but can't remember where...

The issue with USB descriptors should be moved into a separate issue and linked to your comment.

howels, 2022-02-18:
@hydra I confirm - disabling the data caches in the specific line you referenced has resolved the problem in my testing so far. 
Testing consisted of 10 board restarts with MCP USB connection and power to the ESCs so I could observe motor RPM data. Previously I would not get more than 4 startups in a row with 4/4 motors reporting RPM data.

klutvott123, 2022-02-18:
@howels can you please try this
[betaflight_4.3.0_IFLIGHT_H743_AIO_V2_norevision.zip](https://github.com/betaflight/betaflight/files/8100950/betaflight_4.3.0_IFLIGHT_H743_AIO_V2_norevision.zip)
 Not sure which target you actually need for that thing so this file contains IFLIGHT_H743_AIO_V2 and V1 and the unified target.

howels, 2022-02-18:
> @howels can you please try this [betaflight_4.3.0_IFLIGHT_H743_AIO_V2_norevision.zip](https://github.com/betaflight/betaflight/files/8100950/betaflight_4.3.0_IFLIGHT_H743_AIO_V2_norevision.zip) Not [...]

Sure, what are the changes in this build?

klutvott123, 2022-02-19:
Invalidates the D cache per motor.
This is how it currently works
https://github.com/betaflight/betaflight/blob/ade559731c628d6fe1e1ad6bc9993a143e5f4856/src/main/drivers/dshot_bitbang.c#L496-L508

Looks to me like the cache wont be invalidated if the motors are on the same port and only partially invalidated if some are on the same port. I'm not very familiar with this particular code so I might be wrong. I don't have the equipment to test it either lol

SteveCEvans, 2022-02-19:
I’ll sort this out this evening. Thanks for your investigations into this.

howels, 2022-02-19:
> Invalidates the D cache per motor. This is how it currently works
>
> https://github.com/betaflight/betaflight/blob/ade559731c628d6fe1e1ad6bc9993a143e5f4856/src/main/drivers/dshot_bitbang.c#L496-L508 [...]

Worked ok for the first 3 boots, then I hit this - so it isn't as reliable as disabling the data cache: 
[image]

Restarting the FC in rapid succession seems to be the best way to trigger this condition.

howels, 2022-02-20:
> @hydra I confirm - disabling the data caches in the specific line you referenced has resolved the problem in my testing so far. I have passed a build to another h743 user to confirm and they will test over the weekend. Testing consisted of 10 board restarts with MCP USB connection and power to the ESCs so I could observe motor RPM data. Previously I would not get more than 4 startups in a row with 4/4 motors reporting RPM data.

Another report from a user with the JHEH743 board saying that the build with data cache disabled allows bidirectional dshot to work.

SteveCEvans, 2022-02-20:
See https://github.com/betaflight/betaflight/issues/11155#issuecomment-1046136342 Diatone are sending me the ESCs which are mentioned in this issue.

hydra, 2022-02-21:
@alphacharlie the mentioned 2.2.1 (timer breaks) and 2.2.15 (hrtim) do not apply.  

we are not using timer breaks or the hrtim peripheral.

[image]

Not used.

[image]

Not used.

alphacharlie, 2022-02-21:
@hydra - Thanks for the info.  (And thanks to you and everybody on the Betaflight team for making such a great piece of software!)

I'm very much hoping this issue gets resolved. (But I'm not holding my breath. I still wonder if there isn't something wrong in the H743 hardware itself causing this. )

Cheers!

konehead, 2022-03-08:
I have the same JHEMCU H743 AIO board and am experiencing the same issue.
Just wanted to add my name here, in the case you need any additional testers for this board; I'm happy to help.
Thanks to all for the hard work on betaflight!

KevlarFPV, 2022-03-10:
im having the same issue with the I flight Beast f7 aio v1.2 had to factory rest mine to work again hope there will be a fix to these issues.

howels, 2022-03-10:
> im having the same issue with the I flight Beast f7 aio v1.2 had to factory rest mine to work again hope there will be a fix to these issues.

Did your iFlight H743 AIO v2  fail to boot and fail to connect with configurator (requiring boot button to recover) or did you factory reset just to get back to default settings?

howels, 2022-04-01:
I have added the JHEH743 (https://github.com/betaflight/unified-targets/pull/569) and combined with the fix to lower GPIO pin speed on H7 (https://github.com/betaflight/betaflight/pull/11494) this issue is now resolved.

closed 2022-04-01 by howels

konehead, 2022-04-01:
> I have added the JHEH743 ([betaflight/unified-targets#569](https://github.com/betaflight/unified-targets/pull/569)) and combined with the fix to lower GPIO pin speed on H7 (#11494) this issue is now resolved.

This is awesome that we now have an official target! Thank you.
I'm not sure the DShot issue is fixed though?  I updated my board with the new target with a full chip erase and enabled BDshot and I still have 100% errors on motors 1 & 3.  Using Bluejay version 0.16.  
Is there something else I need to do?
Thanks,

howels, 2022-04-01:
> This is awesome that we now have an official target! Thank you. I'm not sure the DShot issue is fixed though? I updated my board with the new target with a full chip erase and enabled BDshot and I [...]

Have you enabled bitbang? I built BF from master today and together with BlueJay (Beacon Test version, not the latest version) the latest BF code spins all 4 motors with RPM data.

konehead, 2022-04-01:
> > > I have added the JHEH743 ([betaflight/unified-targets#569](https://github.com/betaflight/unified-targets/pull/569)) and combined with the fix to lower GPIO pin speed on H7 (#11494) this issue is now resolved.
> > 
> > 
> > This is awesome that we now have an official target! Thank you. I'm not sure the DShot issue is fixed though? I updated my board with the new target with a full chip erase and enabled BDshot and I still have 100% errors on motors 1 & 3. Using Bluejay version 0.16. Is there something else I need to do? Thanks,
> 
> Have you enabled bitbang? I built BF from master today and together with BlueJay (Beacon Test version, not the latest version) the latest BF code spins all 4 motors with RPM data. Note - PR was only merged 12 hours ago so might not be in the latest nightly downloads yet.
> 
> [betaflight_4.3.0_STM32H743_bef4c9a25.hex.zip](https://github.com/betaflight/betaflight/files/8398197/betaflight_4.3.0_STM32H743_bef4c9a25.hex.zip)

Apologies, the first time I tried only BF RC3 release.  
I just tried the latest nightly build and everything is working perfect !!!
Thank you so much for this fix.  Really appreciate your time and effort.

howels, 2022-04-01:
Was mainly the smarter people who understood how the motor pins behave on the H7 but big thanks to everyone involved!