betaflight/betaflight #11382: Problem with Mixer mode for BETAFPVF4SX1280 target - BF4.3.0 - RC3

EastCoaster, 2022-02-03:
### Describe the bug

No motors deteced after upgrading to RC3.

### To Reproduce

Installed RC3 and got the following error at the motor screen.
[image]

### Expected behavior

Motors to show on the motor tab

### Flight controller configuration

# 

# diff all

# version
# Betaflight / BETAFPVF4SX1280 (BHER) 4.3.0 Nov  3 2021 / 15:43:00 (0265565fa) MSP API: 1.44

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name BETAFPVF4SX1280
manufacturer_id BEFH
mcu_id 004200413430510333353839
signature 

# name: Meteor65-ELRS

# feature
feature -LED_STRIP
feature -ANTI_GRAVITY

# serial
serial 0 1 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# map
map TAER1234

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 35 2 1300 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 100 200 350

# master
set gyro_notch1_hz = 400
set gyro_notch2_hz = 200
set acc_trim_pitch = 2
set acc_calibration = 108,-127,111,1
set max_check = 2000
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 300
set current_meter = NONE
set ibata_scale = 179
set small_angle = 180
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_vbat_pos = 2433
set osd_rssi_pos = 2241
set osd_link_quality_pos = 2273
set osd_rssi_dbm_pos = 2305
set osd_tim_1_pos = 386
set osd_tim_2_pos = 2455
set osd_remaining_time_estimate_pos = 33
set osd_flymode_pos = 2423
set osd_throttle_pos = 2401
set osd_vtx_channel_pos = 2080
set osd_crosshairs_pos = 2253
set osd_current_pos = 343
set osd_craft_name_pos = 2441
set osd_home_dist_pos = 98
set osd_flight_dist_pos = 130
set osd_warnings_pos = 2378
set osd_avg_cell_voltage_pos = 289
set osd_disarmed_pos = 2315
set osd_esc_tmp_pos = 163
set osd_esc_rpm_pos = 150
set osd_core_temp_pos = 278
set vtx_band = 4
set vtx_channel = 4
set vtx_power = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5800
set vcd_video_system = NTSC
#set expresslrs_hybrid_switches = ON
set name = Meteor65

profile 0

profile 1

profile 2

# profile 2
set p_pitch = 51
set i_pitch = 99
set d_pitch = 42
set f_pitch = 105
set p_roll = 46
set i_roll = 94
set d_roll = 39
set f_roll = 99
set p_yaw = 50
set i_yaw = 99
set f_yaw = 99
set d_min_roll = 25
set d_min_pitch = 28

# restore original profile selection
profile 2

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 

### Flight controller

BETAFPVF4SX1280

haslinghuis, 2022-02-03:
Please also provide `resource show all`

EastCoaster, 2022-02-04:
> 

I did not capture that at the time of the issue but did grab a screenshot of the resource command.

haslinghuis, 2022-02-04:
`dma show` and `timer show`

k-trahan, 2022-02-04:
I find with this board that I have to set `dshot_bitbang = OFF`.

Martivip, 2022-02-04:
@haslinghuis please look at #11375 I have the same issue

haslinghuis, 2022-02-04:
Duplicate of #11375

closed 2022-02-04 by haslinghuis

SteveCEvans, 2022-02-04:
Turn off soft serial.