betaflight/betaflight #11386: Code documentation

David-OConnor, 2022-02-04:
### Is your feature request related to a problem? Please describe

This code base is very large, and is split across many sub-directories and files. The overall documentation comments appear (at first glance, to someone not familiar with many C code bases) to be quite sparse. For example, let's look at one of the barometer drives: https://github.com/betaflight/betaflight/blob/master/src/main/drivers/barometer/barometer_bmp085.c.

The comment at the top is this:
```
/*
 * This file is part of Cleanflight and Betaflight.
 *
 * Cleanflight and Betaflight are free software. You can redistribute
 * this software and/or modify this software under the terms of the
 * GNU General Public License as published by the Free Software
 * Foundation, either version 3 of the License, or (at your option)
 * any later version.
 ...
 ```

This is repeated on most files, and could be removed except for in the top-level license file (DRY). Instead, something like this might be used:

```
/*
* This module contains code related to using the BMP085 IC barometer.
*  It contains register definitions, and common functions for configuring the IC,
* and taking measurements. See [https://asf.com/BMP085_datasheet.pdf](The datasheet) for details
...
```

Then, over each reg write etc, document the section or table of the datasheet in a comment, and/or an excerpt.

In addition, I think a readme etc file in each directory that describes what the files in the directory do would help. I think this would be a relatively small undertaking for someone familiar with the code base, and would significantly approve its approachability.

### Describe the solution you'd like

See above

### Describe alternatives you've considered

See above

etracer65, 2022-02-04:
Betaflight is open source and all code is through contribution. So feel free to contribute additional documentation as you see fit.

etracer65, 2022-02-04:
And you might want to look at the exiting documentation. https://github.com/betaflight/betaflight/tree/master/docs

Finally, each source file must retain the license information details in the header as currently included. This information is intentionally added to every file so that if reused elsewhere the license and source details are preserved. So we wouldn't want to change this.

PR #11490 (2022-04-22): Added doc comment requirement to styleguide
Betaflight has a large and expanding codebase, and is a long-term open source project with many contributors. Short explanations of each module, data structure etc would improve long-term maintainability, and provide better structure and scope for files. It would also make the code base more approachable for new contributors, and existing contributors to work on sections they're new to. 

One way to approach this is to require new additions, and modifications to add this documentation. This requires no initial action, other than a change to the style [... cut]
closed 2022-03-14 by github-actions[bot]