betaflight/betaflight #11387: ~0.5s input delay on BF4.3 Rc3, ELRS 2.2 and Iflight f7 Twing when running dshot600/8k pid. Resolved by dshot300/4k pid setting

AllTheNamesWereTaken-EvenThisOne, 2022-02-05:
### Describe the bug

As title says, im getting massive input delay when running dshot600 and 8k pids on BF 4.3 Rc3. This is resolved when going down to dshot300, 4k pids. 

### To Reproduce

Use Dshot600, 8k pids
Use ELRS 2.2
On a Iflight f7 Twing FC

### Expected behavior

No input delay

### Flight controller configuration

```

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Feb  1 2022 / 20:14:54 (3267f0417) MSP API: 1.44
# config: manufacturer_id: IFRC, board_name: IFLIGHT_F722_TWING, version: c2b8c0d3, date: 2021-06-18T03:42:42Z

# start the command batch
batch start

board_name IFLIGHT_F722_TWING
manufacturer_id IFRC

# feature
feature GPS
feature TELEMETRY

# serial
serial 0 64 115200 57600 0 115200
serial 1 1 115200 57600 0 115200
serial 3 2 115200 57600 0 115200

# beeper
beeper -RX_LOST
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# aux
aux 0 0 0 1625 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 6 6 1300 1700 0 0
aux 4 27 5 1750 2100 0 0
aux 5 13 3 1300 1700 0 0
aux 6 31 7 1850 2100 0 0
aux 7 35 2 1600 2075 0 0

# master
set gyro_lpf1_static_hz = 500
set gyro_lpf2_static_hz = 1000
set gyro_to_use = FIRST
set dyn_notch_count = 2
set dyn_notch_q = 500
set dyn_notch_min_hz = 120
set gyro_lpf1_dyn_min_hz = 500
set gyro_lpf1_dyn_max_hz = 1000
set acc_calibration = 42,10,4,1
set serialrx_provider = CRSF
set blackbox_sample_rate = 1/2
set dshot_idle_value = 450
set dshot_bidir = ON
set motor_output_reordering = 3,2,1,0,4,5,6,7
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set simplified_gyro_filter_multiplier = 200
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_vbat_pos = 340
set osd_rssi_pos = 2544
set osd_link_quality_pos = 2405
set osd_rssi_dbm_pos = 2437
set osd_tim_1_pos = 2469
set osd_tim_2_pos = 2501
set osd_flymode_pos = 2071
set osd_current_pos = 2515
set osd_mah_drawn_pos = 2485
set osd_gps_speed_pos = 2510
set osd_gps_sats_pos = 2058
set osd_home_dir_pos = 2065
set osd_home_dist_pos = 2066
set osd_altitude_pos = 2060
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2540
set osd_battery_usage_pos = 264
set osd_disarmed_pos = 2533
set osd_flip_arrow_pos = 2287
set osd_efficiency_pos = 2454
set debug_mode = GYRO_SCALED
set gyro_1_sensor_align = CW180
set gyro_1_align_yaw = 1800
set gyro_2_sensor_align = CW270
set gyro_2_align_yaw = 2700
set rpm_filter_fade_range_hz = 100

profile 0

# profile 0
set dterm_lpf1_dyn_expo = 10
set vbat_sag_compensation = 100
set p_pitch = 59
set i_pitch = 106
set d_pitch = 58
set f_pitch = 158
set p_roll = 51
set i_roll = 92
set d_roll = 53
set f_roll = 138
set d_min_roll = 43
set d_min_pitch = 46
set feedforward_averaging = 2_POINT
set feedforward_jitter_factor = 12
set dyn_idle_min_rpm = 35
set dyn_idle_p_gain = 35
set simplified_pids_mode = RP
set simplified_d_gain = 145
set simplified_pi_gain = 115
set simplified_dmax_gain = 70
set simplified_feedforward_gain = 115
set simplified_pitch_d_gain = 95
set simplified_pitch_pi_gain = 110

rateprofile 0

# rateprofile 0
set roll_rc_rate = 20
set pitch_rc_rate = 20
set yaw_rc_rate = 20
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 74
set pitch_srate = 74
set yaw_srate = 74

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: CAMERA_CONTROL
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: GYRO_EXTI
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: GYRO_CS 1
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: PINIO 3
B09: FLASH_CS
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: GYRO_CS 2
C04: LED 1
C05: FREE
C06: MOTOR 3
C07: MOTOR 1
C08: MOTOR 4
C09: MOTOR 2
C10: SERIAL_TX 4
C11: SERIAL_RX 4
C12: FREE
C13: PINIO 1
C14: PINIO 2
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1 : CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

Iflight succex-d TwinG F7

### Other components

BetaFPV ELRS rx updated to 2.2

### How are the different components wired up

Correctly and super clean

AndroidGX, 2022-02-05:
Since we have the same issue, maybe we can regroup both in my ticket ? https://github.com/betaflight/betaflight/issues/11385
se everyone could update directly on it ? 👍

McGiverGim, 2022-02-08:
duplicated https://github.com/betaflight/betaflight/issues/11385

closed 2022-02-08 by McGiverGim