betaflight/betaflight #11389: Naked GoPro failed to turn on connected to LED pin with LED_STRIP changed to PINIO. BF 4.3-RC3

YuriDenison, 2022-02-06:
### Describe the bug

I resource `LED_STRIP` pin to `PINIO 1` and set up the `PINIO_BOX` so I can trigger gopro recording from the radio using `USER 1`.

I have similar setup on 3 quads with different FC
- GEPRC GEP-F722-45A AIO (geprc cinelog 35)
- IFlight Beast H7 55A AIO (custom build)
- MAMBA DJI F722 MK2 (diatone roma F5)

On BF 4.2.9 and 4.2.11 everything worked fine on all quads. GoPro turns on successfuly right after battery is connected.
Recording can be successfullly started/stopped on button click on the radio.

On BF 4.3-RC3:
- On MAMBA DJI F722 MK2 works fine. 
- On Geprc and iFlight Gopro doesn't turn on. The red led is on (power is ok).

Similar [issue](https://github.com/betaflight/betaflight/issues/10524) was reported a year ago, but was closed.
GoPro - [naked gerpc 8](https://geprc.com/wp-content/uploads/2021/05/geprc-gopro8-naked-User-Manual-Set-Guide.pdf)

### To Reproduce

Yellow wire from gopro should be soldered to `LED` pin on FC.

```
resource LED_STRIP 1 NONE
resource PINIO 1 A08
set pinio_box = 40,255,255,255
```

### Expected behavior

GoPro turns on successfuly right after battery is connected.
Recording can be successfullly started/stopped on button click on the radio.

### Flight controller configuration

```
 # 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Feb  1 2022 / 20:14:54 (3267f0417) MSP API: 1.44
# config: manufacturer_id: GEPR, board_name: GEPRC_F722_AIO, version: d3484d96, date: 2021-04-28T07:05:45Z

# start the command batch
batch start

board_name GEPRC_F722_AIO
manufacturer_id GEPR

# name: CineLog35

# resources
resource MOTOR 1 C07
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C08
resource LED_STRIP 1 NONE
resource PINIO 1 A01

# feature
feature TELEMETRY

# serial
serial 0 1 115200 57600 0 115200
serial 4 0 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1825 2100 0 0
aux 1 1 1 1300 1675 0 0
aux 2 13 2 1725 2100 0 0
aux 3 35 1 1725 2100 0 0
aux 4 40 3 1900 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 BAND_D   D FACTORY 5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set acc_calibration = 23,8,-41,1
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 1950
set rssi_channel = 9
set serialrx_provider = CRSF
set dshot_idle_value = 500
set dshot_bidir = ON
set align_board_roll = 180
set align_board_yaw = 0
set vbat_max_cell_voltage = 435
set vbat_min_cell_voltage = 340
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set osd_vbat_pos = 0
set osd_rssi_pos = 2452
set osd_link_quality_pos = 404
set osd_flymode_pos = 26
set osd_current_pos = 224
set osd_mah_drawn_pos = 2484
set osd_craft_name_pos = 14
set osd_warnings_pos = 2377
set osd_avg_cell_voltage_pos = 2516
set debug_mode = GYRO_SCALED
set pinio_box = 40,255,255,255
set gyro_1_sensor_align = CW180
set gyro_1_align_yaw = 1800
set name = CineLog35

profile 0

# profile 0
set feedforward_smooth_factor = 50
set feedforward_jitter_factor = 10

rateprofile 0

# end the command batch
batch end

```
```
  resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: CAMERA_CONTROL
A01: PINIO 1
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: GYRO_EXTI
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: GYRO_CS 1
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: FREE
B09: FLASH_CS
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: LED 1
C05: FREE
C06: MOTOR 2
C07: MOTOR 1
C08: MOTOR 4
C09: MOTOR 3
C10: SERIAL_TX 4
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1 : CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```

### Flight controller

GEP-F722-45A AIO

carbendi, 2022-02-06:
@YuriDenison  I was the one mentioned the problem about a year ago. And trying all the combinations I dont know what in the code is changed but I was able to make it work with inverting the pinio function. And changing the record button target orientation. 

If you like you can give it a try to these the way I am using with betafpv becs in your own risk; 

set pinio_config = 129,1,1,1
(this inverts the pinio function of the "40 in your set pinio_box = 40,255,255,255 ")
and
Change USER1 (or whichever user you are using for record swith) to yellow part at the right side not covering the middle. 1750 to 2100 should work.

This is how I have been using. 

Please let me know if you try it and if it works.

closed 2022-03-19 by github-actions[bot]

Marcel-Hartmann, 2024-03-31:
still open

Marcel-Hartmann, 2024-03-31:
Already open I can't use it with the NamelessRC Bec anyways using pinio_config 129,1,1,1 or 1,1,1,1 doesn't matter. I got 3,3 Volts if I switch my user1 which I got seated up like set pinio_box=40,255,255,255. All debugging works but BEC did not trigger the record start/stop functionality. Any Ideas?

carbendi, 2024-03-31:
@Marcel-Hartmann  there are 2 small pads they you have to short in nameless rc bec if you want o to use the remote rec option. 
hace you done that? 
[image]

Marcel-Hartmann, 2024-04-01:
Hi, thanks for answering. So I have tried your suggestion and got weird result. If I solder the two little PINs as you suggested, then the GOPRO9 gets on by pressing on after putting in the balancer cable into the GoPro. BUT if I add the LIPO, the GoPro went off. I desoldered the two PINs so that they are not connected, and the GoPro gets on and by adding a LIPO it stays on, but anyways it does not work for me.

Or should I use the AUX switched RX Cable and wire it to that little pad? Any suggestions will be appreciated.

Have a nice Easter

Marcel