betaflight/betaflight #11392: Unable to build STM32 UNIFIED when using BMI160

miguelwon, 2022-02-07:
### Describe the bug

I have a custom board that includes a BMI160 sensor. The board is built around a STM32F411CEU, for which reason I'm trying to [build](https://github.com/betaflight/betaflight/blob/master/docs/development/Building%20in%20Ubuntu.md) using STM32F411 target (from STM32_UNIFIED). The default [target.h](https://github.com/betaflight/betaflight/blob/3267f0417f2f597555d5a804648cbb3c114faf1f/src/main/target/STM32_UNIFIED/target.h#L240) file compiles with no issue, but has soon as I change 

`#define USE_ACCGYRO_BMI270`

to 

`#define USE_ACCGYRO_BMI160`

I get the following compilation error:

```
%% (optimised) ./src/main/drivers/usb_msc_f4xx.c 
%% (optimised) ./src/main/msc/usbd_storage.c 
./src/main/drivers/accgyro/accgyro_spi_bmi160.c: In function 'bmi160AccRead':
./src/main/drivers/accgyro/accgyro_spi_bmi160.c:278:40: warning: passing argument 2 of 'spiReadWriteBufRB' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  278 |     spiReadWriteBufRB(&acc->gyro->dev, bmi160_tx_buf, bmi160_rx_buf, BUFFER_SIZE);   // receive response
      |                                        ^~~~~~~~~~~~~
In file included from ./src/main/drivers/accgyro/accgyro_spi_bmi160.c:42:
./src/main/drivers/bus_spi.h:162:56: note: expected 'uint8_t *' {aka 'unsigned char *'} but argument is of type 'const uint8_t *' {aka 'const unsigned char *'}
  162 | bool spiReadWriteBufRB(const extDevice_t *dev, uint8_t *txData, uint8_t *rxData, int length);
      |                                               ~~~~~~~~~^~~~~~
./src/main/drivers/accgyro/accgyro_spi_bmi160.c: In function 'bmi160GyroRead':
./src/main/drivers/accgyro/accgyro_spi_bmi160.c:304:24: error: 'acc' undeclared (first use in this function)
  304 |     spiReadWriteBufRB(&acc->gyro->dev, bmi160_tx_buf, bmi160_rx_buf, BUFFER_SIZE);   // receive response
      |                        ^~~
./src/main/drivers/accgyro/accgyro_spi_bmi160.c:304:24: note: each undeclared identifier is reported only once for each function it appears in
./src/main/drivers/accgyro/accgyro_spi_bmi160.c:304:40: warning: passing argument 2 of 'spiReadWriteBufRB' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  304 |     spiReadWriteBufRB(&acc->gyro->dev, bmi160_tx_buf, bmi160_rx_buf, BUFFER_SIZE);   // receive response
      |                                        ^~~~~~~~~~~~~
In file included from ./src/main/drivers/accgyro/accgyro_spi_bmi160.c:42:
./src/main/drivers/bus_spi.h:162:56: note: expected 'uint8_t *' {aka 'unsigned char *'} but argument is of type 'const uint8_t *' {aka 'const unsigned char *'}
  162 | bool spiReadWriteBufRB(const extDevice_t *dev, uint8_t *txData, uint8_t *rxData, int length);
      |                                               ~~~~~~~~~^~~~~~
./src/main/drivers/accgyro/accgyro_spi_bmi160.c: In function 'bmi160SpiAccInit':
./src/main/drivers/accgyro/accgyro_spi_bmi160.c:324:21: error: 'accDev_t' {aka 'struct accDev_s'} has no member named 'dev'
  324 |     BMI160_Init(&acc->dev);
      |                     ^~
./src/main/drivers/accgyro/accgyro_spi_bmi160.c: In function 'bmi160SpiAccDetect':
./src/main/drivers/accgyro/accgyro_spi_bmi160.c:332:26: error: 'accDev_t' {aka 'struct accDev_s'} has no member named 'dev'
  332 |     if (bmi160Detect(&acc->dev) == MPU_NONE) {
      |                          ^~
%% (optimised) ./src/main/msc/usbd_storage_sd_spi.c 
%% (optimised) ./src/main/msc/usbd_storage_sdio.c 
Makefile:433: recipe for target 'obj/main/STM32F411/drivers/accgyro/accgyro_spi_bmi160.o' failed
make[1]: *** [obj/main/STM32F411/drivers/accgyro/accgyro_spi_bmi160.o] Error 1
make[1]: *** Waiting for unfinished jobs....
%% (optimised) ./src/main/msc/emfat.c 
%% (optimised) ./src/main/msc/usbd_storage_emfat.c 
%% (optimised) ./src/main/msc/emfat_file.c 
make[1]: Leaving directory '/home/mwon/betaflight'
Makefile:575: recipe for target 'hex' failed
make: *** [hex] Error 2
```

### To Reproduce

One Ubuntu, change line 240 of [target.h](https://github.com/betaflight/betaflight/blob/master/src/main/target/STM32_UNIFIED/target.h#L240) from

`#define USE_ACCGYRO_BMI270`

to 

`#define USE_ACCGYRO_BMI160`

and run:

`make TARGET=STM32F411`

### Expected behavior

The expected behavior is a new compiled custom STM32 Unified build.

### Flight controller configuration

I'm still trying to build the FC firmware.

### Flight controller

Custom

klutvott123, 2022-04-13:
@miguelwon Could you try this? https://github.com/klutvott123/betaflight/tree/fix-bmi160-driver
I don't have the hardware to test, but here's a STM32F411 binary with bmi160 driver included. bmi270 driver is removed to make it fit in flash.
[betaflight_4.3.0_STM32F411_bmi160.zip](https://github.com/betaflight/betaflight/files/8484321/betaflight_4.3.0_STM32F411_bmi160.zip)

miguelwon, 2022-04-14:
I couldn't make it work. Tried to compile from your branch and I got the following error:

```
miguelwon@Miguels-MacBook-Air betaflight_fix_bmi160 % make TARGET=UBIONE
/Library/Developer/CommandLineTools/usr/bin/make -j ./obj/betaflight_4.3.0_UBIONE_1c2fdcdc5.hex
%% (optimised) ./src/main/startup/system_stm32f4xx.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu.c 
%% (optimised) ./src/main/drivers/adc_stm32f4xx.c 
%% (optimised) ./src/main/drivers/bus_i2c_stm32f10x.c 
%% (optimised) ./src/main/drivers/bus_spi_stdperiph.c 
%% (optimised) ./src/main/drivers/dma_stm32f4xx.c 
%% (optimised) ./src/main/drivers/dshot_bitbang.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_decode.c 
%% (size optimised) ./src/main/drivers/inverter.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_stdperiph.c 
%% (size optimised) ./src/main/drivers/light_ws2811strip_stdperiph.c 
%% (size optimised) ./src/main/drivers/transponder_ir_io_stdperiph.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_shared.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot.c 
%% (optimised) ./src/main/drivers/serial_uart_stdperiph.c 
%% (optimised) ./src/main/drivers/serial_uart_stm32f4xx.c 
%% (optimised) ./src/main/drivers/system_stm32f4xx.c 
%% (optimised) ./src/main/drivers/timer_stm32f4xx.c 
%% (optimised) ./src/main/drivers/persistent.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_bmi160.c 
%% (optimised) ./src/main/build/build_config.c 
%% (optimised) ./src/main/drivers/barometer/barometer_dps310.c 
cc -mthumb -mcpu=cortex-m4 -march=armv7e-m -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Wdouble-promotion  -I./src/main -I./src/main/target -I./src/main/startup -I./lib/main/STM32F4/Drivers/STM32F4xx_StdPeriph_Driver/inc -I./lib/main/STM32_USB_OTG_Driver/inc -I./lib/main/STM32_USB_Device_Library/Core/inc -I./lib/main/STM32_USB_Device_Library/Class/cdc/inc -I./lib/main/STM32_USB_Device_Library/Class/hid/inc -I./lib/main/STM32_USB_Device_Library/Class/hid_cdc_wrapper/inc -I./lib/main/STM32_USB_Device_Library/Class/msc/inc -I./lib/main/STM32_USB-FS-Device_Driver/inc -I./lib/main/CMSIS/Core/Include -I./lib/main/STM32F4/Drivers/CMSIS/Device/ST/STM32F4xx -I./src/main/vcpf4 -I./lib/main/MAVLink -I./src/main/target/UBIONE -I./lib/main/CMSIS/DSP/Include -I./lib/main/google/olc  -std=gnu11 -Wall -Wextra -Wunsafe-loop-optimizations -Wdouble-promotion -ffunction-sections -fdata-sections -fno-common -pedantic  -DSTM32F411xE -DHSE_VALUE=8000000 -DARM_MATH_MATRIX_CHECK -DARM_MATH_ROUNDING -D__FPU_PRESENT=1 -DUNALIGNED_SUPPORT_DISABLE -DARM_MATH_CM4 -DTARGET_FLASH_SIZE=512 -DHSE_VALUE=8000000 -D_GNU_SOURCE -DUSE_STDPERIPH_DRIVER -DUBIONE  -DSTM32F4 -D'__FORKNAME__="betaflight"' -D'__TARGET__="UBIONE"' -D'__REVISION__="1c2fdcdc5"' -save-temps=obj -MMD -MP     ./src/main/drivers/accgyro/accgyro_spi_bmi160.c   -o drivers/accgyro/accgyro_spi_bmi160
%% (optimised) ./src/main/build/debug.c 
%% (optimised) ./src/main/build/debug_pin.c 
%% (optimised) ./src/main/target/UBIONE/target.c 
%% (size optimised) ./src/main/main.c 
%% (optimised) ./src/main/pg/adc.c 
%% (optimised) ./src/main/pg/beeper.c 
%% (optimised) ./src/main/pg/beeper_dev.c 
%% (optimised) ./src/main/pg/board.c 
%% (optimised) ./src/main/pg/bus_i2c.c 
%% (optimised) ./src/main/pg/bus_quadspi.c 
%% (optimised) ./src/main/pg/bus_spi.c 
%% (optimised) ./src/main/pg/dashboard.c 
%% (optimised) ./src/main/pg/displayport_profiles.c 
%% (optimised) ./src/main/pg/dyn_notch.c 
%% (optimised) ./src/main/pg/flash.c 
%% (optimised) ./src/main/pg/gyrodev.c 
%% (optimised) ./src/main/pg/max7456.c 
%% (optimised) ./src/main/pg/mco.c 
%% (optimised) ./src/main/pg/motor.c 
%% (optimised) ./src/main/pg/pg.c 
%% (optimised) ./src/main/pg/pin_pull_up_down.c 
%% (optimised) ./src/main/pg/piniobox.c 
%% (optimised) ./src/main/pg/rcdevice.c 
%% (optimised) ./src/main/pg/pinio.c 
%% (optimised) ./src/main/pg/rx.c 
%% (optimised) ./src/main/pg/rx_pwm.c 
%% (optimised) ./src/main/pg/rx_spi.c 
%% (optimised) ./src/main/pg/rx_spi_cc2500.c 
clang: error: the clang compiler does not support '-march=armv7e-m'
clang: warning: optimization flag '-fsingle-precision-constant' is not supported [-Wignored-optimization-argument]
clang: error: the clang compiler does not support '-march=armv7e-m'
clang: warning: optimization flag '-fsingle-precision-constant' is not supported [-Wignored-optimization-argument]
clang: error: the clang compiler does not support '-march=armv7e-m'
clang: warning: optimization flag '-fsingle-precision-constant' is not supported [-Wignored-optimization-argument]
clang: error: the clang compiler does not support '-march=armv7e-m'
%% (optimised) ./src/main/pg/rx_spi_expresslrs.c 
%% (optimised) ./src/main/pg/scheduler.c 
make[1]: *** [drivers/accgyro/accgyro_spi_bmi160] Error 1
make[1]: *** Waiting for unfinished jobs....
%% (optimised) ./src/main/pg/sdio.c 
%% (optimised) ./src/main/pg/sdcard.c 
make: *** [hex] Error 2
```

I have these definitions in `target.h`:

```
#define USE_SPI
#define USE_SPI_DEVICE_1

#define SPI1_SCK_PIN            PA5
#define SPI1_MISO_PIN           PA6
#define SPI1_MOSI_PIN           PA7

#define GYRO_1_CS_PIN          PA4
#define GYRO_1_SPI_INSTANCE    SPI1
#define GYRO_1_ALIGN      CW180_DEG

#define USE_ACCGYRO_BMI160
```

klutvott123, 2022-04-14:
I assume you have made a legacy target for your board. The unified STM32F411 with bmi160 driver included compiles just fine here in ubuntu on windows 10. Which clang version are you using? For Betaflight we're using version 10 I believe

miguelwon, 2022-04-14:
Ok, the unified STM32F411 with bmi160 compiled. Even so, I'm not able to make it work. Sorry. Is not working with INAV either so perhaps is related to my hardware.

SteveCEvans, 2022-05-15:
@miguelwon Please try https://github.com/betaflight/betaflight/pull/11549. This adds DMA support to the BMI160, but I've not yet been able to test as I don't have any FCs with the BMI160. I've tried the `target.h` changes you mention about and the `STM32F411` target builds fine.

closed 2022-05-25 by blckmn

miguelwon, 2022-06-25:
@SteveCEvans just come back to this issue. It compiles but when everything is connected and flashed, it does not detect the gyro.

SteveCEvans, 2022-06-25:
I could really do with having some hardware with this gyro. Would it be possible to get a sample?

miguelwon, 2022-06-25:
Ok, send me your address, please. I'll see what I can do.

SteveCEvans, 2022-06-29:
@miguelwon Please try https://github.com/betaflight/betaflight/pull/11707, with the patch below to use the BMI160 rather than the BMI270.

```
diff --git a/src/main/target/STM32_UNIFIED/target.h b/src/main/target/STM32_UNIFIED/target.h
index 71519c6..85fe88b 100644
--- a/src/main/target/STM32_UNIFIED/target.h
+++ b/src/main/target/STM32_UNIFIED/target.h
@@ -294,7 +294,7 @@
 #define USE_ACC_SPI_ICM20689
 #define USE_GYRO_SPI_ICM20689
 #define USE_ACCGYRO_LSM6DSO
-#define USE_ACCGYRO_BMI270
+#define USE_ACCGYRO_BMI160
 #define USE_GYRO_SPI_ICM42605
 #define USE_GYRO_SPI_ICM42688P
 #define USE_ACC_SPI_ICM42605
```