betaflight/betaflight #11396: Disarm/Fail safe HGLRC F722

Omega1978, 2022-02-09:
### Describe the bug

New Apex 5 build with Hglrc Zeus F722, 48 A ESC with Bluejay, Crossfire nano rx se, BF CR3.
After first flight on landing, warning message !FS! on mode tab and disarm in flight.
Second flight in air same message with disarm and crash.

### To Reproduce

second flight same issue

### Expected behavior

Locate the issue

### Flight controller configuration

```
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Feb  1 2022 / 20:14:54 (3267f0417) MSP API: 1.44
# config: manufacturer_id: HGLR, board_name: HGLRCF722, version: 836acaf9, date: 2021-10-31T19:48:38Z

# start the command batch
batch start

board_name HGLRCF722
manufacturer_id HGLR

# name: HGLRCF722

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL
feature MOTOR_STOP
feature TELEMETRY
feature OSD

# serial
serial 2 8192 115200 57600 0 115200
serial 3 1 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# map
map EATR1234

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 3 900 1300 0 0
aux 2 2 3 1300 1700 0 0
aux 3 13 2 1300 2100 0 0
aux 4 28 0 1700 2100 0 0

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = -15,10,-97,1
set rssi_channel = 12
set serialrx_provider = CRSF
set dshot_bidir = ON
set osd_vbat_pos = 54
set osd_rssi_pos = 2521
set osd_tim_1_pos = 470
set osd_flymode_pos = 2464
set osd_mah_drawn_pos = 2080
set osd_craft_name_pos = 2059
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2048
set osd_disarmed_pos = 2496
set debug_mode = GYRO_SCALED
set name = HGLRCF722

profile 0

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 30
set pitch_expo = 30
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70

# end the command batch
batch end

# 
```
```
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Flight controller

HGLRC ZEUS F722

Abulos, 2022-02-13:
Same issue using 4.3 RC2 - problem goes away reverting to 4.2.11
FC=Iflight SucceX-D F7 V2.1 TwinG
Receiver=TBS crossfire nano

On the bench I noticed when I turned on telemetry I started seeing throttle "dips" in the receiver tab
[image]

When I moved the sticks starting seeing dips on other channels as well:

[image]

Here is a diff that recreates the problem
[BTFL_cli_20220115_123220_diff.txt](https://github.com/betaflight/betaflight/files/8055036/BTFL_cli_20220115_123220_diff.txt)

ea-fpv, 2022-02-22:
#11428 this issue seems very similar to the one posted here. Just different flight controllers.

klutvott123, 2022-03-07:
Do you guys have CH2 connected to the FC? https://intofpv.com/t-psa-tbs-crossfire-tracer-rx-wiring-always-connect-the-rx-ch2-telemetry-wire

ea-fpv, 2022-03-13:
Hi, Glad to report back using this binary: [betaflight_4.3.0_STM32F7X2_norevision.zip](https://github.com/betaflight/betaflight/files/8233961/betaflight_4.3.0_STM32F7X2_norevision.zip)

Procedure followed: 
1.  Replicated the symptom as reported on  #11428
3. Saved backup from [Presets] tab on Configurator: 10.8.0 (c9b7ed8f)
4. Flashed local firmware with provided binary and restored saved backup from the [Presets] tab
5. Successfully :( replicated the issue withe the cli option crsf_use_negotiated_baud on default value (OFF)
[image]
6. Manually changed cli option crsf_use_negotiated_baud = ON
7. Power-cycled FC after SAVE
8. The issue seems to have been resolved (unable to replicate) when the cli option crsf_use_negotiated_baud = ON

Here is the diff in case needed:

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Mar 11 2022 / 18:06:40 (norevision) MSP API: 1.44
# config: manufacturer_id: DIAT, board_name: MAMBAF722, version: ec268d11, date: 2021-07-26T13:10:45Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAF722
manufacturer_id DIAT
mcu_id 0036003c3338510439393832
signature 

# name: QAV-S BF43

# feature
feature LED_STRIP

# serial
serial 0 0 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 5 1 115200 57600 0 115200

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# led
led 0 9,9::COW:11
led 1 10,10::COW:11
led 2 11,11::COW:11
led 3 12,12::COW:11

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 1300 1700 0 0
aux 2 13 2 1700 2100 0 0
aux 3 35 2 1700 2100 0 0

# adjrange
adjrange 0 0 1 900 1300 12 1 0 0
adjrange 1 0 1 1300 1700 12 1 0 0
adjrange 2 0 1 1700 2100 12 1 0 0

# rxrange
rxrange 0 993 2012

# master
set gyro_lpf1_static_hz = 225
[... 176 lines cut ...]
set tpa_breakpoint = 1450

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

Scheduling field tests for next weekend to report back.

NOTE: A noticeable lag was experienced with the option set to crsf_use_negotiated_baud = off (at least 500ms). The values seemed to remain for a while even after the setpoint changed not event worth trying to fly the quad with such a huge lag. Not sure if the default value should be OFF.

Please comment if the expected behaviour was to have this solved with the option ON or vice-versa.

Thank you

klutvott123, 2022-03-13:
@ea-fpv The behaviour you're seeing in the the screenshot of the receiver tab is exactly what it will look like when there's a baud rate mismatch between fc and rx. It then tries to negotiate the baud rate again and you can see that it succeeds eventually and the traces settle down. The strange thing is that with ```crsf_use_negotiated_baud = OFF``` the baud rate should remain at the default 420000 all the time. The rx will still try to negotiate, but the fc shouldn't accept any of the suggested baud rates and everything will remain at default. For this to work it is important to power cycle both FC and RX after changing it. What happens if the RX isn't power cycled is that it will just assume that the baud rate is the last negotiated baud rate and there will be a mismatch. The result is exactly what your receiver tab shows. Did you power cycle everything after flashing?

The reason for it working with ```crsf_use_negotiated_baud = ON``` could be that this issue can be intermittent. The baud rate is likely negotiated to be 2M which I suspect may be pushing it for some setups.

klutvott123, 2022-03-14:
@ea-fpv Thanks for testing. I reproduced the issues with slow updates. Made some changes. Have tested it with baud negotiation on/off telemetry on/off and have verified rx and tx baud rates with my scope. Here are updated binaries that should work properly

[betaflight_4.3.0_STM32G47X_norevision.zip](https://github.com/betaflight/betaflight/files/8241128/betaflight_4.3.0_STM32G47X_norevision.zip)
[betaflight_4.3.0_STM32H743_norevision.zip](https://github.com/betaflight/betaflight/files/8241129/betaflight_4.3.0_STM32H743_norevision.zip)
[betaflight_4.3.0_STM32F7X2_norevision.zip](https://github.com/betaflight/betaflight/files/8241130/betaflight_4.3.0_STM32F7X2_norevision.zip)
[betaflight_4.3.0_STM32F405_norevision.zip](https://github.com/betaflight/betaflight/files/8241131/betaflight_4.3.0_STM32F405_norevision.zip)
[betaflight_4.3.0_STM32F411_norevision.zip](https://github.com/betaflight/betaflight/files/8241132/betaflight_4.3.0_STM32F411_norevision.zip)
[betaflight_4.3.0_STM32F745_norevision.zip](https://github.com/betaflight/betaflight/files/8241133/betaflight_4.3.0_STM32F745_norevision.zip)

Seems like you guys are most easily able to reproduce the jumping channel values right after boot. What would cause this is rebooting the flight controller without power cycling. The RX still uses the previously negotiated baud rate while the FC has fallen back to default because of the reboot. It detects a baud rate mismatch by counting frame errors and this can take a few seconds, but it eventually syncs up again. I'm able to reproduce this and would consider it normal behaviour.

The failsafes in flight is another issue. It would be great if those of you who are able to reproduce the failsafes and dropping channel values a while after booting could test these binaries. Bench testing is fine if you don't feel like flying it. Look at the receiver tab for randomly dropping channel values or the flags in the status tab. 

```set crsf_use_negotiated_baud = OFF``` to force default crsf baudrate and set to ON to enable for comparison. Remember to power cycle FC/RX in between.

ea-fpv, 2022-03-14:
Thanks @klutvott123! Your build seems to be heading the right way.

Just tested [betaflight_4.3.0_STM32F7X2_norevision.zip](https://github.com/betaflight/betaflight/files/8241130/betaflight_4.3.0_STM32F7X2_norevision.zip) and I am glad to report the following:

1.) Unable to reproduce the symptoms with the default cli option crsf_use_negotiated_baud = OFF. This time I made sure to reboot all components (TX, RX, FC) before bench testing and I haven't been able to get the "sync dives/grooves" previously experienced. Also, no more "lag" felt like with the older build. Actually, the quad simulation on the configurator feels "smooth" and "responsive".
2.) After turning on the  crsf_use_negotiated_baud option I was able to reproduce the issue as shown below but this time it kept getting harder and harder to reproduce despite trying (cold)booting up the components in different order. Taking a look at the timestamps of the screenshots will give you an idea of how sparse the symptoms were experienced this time. Eventually got a very lengthy dive but the issue is no longer reproduced as easily as before.
[image]
[image]
[image]
[image]

I will be glad to give this build a try (with crsf_use_negotiated_baud = off) at the field if others have results similar to mine.

Thanks guys

klutvott123, 2022-03-14:
@ea-fpv Thanks a lot for testing! 

So with ```crsf_use_negotiated_baud = OFF``` there are no dropping channel values anymore? If so, it does indeed seem like this issue is caused by baud rate being too high as the baud rate used with the option off is the same as in BF 4.2.

The behaviour with baud rate negotiation ON is the same as the current master branch so that you're still seeing issues with it enabled is as expected.

ea-fpv, 2022-03-14:
Correct. Having the option set to OFF seems to fix this for me.

Abulos, 2022-03-14:
Hi @klutvott123 

Thanks for looking at this.

Yes, both Tx and Rx wires on my crossfire receiver are connected to FC.

Tested 4.3 RC2 (again) and 4.3 RC3 (new) and was able to reproduce throttle dips when I turned telemetry=ON with both firmwares, Note my throttle dips are not at regular intervals. TIme between dips varies.

Tried your latest betaflight_4.3.0_STM32F7X2_norevision.zip on the bench. I could not reproduce the throttle dips with this hex. Tried both crsf_use_negotiated_baud = OFF and crsf_use_negotiated_baud = ON. Could not get throttle dips with either option.
[20220314_klutvott hex trial diff.txt](https://github.com/betaflight/betaflight/files/8244553/20220314_klutvott.hex.trial.diff.txt)

Abulos, 2022-03-18:
[https://github.com/betaflight/betaflight/pull/11472](url)

Abulos, 2022-03-18:
#11472

closed 2022-03-22 by haslinghuis