betaflight/betaflight #11402: Brushed DC Motors Full Speed when armed on RC3

edlang90, 2022-02-14:
### Describe the bug

I've built a custom FC for a brushed whoop with ExpressLRS via SPI. It uses the Happymodel pinout (crazybeesx1280). Once I had everything set up (set motor output to BRUSHED, bind Rx, etc), I tried arming the quad and all the motors sped up to max speed. I tried resetting PIDs, using angle/acro/horizon and disabling airmode (props are on). The accelerometer is aligned properly.

What's weird is that when I use the motors tab I can control the full range of speed of the motors.

Is this a bad config or is it a bug with the code? I'm using 4.3.0 RC3 with the crazybeesx1280 target.

### To Reproduce

Uses custom hardware 

### Expected behavior

I expected the quad to fly normally since it's a regular "official" target with the motor output signal changed. 

### Flight controller configuration

```
  # diff

# version
# Betaflight / CRAZYBEEF4SX1280 (HAMO) 4.3.0 Dec 12 2021 / 15:21:14 (eb873308c) MSP API: 1.44

# start the command batch
batch start

board_name CRAZYBEEF4SX1280

# resources
resource BEEPER 1 NONE

# feature
feature -OSD
feature -AIRMODE
feature LED_STRIP

# serial
serial 0 8192 115200 57600 0 115200
serial 1 65536 115200 57600 0 115200

# led
led 0 8,8::C:8
led 1 7,8::C:8

# aux
aux 0 0 0 1500 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1700 2100 0 0
aux 3 15 2 1700 2100 0 0
aux 4 19 2 1300 1700 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 25 100 200 400
vtxtable powerlabels 25 100 200 350

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 0
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_calibration = -66,-122,173,1
set min_throttle = 1040
set current_meter = NONE
set vbat_scale = 20
set yaw_motors_reversed = ON
set simplified_gyro_filter = OFF
set simplified_gyro_filter_multiplier = 0
set vtx_band = 4
set vtx_channel = 8
set vtx_power = 1
set vtx_freq = 5880
set gyro_1_sensor_align = CW180
set gyro_1_align_yaw = 1800
set expresslrs_uid = 196,211,63,57,143,135
set expresslrs_rate_index = 2

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 0
set dterm_lpf1_dyn_max_hz = 0
set dterm_lpf1_static_hz = 0
set dterm_lpf2_static_hz = 0
set simplified_pids_mode = OFF
set simplified_dterm_filter = OFF
set simplified_dterm_filter_multiplier = 0

rateprofile 0

# end the command batch
batch end

```
```
  # resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: LED_STRIP
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI_EXPRESSLRS_RESET
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_EXPRESSLRS_BUSY
A14: PREINIT 2
A15: RX_SPI_CS
B00: ADC_BATT
B01: FREE
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: PREINIT 4
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: FREE
D02: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 1
TIM3: FREE
TIM4:
    CH1 : MOTOR 2
    CH2 : MOTOR 3
    CH3 : MOTOR 4
TIM5:
    CH1 : LED_STRIP
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: LED_STRIP
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

Custom board, 4.3.0 RC3

haslinghuis, 2022-02-17:
You have totally disable gyro and dterm filtering. Please check with default settings.

klutvott123, 2022-02-20:
@edlang can you try this please
[betaflight_4.3.0_CRAZYBEEF4SX1280_e4a780d9b.zip](https://github.com/betaflight/betaflight/files/8105209/betaflight_4.3.0_CRAZYBEEF4SX1280_e4a780d9b.zip)

edlang90, 2022-02-21:
Hey, thanks for your reply. I tried that build but I'm still getting a "no_gyro" error. Can't try the motors because it won't let me arm (works from the motors tab though). Here's my resource show all:

```

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: PREINIT 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI_EXPRESSLRS_RESET
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_EXPRESSLRS_BUSY
A14: PREINIT 2
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D02: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 1
TIM3: FREE
TIM4:
    CH1 : MOTOR 2
    CH2 : MOTOR 3
    CH3 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```

klutvott123, 2022-02-21:
The test firmware I uploaded could help with the full throttle issue only. Since you're using a different gyro you will need to add the necessary defines for that gyro and compile the firmware yourself

edlang90, 2022-02-21:
Oh I see. What did you change on this test firmware?

klutvott123, 2022-02-21:
This https://github.com/betaflight/betaflight/pull/11429

edlang90, 2022-02-27:
Hey, just tested out a custom build with your PR #11429 and it works great on brushed motors as well

haslinghuis, 2022-03-29:
Fixed in #11429

closed 2022-03-29 by haslinghuis