betaflight/betaflight #11405: Joystick mode

michael9544, 2022-02-15:
### Describe the bug

when set to usb_hid_cdc = on, it is no longer possible to connect to the configurator

### To Reproduce

set to usb_hid_cdc = on

### Expected behavior

when set to usb_hid_cdc = on, it is no longer possible to connect to the configurator

### Flight controller configuration

# version
# Betaflight / STM32F405 (S405) 4.3.0 Jan 31 2022 / 22:59:18 (3267f0417) MSP API: 1.44
# config: manufacturer_id: FLON, board_name: REVOLTOSD, version: 94d0da35, date: 2020-02-05T22:15:56Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name REVOLTOSD
manufacturer_id FLON
mcu_id 0047002d324d501920333256
signature 

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL

# serial
serial 2 64 115200 57600 0 115200

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 1177,-176,-409,1
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set pid_process_denom = 2

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration

### Flight controller

REVOLTOSD

haslinghuis, 2022-02-15:
In Options tab enable `show all serial devices`:

[image]

michael9544, 2022-02-15:
Thanks very much

closed 2022-02-15 by michael9544