betaflight/betaflight #11407: Support for AirSim simulator

nikhil-sethi, 2022-02-16:
### Is your feature request related to a problem? Please describe

This feature requests support for [AirSim](https://github.com/microsoft/AirSim): a powerful graphics simulator for drones, as a new simulation platform for the SITL target. 
I have a working implementation ready at my [fork](https://github.com/nikhil-sethi/betaflight/tree/airsim-sitl) and I can create a PR if the feature request is welcomed.
I have also created a corresponding [issue](https://github.com/microsoft/AirSim/issues/4351#issue-1137213999) and [PR](https://github.com/microsoft/AirSim/pull/4357#issue-1139759720) over at AirSim. I welcome any criticism from the maintainers and specific requirements that I should address.
Video for the implementation: https://youtu.be/sVEfow3y6rU

### Describe the solution you'd like

The implementation is built on top of the work done by @xuhao1 in this PR: #11303. Note that changes in files are not specific to AirSim and can just be considered an improvement in the SITL target. This is because like RealFlight, AirSim also requires raw PWM values. 
I have just added some directives which makes it easier to extend to other simulators instead of creating multiple threads and data types for each simulator (there will be only one simulator running at a time anyways). 
Note that this should be tested with Realflight which I don't have but maybe @xuhao1 can help with that.

### Describe alternatives you've considered

Currently the SITL target simulates with Gazebo through the Ardupilot plugin which is repetitive and unnecessary.  Gazebo can be clunky at times to extend and supporting AirSim can also provide better graphics for FPV training. 
Please let me know how I should proceed. Thank you for reading!

haslinghuis, 2022-02-17:
Yes please go - along. Please take in consideration we are in code freeze. It will take some time to merge the mentioned PR which your building upon.

nikhil-sethi, 2022-02-22:
Hello @haslinghuis. Sure, time is not much of an issue unless the folks at airsim want to wait for betaflight to merge the PR. I'll still open it and let people review in the meantime.

haslinghuis, 2023-02-20:
@nikhil-sethi we had some updates. Please update.

closed 2026-09-26 by blckmn