betaflight/betaflight #11408: missing target

deltawhiskey1147, 2022-02-17:
### Describe the bug

i cant find the correct target for a Foxeer F722 flight controller. the information provided with the flight controller said the target is called foxeerf722v2

### To Reproduce

open the target list and start scrolling

### Expected behavior

you wont find the target FOXEERF722V2

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```

### Flight controller

foxeerF722

haslinghuis, 2022-02-17:
Please update configurator to 10.8RC3

etracer65, 2022-02-17:
Looks fine to me.

[image]

deltawhiskey1147, 2022-02-18:
Thank you for your help. I’m not sure why my list isnt showing the correct target. I think it’s because I’m using a chrome book. Nothing but issues since day one

haslinghuis, 2022-02-19:
Configurator on Chrome is not supported anymore.