betaflight/betaflight #11409: Receiver glitch after first power cycle

itwillbefun, 2022-02-17:
### Describe the bug

4.3 flashed, set up receiver and check receiver tab. Everything is responding accurately. Unplug, replug back in, and then there is a massive lag/glitch response in the receivers tab. 

Running the Holybro KakuteF7 HDV with crossfire diversity.

### To Reproduce

flash, set up receiver port and receiver tab. Save. Power cycle. Glitch/lag.

### Expected behavior

to keep working

### Flight controller configuration

```
  # version
# Betaflight / STM32F745 (S745) 4.3.0 Oct 25 2021 / 03:43:24 (7495d7f96) MSP API: 1.44
# config: manufacturer_id: HBRO, board_name: KAKUTEF7HDV, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

board_name KAKUTEF7HDV
manufacturer_id HBRO

# serial
serial 3 64 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# master
set acc_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = CRSF
set gyro_1_align_yaw = 2700

profile 0

rateprofile 0

# end the command batch
batch end

```
```
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: LED 1
A03: FREE
A04: SDCARD_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: SDCARD_DETECT
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : MOTOR 3
    CH2 : MOTOR 4
TIM2: FREE
TIM3:
    CH3 : MOTOR 1
    CH4 : MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: MOTOR 4
DMA2 Stream 3: MOTOR 3
DMA2 Stream 4: ADC
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE```

### Flight controller

Holybro Kakute F7 HDV, Pyrodrone

### Other components

Crossfire diversity with latest firmware

### How are the different components wired up

5v, g, tx/rx 4

itwillbefun, 2022-03-20:
The jumps you see aren’t me. I’m holding the throttle up. And then I’m holding the roll. I had to reflash to get the video and I’m on the road for a job so I used speedybee, but this also does it in the desktop config.

https://user-images.githubusercontent.com/7536425/159148801-5ef724f0-1f74-4585-abd9-9b86798efbc0.MOV

klutvott123, 2022-03-20:
try this please https://github.com/betaflight/betaflight/pull/11472. do you have the telemetry wire connected?

itwillbefun, 2022-03-20:
seems to have worked!

haslinghuis, 2022-04-02:
Fixed in #11472

closed 2022-04-02 by haslinghuis