betaflight/betaflight #11411: ELRS UID Byte Generator tool built into Receiver tab

RitterFPV, 2022-02-17:
### Is your feature request related to a problem? Please describe

Sort of... 
Having a hell of a time getting the SPI receiver boards to bind properly.

Setting up Bind Phrase and just adding that CLI to each build eliminates the Bind Pulse Sync. 

### Describe the solution you'd like

Instead of or in addition to the Bind button on receiver tab when using ELRS. 
instead enter bind phrase and it auto inserts the UID in the CLI. 

then that is added to a "User Preset..." 

### Describe alternatives you've considered

The alternative is to point a link to the UID Byte Generator on https://www.expresslrs.org/2.0/hardware/spi-receivers/
then copy paste to cli... 

### Other information

[image]
[image]

KarateBrot, 2022-02-17:
See https://github.com/betaflight/betaflight-configurator/pull/2721 and https://github.com/betaflight/betaflight/pull/11192

RitterFPV, 2022-02-17:
Brilliant. 
I know I couldn't be the smartest guy in the room. 
also looks like I placed this in wrong side of the project. 
apologies for the noise

closed 2022-02-17 by RitterFPV