betaflight/betaflight #11413: Cannot arm quad when PWM or OneShot125 motor protocol is used with BF 4.3.0-RC3

galmiklos, 2022-02-17:
### Describe the bug

Quad would not arm, although there are no arming disable flags shown.

Motors run smoothly from motors tab in BF Configurator.

When arming switch is flipped on radio, ESCs power up and indicate presence of signal from FC, ARMING buzzer sequence can be heard, but ARMED buzzer sequence never comes, and the motors don't start spinning idle. There is no buzzer sequence suggesting arming disable flags either.

### To Reproduce

Configure motor protocol as PWM or OneShot125, when BF 4.3.0-RC3 is flashed, and try to arm quad.

### Expected behavior

When arming switch is flipped, motors should run idle.

### Flight controller configuration

```
# diff all

# version
# Betaflight / STM32F405 (S405) 4.3.0 Feb  1 2022 / 20:13:01 (3267f0417) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF405STD, version: 310c299f, date: 2020-05-22T01:08:06Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF405STD
manufacturer_id MTKS
mcu_id 003e00164d31500920313235
signature 

# feature
feature -SOFTSERIAL
feature -AIRMODE

# serial
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200

# map
map TAER1234

# aux
aux 0 0 0 1700 1950 0 0
aux 1 1 2 1050 1275 0 0
aux 2 2 2 1375 1625 0 0

# rxrange
rxrange 0 1159 1840
rxrange 1 1159 1840
rxrange 2 1159 1840
rxrange 3 1159 1840

# master
set acc_calibration = 31,-45,-26,1
set mag_hardware = AUTO
set mag_calibration = 0,-135,-147
set rssi_channel = 12
set serialrx_provider = SRXL
set blackbox_sample_rate = 1/16
set min_throttle = 1080
set use_unsynced_pwm = OFF
set motor_pwm_protocol = ONESHOT125
set motor_pwm_rate = 480
set small_angle = 180
[... 1210 lines cut ...]
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# end the command batch
batch end
```

### Flight controller

Matek F405 STD (new version with BMP280 baro) from getfpv.com

### Other components

Matek FCHUB-6S PDB
ESCs: HobbyKing 30A UBEC, flashed with BLHeli rev14.9 for Atmel
Motors: Turnigy aerodrive D2830/11 1000KV
Receiver: Happymodel Speki+ Micro 2.4GHz Receiver - DSM2/DSMX/SRXL

### Add any other context about the problem that you think might be relevant here

Issue was discussed on intofpv.com.

[https://intofpv.com/t-quad-not-armed-with-no-arming-status-flag](url)

Alveran81, 2022-02-20:
https://github.com/betaflight/betaflight/issues/11363

Nearly Same Problem.

klutvott123, 2022-02-20:
Thanks for reporting. Fixed in #11429

galmiklos, 2022-02-25:
I can see there is noone to test the fix with PWM, I can do it if it helps. Will I have to compile myself, or would there be a binary already compiled by chance?

klutvott123, 2022-02-25:
https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=6022&view=artifacts&pathAsName=false&type=publishedArtifacts Here are all hexes for all targets 👍

galmiklos, 2022-02-26:
I flashed betaflight_4.3.0_STM32F405_31011fa84.hex on my Matek F405 STD FC, with the SimonK ESCs connected with PWM, and the quad armed as expected--the motors started spinning idle.

I kept hearing the "RC Smoothing filter not initialized" beeper sequence after arming, though. I am not sure if it's related, but I didn't have that with BF4.2.11.

closed 2022-02-26 by haslinghuis

klutvott123, 2022-02-26:
@galmiklos Thanks for testing :thumbsup:

There's been a lot of changes in the rc smoothing department too. I just tested PWM and rc smoothing works correctly here so I'm not sure why it's not working for you

galmiklos, 2022-02-28:
What happened was that during the firmware update, rc_smoothing was turned OFF. I turned it back ON, and I don't get the RC_SMOOTHING_INIT_FAIL warning anymore.

It is a good question though, why do I get a warning about initialization, when the feature is not even on. But of course, if one wants to keep it off, the fix is easy, just disable it in the beeper configuration.

klutvott123, 2022-02-28:
Probably good to have the warning for safety reasons. Pids and feedforward won't work optimally without the proper smoothing

miguelwon, 2022-03-18:
Hi,
I'm having a similar problem that can be related. I'm working in a setup where I want to control the FC through MSP (through an RPI and a Bluetooth PS4 DualShock controller). 
Everything seems to work fine from the Receiver menu. I see the MSP commands are being received correctly and, from the Motor section, I read the correct PWM signal of 1000 in each motor when arming is off. Then, as soon I turn on the "I understand the risk" button and send arm signal, BF stops sending the 1000 PWM to the motors and generates some quite erratic signal of around 160. Then, if I start to increase throttle, I suddenly get PWM signals of about 1600 in motors 2 and 3.

I'm using BF v4.3.0 with a custom target that I show bellow

[image]

[image]

```
#define TARGET_BOARD_IDENTIFIER "UB1"
#define USBD_PRODUCT_STRING     "UBIONE"

// ******* LEDs and BEEPER ********

#define LED0_PIN                PC13

#define USE_BEEPER
#define BEEPER_PIN              PB14
#define BEEPER_INVERTED
#define BEEPER_PWM_HZ             2000 // Beeper PWM frequency in Hz

// ******* GYRO and ACC ********

//#define USE_SPI
//#define USE_SPI_DEVICE_1
//#define GYRO_1_SPI_INSTANCE    SPI1
//#define GYRO_1_CS_PIN          PA4
//#define GYRO_1_ALIGN      CW180_DEG

#define USE_SPI
#define USE_SPI_DEVICE_3

#define SPI3_SCK_PIN            PB3
#define SPI3_MISO_PIN           PB4
#define SPI3_MOSI_PIN           PB5

#define GYRO_1_SPI_INSTANCE    SPI3
#define GYRO_1_CS_PIN          PB12
#define GYRO_1_ALIGN      CW0_DEG

#define USE_EXTI
#define USE_GYRO_EXTI
#define GYRO_1_EXTI_PIN         PC14
#define USE_MPU_DATA_READY_SIGNAL
#define ENSURE_MPU_DATA_READY_IS_LOW

#define USE_GYRO
#define USE_GYRO_SPI_ICM20689
#define USE_ACC
#define USE_ACC_SPI_ICM20689

// ******* BARO **********

#define USE_I2C
#define USE_I2C_DEVICE_1
#define I2C_DEVICE              (I2CDEV_1)
#define I2C1_SCL                PB8        // SCL pad
#define I2C1_SDA                PB9        // SDA pad
#define BARO_I2C_INSTANCE       (I2CDEV_1)

#define USE_BARO
#define USE_BARO_DPS310

// ******* SERIAL ********

#define USE_VCP
#define USE_UART2
#define UART2_RX_PIN            PA3
#define UART2_TX_PIN            PA2

#define USE_SOFTSERIAL1
#define USE_SOFTSERIAL2

#define SERIAL_PORT_COUNT       4 //VCP, USART1, USART2, SOFTSERIAL x 2

// ******* ADC ********

#define USE_ADC
#define DEFAULT_VOLTAGE_METER_SOURCE VOLTAGE_METER_ADC
#define ADC_INSTANCE            ADC1  // Default added
#define ADC1_DMA_OPT            0     // DMA 2 Stream 0 Channel 0
#define VBAT_ADC_PIN            PB1

// ******* FEATURES ********

#define USE_RX_MSP
#define DYNBALANCE

//#define DEFAULT_VOLTAGE_METER_SOURCE    VOLTAGE_METER_ADC

#define TARGET_IO_PORTA         0xffff
#define TARGET_IO_PORTB         0xffff
#define TARGET_IO_PORTC         0xffff
#define TARGET_IO_PORTD         (BIT(2))

#define USABLE_TIMER_CHANNEL_COUNT 7
#define USED_TIMERS             ( TIM_N(1) | TIM_N(2) | TIM_N(3) | TIM_N(4) | TIM_N(10) | TIM_N(11) )

```

PR #11429 (2022-02-26): Fix non DSHOT protocols
Non DSHOT protocols are currently broken. At least multishot and oneshot.
Tested and confirmed working with all protocols except brushed and pwm, but that probably works too.

Fixes #11413, fixes #11363