betaflight/betaflight #11416: Configurator hangs after flashing / save & reboot

Hellfire82, 2022-02-18:
### Describe the bug

All of these bugs seem to be related:

1) After successfully flashing the FC via DFU, the FC reboots but the configurator does not display a COM port (only options are 'Manual Selection' & 'Virtual Mode').
* There is a constant single red light on the FC which could be misjudged as the FC being stuck in DFU mode, however I don't think that this is the case (I believe this is because the custom defaults have not been applied).
* I have to restart the configurator in order for a COM port to be displayed.
* After connecting, the option to apply custom defaults is presented & the lights on the FC then display as normal.

2) In addition when selecting save & reboot, the FC reboots but the configurator hangs.
* I have to restart the configurator in order to return functionality.

3) In addition, when typing 'save' or 'bl' into the CLI, the FC reboots but the configurator hangs.
* I have to restart the configurator in order to return functionality (the latter with the FC in DFU mode).

[image]

### To Reproduce

1) Flash FC to any version (currently using 4.3 RC1)
* No COM port available so restart configurator.
* After restart, connect & apply custom defaults.

2) Select save & reboot.
* Restart configurator to return functionality.

3) Type save or bl into the CLI.
* Restart configurator to return functionality.

### Expected behavior

1) After successfully flashing the FC, the COM port should be displayed, which should then allow me to connect straight away & apply custom defaults.

2) When selecting save & reboot, the configurator should reboot & return to the previous tab that I was using.

3) When typing save or bl into the CLI, the configurator should reboot & return me to either the previous tab that I was using, or in respect to the latter, the home page.

### Setup

- Configurator version: 10.9.0-debug-90f15faf
- Windows 8.1
- Aikon F7 Mini v 1.1

* The ImpulseRC driver fixer has been ran on multiple occasions successfully.
* I installed configurator version 10.7.1 & all of the above issues were NOT present.

ctzsnooze, 2022-04-15:
@Hellfire82 please advise if this issue is fixed in current master and latest Configurator.

Hellfire82, 2022-04-15:
@ctzsnooze, @haslinghuis I am currently using configurator RC4 & I am still experiencing the same issues that I last mentioned on February 14th.
After my last comment, I was unable to find this thread for some reason (like it had been closed), so I couldn't provide any more feedback.

haslinghuis, 2022-04-15:
Confirm issue is intermittent and still present. Still looks as USB driver or firmware issue.

Hellfire82, 2022-04-15:
@haslinghuis I am now using BF configurator RC5 & can confirm that all of the aforementioned issues are still present. There is nothing intermittent about this as all of the issues happen every time.
Correct me if I'm wrong, but if it was a USB driver issue, then I would have the same issues with BF configurator 10.7.2 (which worked perfectly when it was installed).
One thing I noticed with RC5 is that I am no longer able to display the console by pressing F12?

haslinghuis, 2022-04-15:
Yes RC is a production build, only nightlies have the console.

Looking at related issues here:

- https://github.com/betaflight/betaflight/issues/11506
- https://github.com/betaflight/betaflight/issues/11416
- https://github.com/betaflight/betaflight/issues/11157

haslinghuis, 2022-04-28:
Upon further testing my SpeedyBee F7 V1 Mini stack does have the same issue using default configuration, but intermittent. 

Using TBS 69 here without LIPO attached so it's not powered / interfering with the reboot.

With 4,2.0 firmware and a very old configurator build from 31-12-2020 had to reboot 15 times to make it fail, on 4.3 firmware it fails a little more.

HowdyImSWIM, 2022-05-06:
This is my first ever comment on Github so I apologize in advance in the case that any of the following isn't how things work on here. 

That said, I am having the **exact** same issue on RC6 as Hellfire, and issue did not exist on any earlier versions/releases. I encountered these issues of "hanging", double clicking connect, plugging in before opening BF Config to be able to connect etc etc same exact issues to a T from very first time trying to use 4.3 which was with RC4 [I believe?] and if I'm right about RC4 being first then the issues have persisted/carried over with **NO CHANGES** for better or worse through my jump to 10.8 RC6 /BF 4.38. 

Again the details of issue are identical in literally every single way besides my fc and quad;
IFlight Nazgul 5 V2 4s 
# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Apr 25 2022 / 01:06:24 (9d71184) MSP API: 1.44
# config: manufacturer_id: IFRC, board_name: IFLIGHT_SUCCEX_E_F7, version: c2b8c0d3, date: 2021-06-18T03:42:42Z

If you would like more info i'll do my best to provide all that may be requested and at my earliest convenience; fair warning very busy this week so yeah. Hope I didn't mess this open issue comment thread up in anyway again long time GitHub user but first time commenter. Cheers.

haslinghuis, 2022-05-07:
On STM32F7X2 bootloader uses `USART1_RX` (PA10) or `USART1_TX`( PA9), and `USART3_RX` (PB11) or `USART3_TX` (PB10) which should be avoided for peripherals powered using 5V to not interfere with the (re)boot process.

Sparky5150, 2022-05-19:
I had the same problem and I just fixed it

haslinghuis, 2022-05-19:
Tell us more ;-)

Sparky5150, 2022-05-19:
Ok...plugged in my quad and then opened betaflight.  Then opened Zadig 2.7 and selected options-list all devices.  The stm32 was listed there so I selected it (all other usb devices were unplugged).  When I went back to betaflight it still didn't work so I ran irc driver fixer.  When complete it said reboot my computer and so I did.  Can't remember if I plugged the quad in first or opened betaflight first but it worked.

Sparky5150, 2022-05-19:
Does any of that make sense???

haslinghuis, 2022-05-19:
Sure most of the time we don't need `Zadig` only `IRC driver fix` but if it worked for you it makes sense.

haslinghuis, 2022-07-26:
@Hellfire82 

Please check firmware https://github.com/betaflight/betaflight/pull/11764 with configurator: https://github.com/betaflight/betaflight-configurator/pull/2983

Hellfire82, 2022-07-28:
@haslinghuis Using the aforementioned configurator:

1) The FC has to be plugged in via USB before the configurator is loaded, or else the COM port doesn't display.
2) Save & Reboot takes you back to the welcome screen, but doesn't go back to the previously used tab.
3) Saving in CLI leaves you in the CLI tab. You have to press disconnect then connect to get back to the previously used tab.

These are the complete console outputs for the 2 actions that I performed:

Save & Reboot:
[image]

Save in CLI:
[image]

I also previously attempted the solution provided by @Sparky5150 but unfortunately I saw no change.

haslinghuis, 2022-07-28:
@Hellfire82 you also need to flash the firmware provided. Use `detect` button before `load firmware local`

Hellfire82, 2022-07-31:
@haslinghuis I've installed the latest nightly configurator & flashed the latest development build:

[image]

Save & reboot now doesn't return to the welcome screen. Disconnect has to be pressed twice, followed by connect to return to the last used tab. Below is the console output for this action:

[image]
[image]
[image]

haslinghuis, 2022-07-31:
What happens when you execute this line in devtools?
```
        MSP.send_message(MSPCodes.MSP_SET_REBOOT, false, false, reinitializeConnection);
```

Hellfire82, 2022-08-05:
> What happens when you execute this line in devtools?
> 
> ```
>         MSP.send_message(MSPCodes.MSP_SET_REBOOT, false, false, reinitializeConnection);
> ```

When connected and the above line is executed in devtools, the FC reboots but the configurator stays on the last tab that I was using. It takes 2 clicks on disconnect followed by one on connect to get back to the last used tab.

[image]

haslinghuis, 2022-08-05:
The `reboot request accepted` message let's us know the reboot request has been sent to the FC.

My conclusion is that the reset is not happening or USB is struggling to release the port.

It should show something like
[image]

closed 2023-12-17 by haslinghuis