betaflight/betaflight #11417: USB device descriptors missing

howels, 2022-02-18:
### Describe the bug

> 
RC3 with this fix seems good - latest master has developed some issues connecting to USB where device-descriptors aren't being sent but that doesn't seem related to this issue.
All 4 motors report RPM data and they seem to do so reliably across 10+ FC boots! Huge thanks to you @hydra for your expertise here. Do we incur a noticeable performance penalty disabling the data cache?

Here's the lack of USB descriptors after flashing latest master.

```
[44539.525801] usb 7-2.4.1: new full-speed USB device number 41 using xhci_hcd
[44552.881729] usb 7-2.4.1: device descriptor read/64, error -32
[44561.273671] usb 7-2.4.3: new full-speed USB device number 42 using xhci_hcd
[44576.701571] usb 7-2.4.3: device descriptor read/64, error -110
[44592.305468] usb 7-2.4.3: device descriptor read/64, error -110
[44592.497459] usb 7-2.4.3: new full-speed USB device number 43 using xhci_hcd
[44607.925381] usb 7-2.4.3: device descriptor read/64, error -110
[44623.545267] usb 7-2.4.3: device descriptor read/64, error -110
[44623.657333] usb 7-2.4-port3: attempt power cycle
[44624.265252] usb 7-2.4.3: new full-speed USB device number 44 using xhci_hcd
[44629.345260] xhci_hcd 0000:0b:00.3: Timeout while waiting for setup device command
[44634.721227] xhci_hcd 0000:0b:00.3: Timeout while waiting for setup device command
[44634.929191] usb 7-2.4.3: device not accepting address 44, error -62
[44635.009192] usb 7-2.4.3: new full-speed USB device number 45 using xhci_hcd
[44640.101175] xhci_hcd 0000:0b:00.3: Timeout while waiting for setup device command
[44645.473131] xhci_hcd 0000:0b:00.3: Timeout while waiting for setup device command
[44645.681113] usb 7-2.4.3: device not accepting address 45, error -62
[44645.681258] usb 7-2.4-port3: unable to enumerate USB device

```

Normal connections look like:

```
[44484.734161] usb 7-2.4.3: new full-speed USB device number 40 using xhci_hcd
[44484.854157] usb 7-2.4.3: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[44484.854168] usb 7-2.4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[44484.854172] usb 7-2.4.3: Product: IFLIGHT_H743_AIO_V2
[44484.854174] usb 7-2.4.3: Manufacturer: Betaflight
[44484.854176] usb 7-2.4.3: SerialNumber: 387E347B3330
[44484.891202] cdc_acm 7-2.4.3:1.0: ttyACM0: USB ACM device
```

### To Reproduce

Any build from master on the date of this issue - attempts to connect usb after flashing results in failure to detect usb device.

### Expected behavior

USB device descriptors sent by the FC.

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```

### Flight controller

H743 test build - iFlight H743 AIO v2

howels, 2022-02-18:
As discovered in https://github.com/betaflight/betaflight/issues/11376#issuecomment-1044879120

ctzsnooze, 2022-04-15:
@howels Please re-test with master.
If fixed, please close this issue.

howels, 2022-04-15:
issue was closed with RC4

closed 2022-04-15 by howels