betaflight/betaflight #11426: SPEEDYBEE F7 V2 Betaflight 4.3 Crossfire Issue

N3dFlow, 2022-02-20:
### Describe the bug

Low signal from Crossfire using Betaflight 4.3 RC 3 for SpeedyBee F7 V2 Stack.

### To Reproduce

[image]
 

### Expected behavior

I flashed back 4,2 and Crossfire work perfectly

### Flight controller configuration

```
  # version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Feb  1 2022 / 20:14:54 (3267f0417) MSP API: 1.44
# config: manufacturer_id: SPBE, board_name: SPEEDYBEEF7V2, version: 55694498, date: 2021-05-15T13:56:51Z

# start the command batch
batch start

board_name SPEEDYBEEF7V2
manufacturer_id SPBE

# name: Air5

# feature
feature -LED_STRIP
feature -ESC_SENSOR

# serial
serial 1 1 115200 57600 0 115200

# beacon
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1925 2100 0 0
aux 1 1 2 1350 1650 0 0
aux 2 13 1 1800 2100 0 0
aux 3 35 2 1850 2100 0 0

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 0
set dyn_notch_count = 1
set dyn_notch_q = 500
set dyn_notch_max_hz = 500
set gyro_lpf1_dyn_min_hz = 0
set acc_calibration = 32,15,-112,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 2000
set fpv_mix_degrees = 25
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set beeper_dshot_beacon_tone = 2
set small_angle = 180
set simplified_gyro_filter = OFF
set osd_vbat_pos = 14828
set osd_rssi_pos = 14349
set osd_link_quality_pos = 14342
set osd_flymode_pos = 14362
set osd_craft_name_pos = 14336
set osd_display_name_pos = 0
set osd_flight_dist_pos = 14353
set osd_compass_bar_pos = 14474
set osd_warnings_pos = 14696
set osd_avg_cell_voltage_pos = 14821
set osd_disarmed_pos = 14602
set osd_flip_arrow_pos = 14734
set debug_mode = GYRO_SCALED
set rpm_filter_harmonics = 2
set name = Air5

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 80
set dterm_lpf1_dyn_max_hz = 110
set dterm_lpf1_dyn_expo = 8
set dterm_lpf1_type = BIQUAD
set dterm_lpf2_static_hz = 0
set yaw_lowpass_hz = 0
set simplified_dterm_filter = OFF

rateprofile 0

# rateprofile 0
set roll_rc_rate = 3
set pitch_rc_rate = 2
set yaw_rc_rate = 2
set roll_expo = 58
set pitch_expo = 58
set yaw_expo = 58
set roll_srate = 107
set pitch_srate = 104
set yaw_srate = 56

# end the command batch
batch end
```
```
  Currently active IO resource assignments:
(reboot to update)
--------------------
A00: CAMERA_CONTROL
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 1
A09: MOTOR 2
A10: MOTOR 3
A11: USB
A12: USB
A13: SWD
A14: FLASH_CS
A15: PINIO 1
B00: FREE
B01: FREE
B02: BEEPER
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: SERIAL_TX 1
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: MOTOR 4
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1 : CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 1
    CH4 : DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 3
```

### Flight controller

SPEEDYBEE F7 V2

XsmStark, 2022-03-01:
We have tried to flash theSPEEDYBEE F7 V2 to version 4.3RC3 for many times, but failed to find the problem you described during the process. We recommend you to double check your port settings or reset your SPEEDYBEE F7 V2.

haslinghuis, 2022-04-01:
Betaflight firmware 4,3 is using CRSF v3. TBS has confirmed there are some issues.
We did not have a problem with 3 receivers on v6.07, The issue is addressed in TBS firmware V6,17

closed 2022-04-01 by haslinghuis