betaflight/betaflight #11427: No custom defaults for IFLIGHT_H743_AIO_V2

senbax, 2022-02-20:
### Describe the bug

It seems like there are no custom defaults for IFLIGHT_H743_AIO_V2 on Betaflight 4.3.*
For example Gyro alignment and LEDs are not working.

### To Reproduce

Flash Betaflight 4.3.* onto IFLIGHT_H743_AIO_V2
Click "Connect"
-> button "Apply custom defaults" does not appear

### Expected behavior

Flash Betaflight 4.3.* onto IFLIGHT_H743_AIO_V2
Click "Connect"
-> button "Apply custom defaults" appears and custom defaults get applied

### Flight controller configuration

```
# version
# Betaflight / IFLIGHT_H743_AIO_V2 (IF7A) 4.3.0 Feb 13 2022 / 23:26:46 (7b4415f06) MSP API: 1.44

# start the command batch
batch start

board_name IFLIGHT_H743_AIO_V2

# master
set gyro_to_use = FIRST

profile 0

rateprofile 0

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: FREE
B01: FREE
B02: FREE
B03: CAMERA_CONTROL
B04: FREE
B05: FREE
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: PREINIT 2
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_EXT
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: ADC_RSSI
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: LED 1
C14: FREE
C15: FREE
D00: GYRO_EXTI
D01: FREE
D02: BEEPER
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: SPI_SCK 4
E03: FREE
E04: OSD_CS
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : LED_STRIP
TIM2:
    CH2 : CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MOSI 1
DMA1 Stream 1: SPI_MISO 1
DMA1 Stream 2: SPI_MOSI 2
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 3
DMA1 Stream 5: SPI_MISO 3
DMA1 Stream 6: SPI_MOSI 4
DMA1 Stream 7: SPI_MISO 4
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: LED_STRIP
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

IFLIGHT_H743_AIO_V2

McGiverGim, 2022-02-21:
The unified target file must be created by the manufacture. Usually for IFlight is @Linjieqiang who does that.

We can't create it because we don't know the internals of the FC.

@Linjieqiang is a unified target for that on the go?

Linjieqiang, 2022-02-21:
Okay, I will take the time to complete it. @McGiverGim

Linjieqiang, 2022-02-22:
@senbax Hello, fixed on this: https://github.com/betaflight/unified-targets/pull/566

Linjieqiang, 2022-02-24:
@senbax It's time to close this issue.

closed 2022-02-24 by haslinghuis