betaflight/betaflight #11436: usb_hid_cdc not functioning properly

GabeOch, 2022-02-24:
### Describe the bug

If I enter into the CLI:  set usb_hid_cdc = ON, it will make the computer recognize it as a game controller, but with Betaflight 4.2 and now 4.3, one channel will not work at all.  In the sim I use, Flowstate, the yaw channel does not appear.  In an RC airplane Sim I also use, the throttle channel does not appear.  The command functions normally on Betaflight 4.1.  Additionally, when this mode is on, Betaflight Configurator 10.8.0 does not show the flight controller in the drop-down list of flight controllers connected after I enter the command, while Configurator 10.7.2 does.
Note:  This is NOT a problem when flying in real life.  All channels work as expected when flying.  This is only a problem in a sim.

### To Reproduce

Enter, "set usb_hid_cdc = ON" in the CLI, save, open sim (mine, as mentioned, is Flowstate).  Go to Settings>Controller.  Yaw channel does not work.
Open Betaflight Configurator 10.8.0.  Drop-down list of connected flight controllers is blank except for "Manual Selection"

### Expected behavior

The yaw channel should work when I move the sticks in the sim.
The drop-down list in the configurator should show my flight controller.

### Flight controller configuration

```
# diff

# version
# Betaflight / STM32F411 (S411) 4.3.0 Feb  1 2022 / 20:13:25 (3267f0417) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF411RX, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

board_name MATEKF411RX
manufacturer_id MTKS

# name: TinyHawk II

# feature
feature -RX_SPI
feature RX_SERIAL

# serial
serial 0 64 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1075 1625 0 0
aux 1 1 2 1725 1925 0 0
aux 2 1 1 950 1350 0 0
aux 3 13 2 1400 1600 0 0
aux 4 19 0 1300 1700 0 0
aux 5 30 2 1300 1700 0 0
aux 6 35 1 950 1350 1 0
aux 7 35 2 1300 1700 1 0

# adjrange
adjrange 0 0 1 1125 1200 29 2 0 0

# rxrange
rxrange 0 1841 1159
rxrange 1 1159 1841
rxrange 2 1841 1159
rxrange 3 1159 1841

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 100 200

# rxfail
rxfail 6 s 1800
rxfail 8 s 1000

# master
set dyn_notch_count = 4
set dyn_notch_q = 299
set acc_calibration = -74,-44,60,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1159
set max_check = 1841
set rssi_channel = 9
set rssi_scale = 120
set rssi_offset = -20
set fpv_mix_degrees = 40
set serialrx_provider = SPEK2048
set rx_min_usec = 750
set rx_max_usec = 2250
set rx_spi_protocol = FRSKY_D
set min_throttle = 1080
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set failsafe_delay = 20
set failsafe_off_delay = 30
set failsafe_throttle = 1250
set bat_capacity = 450
set vbat_max_cell_voltage = 435
set vbat_warning_cell_voltage = 345
set current_meter = NONE
set vbat_duration_for_warning = 40
set vbat_duration_for_critical = 15
set ibatv_scale = 11
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 20
set yaw_deadband = 10
set osd_units = IMPERIAL
set osd_warn_batt_not_full = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_rssi = ON
set osd_rssi_alarm = 76
set osd_cap_alarm = 450
set osd_tim1 = 0
set osd_vbat_pos = 364
set osd_rssi_pos = 2137
set osd_tim_2_pos = 14710
set osd_flymode_pos = 14402
set osd_throttle_pos = 14456
set osd_vtx_channel_pos = 353
set osd_crosshairs_pos = 8397
set osd_ah_sbar_pos = 8398
set osd_ah_pos = 8270
set osd_craft_name_pos = 14378
set osd_display_name_pos = 394
set osd_compass_bar_pos = 8426
set osd_pid_roll_pos = 353
set osd_pid_pitch_pos = 34
set osd_power_pos = 354
set osd_pidrate_profile_pos = 186
set osd_avg_cell_voltage_pos = 14700
set osd_flip_arrow_pos = 310
set osd_core_temp_pos = 14690
set osd_stick_overlay_left_pos = 4259
set osd_stick_overlay_right_pos = 4276
set osd_rate_profile_name_pos = 23
set osd_pid_profile_name_pos = 2
set osd_profile_name_pos = 354
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_menu_background = GRAY
set vtx_band = 1
set vtx_channel = 7
set vtx_power = 1
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5745
set vcd_video_system = NTSC
set frsky_spi_tx_id = 148,242,0
set frsky_spi_offset = -116
set frsky_spi_bind_hop_data = 106,161,216,15,70,125,180,235,34,89,144,199,254,53,108,163,218,17,72,127,182,237,36,91,146,201,0,55,110,165,220,19,74,129,184,239,38,93,148,203,2,57,112,167,222,21,76,0,0,0
set gyro_1_align_yaw = 1800
set spektrum_spi_num_channels = 12
set name = TinyHawk II

profile 0

# profile 0
set anti_gravity_gain = 10000
set crash_recovery = ON
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10
set iterm_windup = 40
set iterm_limit = 150
set acro_trainer_angle_limit = 30
set p_pitch = 121
set i_pitch = 144
set d_pitch = 109
set f_pitch = 115
set p_roll = 121
set i_roll = 144
set d_roll = 109
set f_roll = 115
set p_yaw = 95
set d_yaw = 45
set f_yaw = 90
set angle_level_strength = 60
set horizon_transition = 80
set level_limit = 50
set horizon_tilt_effect = 80
set horizon_tilt_expert_mode = ON
set abs_control_gain = 20
set d_min_roll = 0
set d_min_pitch = 0
set launch_control_mode = PITCHONLY
set thrust_linear = 50
set dyn_idle_min_rpm = 35
set simplified_pids_mode = OFF
set simplified_master_multiplier = 120

rateprofile 1

# rateprofile 1
set thr_mid = 25
set thr_expo = 50
set rates_type = BETAFLIGHT
set roll_rc_rate = 90
set pitch_rc_rate = 90
set yaw_rc_rate = 90
set roll_srate = 83
set pitch_srate = 83
set yaw_srate = 83
set tpa_rate = 25
set tpa_breakpoint = 1450

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: ADC_BATT
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: FREE
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : DSHOT_BITBANG 2
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

It is a MATEKF411RX flight controller that came installed on the Emax Tinyhawk 2 I purchased used off Ebay.

### Other components

I installed a Spektrum Satellite receiver and it is fully funtional.  This same problem existed before I installed that and while I was still using the Emax e6 controller, however.

### How are the different components wired up

The Spektrum Satellite is hooked up to the RX1 port on the flight controller.

howels, 2022-02-27:
Are you sure that the "missing" channel did not get mapped as an analogue trigger? I've have seen issues where 6 analogue axis are mapped; LX,LY,LZ,RX,RY,RZ - the X and Y axis are normal joystick axis but the Z channel is assumed to be a trigger instead of the next joystick axis. Previously I created a model with channel 3 unmapped and the last two axis on channels 4 & 5 to avoid mapping onto the trigger axis.

GabeOch, 2022-03-01:
I tried it on the preview of FPV Freerider, and it actually worked properly this time.  I think that since Betaflight 4.1, the channel mapping for the game controller mode changed.  Flowstate, the sim I used before, only supports using 6 channels, and one of the channels missing was the Yaw channel, as mentioned.  It would be nice if the channel mapping was changed back to the original channel mapping, but I suppose it's not a big deal.  I would still like the bug fixed where Betaflight Configurator 10.8.0 doesn't detect the flight controller after usb_cdc_hid is enabled, however.

haslinghuis, 2022-03-01:
@GabeOch in options tab of configurator try enabling 

[image]

GabeOch, 2022-03-01:
Thanks!  That worked.