betaflight/betaflight #11438: ACC Trim via OSD has been broken somewhere during the implementation of the BF 4.3

Martivip, 2022-02-27:
### Describe the bug

BF has possibility to Trim ACC via OSD stick commands, this has been broken somewhere during the BF 4.3 implementation.
Right now with BF4.3 RC3 and STM32F745 (KakuteF7) target when I try to trim 
ACC Pitch or Roll this change is not effective and is not saved in FC, configurator displays "zero" values on both axis 
[image]
Also the Visual Beeper and Beep sound during the applying changes is not always present and with Timing AAC Forward is not present at all.
With target BETAFPVF4SX1280 (ELRS SPI receiver on board) due to changes in this PR https://github.com/betaflight/betaflight/pull/11380 the Trim Commands via OSD are effective due to huge @SteveCEvans contribution (setting are stored at Throttle Low now). But but it needs few improvements, user need to have control of how many increments he doing for pitch or roll trims, I have enabled visual beeper ***** and I don't see this visual beeper in my OSD during the changes, it should blink once per 1 step on pitch or roll increments and should blink on 1s (or maybe 0,5s) intervals this allow the user to have control during the changes.
Generally as I've mention this functionality has been present till BF 4.2.9 has been lost during the BF 4.3 Implementation 

### To Reproduce

Try to adjust, Trim ACC via OSD stick commands

### Expected behavior

ACC trim commands should work as it has been implemented in BF 4.2.9

### Flight controller configuration

**F745 Target**
```
# 

# diff all

# version
# Betaflight / STM32F745 (S745) 4.3.0 Feb  1 2022 / 20:14:27 (3267f0417) MSP API: 1.44
# config: manufacturer_id: HBRO, board_name: KAKUTEF7, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name KAKUTEF7
manufacturer_id HBRO
mcu_id 003700263538510a38333932
signature 

# resources
resource MOTOR 5 NONE
resource CAMERA_CONTROL 1 B03
resource PINIO 1 C09

# timer
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)

# dma
dma SPI_MISO 1 1
# SPI_MISO 1: DMA2 Stream 2 Channel 3
dma SPI_RX 1 1
# SPI_RX 1: DMA2 Stream 2 Channel 3
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature LED_STRIP
feature ESC_SENSOR

# serial
serial 0 2048 115200 57600 0 115200
serial 1 2 115200 115200 0 115200
serial 5 64 115200 57600 0 115200
[... 605 lines cut ...]
Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Flight controller

HolyBro KakuteF7 v1.5

SteveCEvans, 2022-02-28:
The issue with the missing beeps was caused by there being a beep associated with each write to FLASH. As we only write once throttle is lowered, that was lost. Beeps have been resurrected for each trim adjustment. There is thus a slight change in beep behaviour because there will now be a beep per adjustment, and a further beep when the setting are written to FLASH, so an extra beep at the end. I believe this to be reasonable.

Fixed in https://github.com/betaflight/betaflight/pull/11380

Martivip, 2022-03-13:
Hi @SteveCEvans, no it don't work I've tested today with the latest Development Master build no.2648.
I have two setups, one based on Ghost RC Link with STM32F745 (KakuteF7) target and one with ELRS RC Link together with BETAFPVF4SX1280 (ELRS SPI receiver on board) tagret.

For both setups my test observations are the same:
- after applying ACC trim roll, pitch adjustments via OSD stick command those adjustments are not stored in EEPROM of the FC, BF configurator shows zero values after applying them via OSD.
- for Trim ACC Forwards command there is no beep sound and there is no visual beeper indication as well

Martivip, 2022-03-20:
Hi @SteveCEvans ok anter instalation BF build no. 2653
I have two setups, one based on Ghost RC Link with STM32F745 (KakuteF7) target and one with ELRS RC Link together with BETAFPVF4SX1280 (ELRS SPI receiver on board) tagret.

- for ELRS Setup ACC trim rol and ich adjustments made via OSD are saved in FC is OK minor issue is that not always visual beeper is persented in OSD during adjustments
- For Ghost Setup, roll and pitch adjustments are **not saved i**n FC at throttle low so definitely something has been broken during the implementation od "save at throttle low" (but the way this save at throttle low has been implemented only to SPI ELRS FC or for FC without SPI ELRS receiver as well ?) and also there is no visual beeper presentation during the ACC trim forward.

Martivip, 2022-04-03:
@SteveCEvans after installation BF build no. 2662 sill I facing the same issues :

I have two setups, one based on Ghost RC Link with STM32F745 (KakuteF7) target and one with ELRS RC Link together with BETAFPVF4SX1280 (ELRS SPI receiver on board) tagret.

for ELRS Setup ACC trim rol and ich adjustments made via OSD are saved in FC is OK minor issue is that not always visual beeper is persented in OSD during adjustments
For Ghost Setup, roll and pitch adjustments are not saved in FC at throttle low so definitely something has been broken during the implementation of "save at throttle low" (but the way this save at throttle low has been implemented only to SPI ELRS FC or for FC without SPI ELRS receiver as well ?) and also there is no visual beeper presentation during the ACC trim forward.

Martivip, 2022-04-09:
for the BF 4.3 RC4 the same status as above

haslinghuis, 2022-04-11:
Please test https://github.com/betaflight/betaflight/pull/11509

klutvott123, 2022-04-13:
I just tested with crossfire and ghost and it works. I don't have a beeper or OSD so I can't comment on that. 
@Martivip ACC trim stick commands only works in angle or horizon mode. Is your ghost setup in one of these modes?

Martivip, 2022-04-14:
@haslinghuis yes I confirm Acc calibration and trims works, the only issue is that Visual Beeper is not properly displayed in OSD, most of the time is not displayed at all :)

haslinghuis, 2022-04-14:
This PR da0a42aed was solving the visual beeper in 4.3: https://github.com/betaflight/betaflight/pull/11359

For further investigation, the code is a little confusing as visual_beeper is also used for ledstrip.
So we are looking at `osd_warn_visual_beeper` not `ledstrip_visual_beeper`.

ctzsnooze, 2022-04-15:
I tested on an F722 and found that with stick-banging I can change the acc trim, the LED flashes, but the 'visual beeper' does not 'flash' in the OSD at the same time.

klutvott123, 2022-04-15:
The issue could be that showVisualBeeper is both set and reset before it's rendered. 
https://github.com/betaflight/betaflight/blob/21c8d67e9545c9cee2de619ec0e19838de6eb57b/src/main/osd/osd.c#L1207-L1209

klutvott123, 2022-04-16:
At the slow rate the OSD task is running at there's no guarantee that we will catch the case where isBeeperOn() == true.
It would probably work if the beeper code set the visual beeper too. then reset it once rendered.

klutvott123, 2022-04-16:
@Martivip Could you try this, please?
[betaflight_4.3.0_STM32F745_norevision (2).zip](https://github.com/betaflight/betaflight/files/8500683/betaflight_4.3.0_STM32F745_norevision.2.zip)

Martivip, 2022-04-17:
> @Martivip Could you try this, please?
> [betaflight_4.3.0_STM32F745_norevision (2).zip](https://github.com/betaflight/betaflight/files/8500683/betaflight_4.3.0_STM32F745_norevision.2.zip)

@klutvott123 no still no visual beeper in OSD

Martivip, 2022-04-17:
> > @Martivip Could you try this, please?
> > [betaflight_4.3.0_STM32F745_norevision (2).zip](https://github.com/betaflight/betaflight/files/8500683/betaflight_4.3.0_STM32F745_norevision.2.zip)
> 
> @klutvott123 no still no visual beeper in OSD

@klutvott123 precisely speaking to sometimes it has been rendered but really rarely, most of the time not at all

klutvott123, 2022-04-17:
@Martivip Thanks for testing. We'll have to look into this. It's a bit hard to test as I don't have VTX equipment so I can't actually see the OSD

klutvott123, 2022-04-17:
All right, I think this should work. It uses a latching approach. This lets the beeper set `showVisualBeeper` and it is only reset once the "****" has been drawn and the beeper is off.
[betaflight_4.3.0_STM32F745_norevision (3).zip](https://github.com/betaflight/betaflight/files/8502083/betaflight_4.3.0_STM32F745_norevision.3.zip)

@Martivip Could you try it, pelase?

Martivip, 2022-04-17:
> All right, I think this should work. It uses a latching approach. This lets the beeper set `showVisualBeeper` and it is only reset once the "****" has been drawn and the beeper is off. [...]

@klutvott123 Hi Ok it works, but sometimes when I applying ie. one step of ACC trim (direction it doesn't matter), instead of one blink "****" it blinks randomly couple of times two or three.

klutvott123, 2022-04-17:
Is this after bringing throttle back to zero?

Martivip, 2022-04-17:
No just after Acc trim command before throttle low

Martivip, 2022-04-17:
when I applying Acc trim command sometimes Visual beeper is randomly rendered twice and yes I know that throttle low will trigger visual beeper once again :)

Martivip, 2022-04-17:
In general this one extra visual beeper render can mislead the operator that he made an increment by two steps instead of one. We need to keep in mind that visual beeper is mostly used without real beeper and that's why the visual beeper should be rendered only once per one increment step :)

klutvott123, 2022-04-17:
@Martivip I agree, the extra beep is not desired behaviour. I made some changes that hopefully will work 
[betaflight_4.3.0_STM32F745_norevision (4).zip](https://github.com/betaflight/betaflight/files/8502412/betaflight_4.3.0_STM32F745_norevision.4.zip)

Thanks for helping testing this. Much appreiated. I really need to get some equipment so I can test this kind of stuff myself

Martivip, 2022-04-18:
> @Martivip I agree, the extra beep is not desired behaviour. I made some changes that hopefully will work [betaflight_4.3.0_STM32F745_norevision (5).zip](https://github.com/betaflight/betaflight/files/8502449/betaflight_4.3.0_STM32F745_norevision.5.zip)
> 
> Thanks for helping testing this. Much appreciated. I really need to get some equipment so I can test this kind of stuff myself
> 
> Edit: Sorry, I uploaded the wrong binary. Updated with the correct one.

@klutvott123 Hi ok it works as it should right now, perfect !

one more thing, not related to this bug but maybe you can make a quick look, I'm notice that one Acc Trim increment (one beep real or visual beeper) is 2 value increments from the configurator perspective
[image]
i think that via one ACC trim increment should be done by 1 value from configurator perspective, one beep one increment from configurator perspective, what do you think ?

klutvott123, 2022-04-18:
PR to fix this is up https://github.com/betaflight/betaflight/pull/11538

@Martivip Looks like the steps of two is hardcoded. Changes would have to be made to firmware and configurator. We'll have to investigate why it's like this before doing anything. Because this is not a bug, changing it would have to wait until BF 4.4. But I agree, it seems a little strange. https://github.com/betaflight/betaflight/blob/954b7373bbe8a5a64be68dbfa5d030d03f0c56bb/src/main/fc/rc_controls.c#L317-L334

Martivip, 2022-04-18:
@klutvott123 Thank you very much for taking your time for this visual beeper issue :)
And I agree those hard coded increments should be reviewed, definitely to put in BF future releasses backlog.

Martivip, 2022-04-18:
> PR to fix this is up #11538
>
> @Martivip Looks like the steps of two is hardcoded. Changes would have to be made to firmware and configurator. We'll have to investigate why it's like this before doing [...]

@klutvott123 are you sure that steps of two are also hardcoded in configurator ? I'm asking because in configurator we are able to perform increments by one step not two

klutvott123, 2022-04-18:
Just checked and your right. It's just when doing the adjustment with stick commands

Martivip, 2022-04-18:
> Just checked and your right. It's just when doing the adjustment with stick commands

@klutvott123 so I'm not pushing :) if you find some spare time then maybe you will be able to create quick PR for that

klutvott123, 2022-04-18:
I can do it, but it will still be for 4.4 🙂

haslinghuis, 2022-04-18:
Issue will be closed automatically upon merging the fixing PR.

Martivip, 2022-04-19:
> I can do it, but it will still be for 4.4 🙂

please do, but I don't understand why it must wait for 4.4  😑, the is not a new feature this is discrepancy removal between firmware and configurator 😄 that has been found.

closed 2022-04-20 by haslinghuis

haslinghuis, 2022-04-20:
@Martivip please open a new issue report so we can assign it.

Martivip, 2022-04-21:
> @Martivip please open a new issue report for the remaining issue so we can assign it as the original issue is now closed it won't get much attention.

@haslinghuis Done, Thank You

PR #11538 (2022-04-20): Fix OSD visual beeper
Fixes the visual beeper in OSD.

Because of the osd task running at a low rate, there's no guarantee that we'll be able to catch the case where the beeper is on. 
This PR fixes it by having the beeper task turn on the visual beeper for every new beep. 

Binaries for testing: 
[betaflight_4.3.0_STM32F7X2_norevision.zip](https://github.com/betaflight/betaflight/files/8505728/betaflight_4.3.0_STM32F7X2_norevision.zip)
[betaflight_4.3.0_STM32F405_norevision.zip](https://github.com/betaflight/betaflight/files/8505729/betaflight_4.3.0_STM32F405_norevision.zip)
[betaflight_4.3.0 [... cut]