betaflight/betaflight #11440: HolyBro Kakute H7 Mini ESC Firmware Issue (Blackbox)

JPOVPhoto, 2022-02-27:
### Describe the bug

I can't access the onboard flash for Blackbox Data. This happens in the following versions of Betaflight Configurator 10.7, and 10.8

Here is my issue. In the dump I have the following resource defined for the flash memory which is the unified target default value for the Kakute H7 Mini FC...

Code:
resource FLASH_CS 1 A04

In the active IO resource assignment list I have the following...

Code:
A04: PREINIT 2

If the blackbox flash chip had been assigned and initiated properly then you should be seeing the following in the active IO resource assignment list for A04...

Code:
A04: FLASH_CS

This occurs in both 4.2.10 and 4.2.11

### To Reproduce

Go to The Black box tab in Betaflight and attempt to switch blackbox to "onboard flash," but the option isn't there. You can choose serial, or none.

### Expected behavior

I should be able to choose onboard flash.

### Flight controller configuration

```
# diff all

# version
# Betaflight / STM32H743 (SH74) 4.2.11 Nov  9 2021 / 20:29:57 (948ba6339) MSP API: 1.43
# config: manufacturer_id: HBRO, board_name: KAKUTEH7MINI, version: c24ea9c4, date: 2021-12-16T07:02:07Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name KAKUTEH7MINI
manufacturer_id HBRO
mcu_id 002e003f3330511936353839
signature 

# name: JPOVC

# resources
resource MOTOR 1 B03
resource MOTOR 2 B10
resource MOTOR 3 B00
resource MOTOR 4 B01

# feature
feature -RX_PARALLEL_PWM

# map
map TAER1234

# aux
aux 0 0 0 1300 1700 0 0
aux 1 35 2 1700 2100 0 0

# master
set acc_calibration = -13,45,-418,1
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set blackbox_device = SERIAL
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set align_board_roll = 180
set small_angle = 180
set pid_process_denom = 2
set osd_rssi_pos = 2400
set osd_link_quality_pos = 2432
set osd_rssi_dbm_pos = 2502
set osd_avg_cell_voltage_pos = 2496
set debug_mode = GYRO_SCALED
set gyro_1_sensor_align = CW270FLIP
set gyro_1_align_pitch = 1800
set name = JPOVC

profile 0

# profile 0
set motor_output_limit = 66
set auto_profile_cell_count = 6

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```
```
  # resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: PREINIT 2
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: FREE
B03: MOTOR 1
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: MOTOR 2
B11: PINIO 1
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: ADC_CURR
C02: LED 1
C03: FREE
C04: FREE
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: CAMERA_CONTROL
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1:
CH1: CAMERA_CONTROL
TIM2:
CH2: MOTOR 1
CH3: MOTOR 2
TIM3:
CH3: MOTOR 3
CH4: MOTOR 4
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 3
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: MOTOR 1
DMA1 Stream 3: MOTOR 2
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

Holybro Kakute H7 Mini (purchased from Pyrodrone)

### Other components

Caddx Vista (most recent firmware)
TBS Crossfire Micro TX with Nano Diversity (not the most recent firmware)

### How are the different components wired up

Vista is connected to the FC via supplied jet plug. The FC is connected to the ESC via supplied cable and just plugs.
The TBS Nano Diversity is is wired per the diagram on the Holybro website.

haslinghuis, 2022-04-02:
@SteveCEvans any idea about this resource?

[image]

SteveCEvans, 2022-04-21:
The KAKUTEH7MINI FLASH device is reporting an ID of 0xEF3016 which we don't currently support. The fact that I get that far means the config is correct. Now trying to figure out what that device is.

The KAKUTEH7 FLASH device is not responding at all, so that may be more of a challenge.

SteveCEvans, 2022-04-21:
I've faked the same config in the FLASH detection as a W25Q128 and I can write and read BB logs, but I need to confirm the FLASH geometry before raising a PR.

SteveCEvans, 2022-04-21:
Stripped off the con-coat and the mystery is revealed...
[image]

SteveCEvans, 2022-04-21:
I'm seeing four copies of each log. The device is behaving like a 32Mbit device, and the device ID ending in 16 suggests that is the case. Treating this as if the ID is 0xEF4016 gives a 4Mbyte FLASH that works as expected. Query of the device ID with Winbond technical support...

haslinghuis, 2022-04-21:
🤯 Including https://github.com/betaflight/betaflight/pull/11537 did not solve it.

SteveCEvans, 2022-04-21:
Try this image.

[betaflight_4.3.0_STM32H743_norevision.hex.zip](https://github.com/betaflight/betaflight/files/8535716/betaflight_4.3.0_STM32H743_norevision.hex.zip)

SteveCEvans, 2022-04-21:
`flash_info` has been tweaked to dump the device ID. Let me know what yours reports.

haslinghuis, 2022-04-21:
The hex solved the problem:

```
# flash_info
ChipID=0xef3016
Flash sectors=64, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=4194304
Paritions:
  0: FLASHFS   0 63
FlashFS size=4194304, usedSize=0
```

Still configurator thinks it has an SDCard though.

SteveCEvans, 2022-04-22:
Oh dear. the FLASH is a W25X32, marked as a W25Q128.

https://www.winbond.com/resource-files/w25x32a_revb_080709.pdf#page14

KarateBrot, 2022-04-22:
@JPOVPhoto Bad news... this is not a firmware issue. Your onboard flash chip is not what it is supposed to be.

vincentpoont2, 2022-04-24:
Vince with Holybro here.

After the first production cycle, we found that there were a small percentage of the flash on the KakuteH7 Mini being W25X32, but marked as a W25Q128 (what it supposed to have). We immediately checked all our inventory and done reworked at that time, but now it looks like there may be a very small quantity of board with the W25X32 flash that was already shipped. All our Kakute H7 Mini board since then does not have this issue. 

Sorry for seeing this so late.

closed 2022-04-24 by blckmn

vincentpoont2, 2025-01-09:
> Same issue with new version of Kakute h7 mini (1.5). For me after pressing "Activate lash device" it's linking the drive. I can't read anything from there. Then drive gone. But I'm able to get logs through "Save logs into the file" button
> 
> No issues with other boards
> 
> `# flash_info Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018 Partitions: 0: FLASHFS 0 255 FlashFS size=16777216, usedSize=0 `
> 
> CC @vincentpoont2

@vitaly-rudenya Which firmware and configurator version are you running?

vitaly-rudenya, 2025-01-09:
@vincentpoont2 looks like user issue.
I reinstalled the cp210 driver - the problem went away.
Thank you for your trouble

vitaly-rudenya, 2025-01-09:
@vincentpoont2 
[image]