betaflight/betaflight #11441: GPS not detected if configured for 230400 baud

skorokithakis, 2022-02-27:
### Describe the bug

If a GPS is configured for 230400 baud (e.g. if it has been used with ArduPilot), it will always fail to connect to Betaflight, because Betaflight doesn't consider baud rates that high. This means that you can permanently break your GPS with BF if you connect it to AP.

### To Reproduce

Connect a GPS to AP (or configure it for 230400 baud). Connect it to BF, it won't be recognized.

### Expected behavior

If you connect it to BF again, it should work.

### Flight controller configuration

Any configuration.

### Flight controller

Any

### Add any other context about the problem that you think might be relevant here

Here's a similar INAV bug: https://github.com/iNavFlight/inav/issues/6159

Considering higher baud rates when autobauding would fix this.

skorokithakis, 2022-02-27:
Using the following hex message with uCenter to switch to 115200 baud, reconnecting to 115200 baud and then copy/pasting the baud change messages to "send hex message" by hand worked to set the GPS to 115200 permanently:

b5 62 06 00 14 00 01 00 00 00 d0 08 00 00 00 c2 01 00 07 00 07 00 00 00 00 00 c4 96 b5 62 06 00 01 00 01 08 22

etracer65, 2022-02-27:
Sounds like you just shouldn't configure your GPS module with an unsupported baud rate. And if you do, then simply use uCenter to reset it to a supported baud rate. The GPS is in no way "permanently broken". Not sure why you think Betaflight needs to fix this for you.

skorokithakis, 2022-02-27:
The GPS works fine and the baud rate is perfectly supported. Betaflight refuses to work with a GPS that's configured at 230400 baud, which is a bug with Betaflight.

etracer65, 2022-02-27:
Betaflight only supports up to 115200 baud. So it's not a bug. Just because some other software might support faster rates doesn't really have any bearing on Betaflight. If you want to use the GPS with Betaflight then set it to 115200.

skorokithakis, 2022-02-28:
> Betaflight only supports up to 115200 baud.

Yeah, so it doesn't support all the rates the GPS can do.

haslinghuis, 2022-02-28:
@skorokithakis it's not a bug - it's by design.

etracer65, 2022-02-28:
@skorokithakis 

I don't think you understand the meaning of a bug. It's not a bug when Betaflight doesn't do something that it's not documented or designed to do. Just because Betaflight doesn't take advantage of all the features of a particular GPS module doesn't make it some bug that needs fixing.

Closing this as it's not a bug. If you want to open a separate Feature Request to consider adding support for a higher baud rate then go ahead.

closed 2022-02-28 by etracer65

skorokithakis, 2022-02-28:
This isn't about the baud rate that BF should communicate at. This is about autobauding not working for all baud rates. Autobaud should detect the GPS at all supported baud rates and configure it down to whatever it wants. It's not "autobaud" if it doesn't work on half the baud rates.

etracer65, 2022-02-28:
Again, you don't understand how it works. "Auto-baud" in this implementation means it will try the baud rates that **Betaflight supports**. That means 115200, 57600, 38400, 19200, and 9600. Betaflight doesn't support 230400 so that why it doesn't work. This is not a bug, but a design decision. Set your GPS module to max supported 115200 baud and it will work fine.

haslinghuis, 2022-02-28:
@skorokithakis

> Closing this as it's not a bug. If you want to open a separate Feature Request to consider adding support for a higher baud rate then go ahead.

Please open a separate Feature Request as suggested. Autobaud is obvious limited to the implemented rates.