betaflight/betaflight #11448: Betaflight-tx-lua-scripts are not saved with SPI rx

bike797, 2022-03-02:
### Describe the bug

Betaflight-tx-lua-scripts are not saved with SPI rx.
 . Crazybee Express spi receiver
 . Crzybee Frsky spi receiver
But external uart receiver is normal operation.

### To Reproduce

 - Reproduce video
  . https://youtu.be/ppPBdMLSKlc 

- BetaFlight 4.3 RC3 with SPI receiver.
  . [Happymodel ELRS X1 AIO 2-4S Flight controller](http://www.happymodel.cn/index.php/2021/08/13/happymodel-elrs-x1-aio-2-4s-flight-controller-built-in-spi-2-4g-elrs-and-12a-esc-for-toothpick/)
  . [CrazybeeX v1.0 All-IN-ONE flight controller 1-2s compatible for brushless whoops](http://www.happymodel.cn/index.php/2020/05/28/crazybeex-v1-0-all-in-one-flight-controller-1-2s-compatible-for-brushless-whoops/)
- Betaflight-tx-lua-scripts [v20220204.5899](https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases/tag/v20220204.5899)

### Expected behavior

The fc setting value can be changed through the tx lua script of the radio, but it is not saved in the FC.
Reverts to previous value when rebooting

The setting value is normally changed and saved through the OSD stick command. (with spi receiver)

### Flight controller configuration

```
# diff all

# version
# Betaflight / CRAZYBEEF4SX1280 (HAMO) 4.3.0 Feb  1 2022 / 20:05:07 (3267f0417) MSP API: 1.44

board_name CRAZYBEEF4SX1280
manufacturer_id HAMO
mcu_id 002300614e56500720363955
signature 

# name: CRUX35

# resources
resource LED_STRIP 1 NONE
resource CAMERA_CONTROL 1 A00
resource FLASH_CS 1 NONE

# dma
dma pin A09 NONE
dma pin A10 NONE

# feature
feature -AIRMODE
feature -ANTI_GRAVITY
feature MOTOR_STOP
feature GPS

# serial
serial 0 8192 115200 57600 0 115200
serial 1 2 115200 57600 0 115200

# map
map TAER1234

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 46 3 1700 2100 0 0
aux 4 13 2 1325 1725 0 0
aux 5 19 6 1725 2100 0 0
aux 6 28 1 1300 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 400 200 100 25
vtxtable powerlabels 25 400 200 100 25

# master
set gyro_lpf1_static_hz = 0
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 0
set acc_trim_pitch = 60
set acc_trim_roll = -4
set acc_calibration = -18,-28,-112,1
set serialrx_provider = CRSF
set blackbox_sample_rate = 1/1
set blackbox_device = NONE
set dshot_idle_value = 500
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set failsafe_procedure = GPS-RESCUE
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 290
set vbat_warning_cell_voltage = 360
set ibata_scale = 470
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_min_dth = 60
set deadband = 1
set yaw_deadband = 1
set report_cell_voltage = ON
set osd_warn_batt_not_full = OFF
set osd_warn_batt_critical = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_warn_link_quality = ON
set osd_warn_rssi_dbm = ON
set osd_rssi_dbm_alarm = -98
set osd_cap_alarm = 353
set osd_alt_alarm = 0
set osd_vbat_pos = 439
set osd_rssi_pos = 225
set osd_link_quality_pos = 2177
set osd_link_tx_power_pos = 2145
set osd_rssi_dbm_pos = 2209
set osd_tim_1_pos = 33
set osd_tim_2_pos = 2081
set osd_flymode_pos = 2401
set osd_throttle_pos = 2337
set osd_vtx_channel_pos = 2099
set osd_current_pos = 2454
set osd_mah_drawn_pos = 2392
set osd_craft_name_pos = 2433
set osd_gps_speed_pos = 2125
set osd_gps_lat_pos = 16618
set osd_gps_sats_pos = 2113
set osd_home_dir_pos = 2118
set osd_home_dist_pos = 2120
set osd_altitude_pos = 2130
set osd_debug_pos = 65
set osd_pidrate_profile_pos = 2369
set osd_warnings_pos = 14506
set osd_avg_cell_voltage_pos = 2423
set osd_disarmed_pos = 2410
set osd_esc_tmp_pos = 140
set osd_esc_rpm_pos = 257
set osd_rtc_date_time_pos = 104
set osd_core_temp_pos = 2092
set osd_total_flights_pos = 2088
set osd_stat_rtc_date_time = ON
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_max_fft = ON
set osd_stat_min_rssi_dbm = ON
set debug_mode = BATTERY
set cpu_overclock = 120MHZ
set vtx_band = 5
set vtx_channel = 6
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5843
set camera_control_ref_voltage = 327
set camera_control_internal_resistance = 270
set camera_control_button_resistance = 460,280,160,78,40
set flash_spi_bus = 0
set expresslrs_uid = 0,0,0,0,0,0
set expresslrs_switch_mode = WIDE
set expresslrs_model_id = 12
set rpm_filter_fade_range_hz = 100
set stats_min_armed_time_s = 50
set stats_total_flights = 3
set stats_total_time_s = 757
set stats_total_dist_m = 8022
set name = CRUX35

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 90
set dterm_lpf1_dyn_max_hz = 180
set dterm_lpf1_dyn_expo = 7
set dterm_lpf1_static_hz = 90
set dterm_lpf2_static_hz = 180
set vbat_sag_compensation = 100
set iterm_relax = RPY
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set feedforward_transition = 21
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65
set feedforward_jitter_factor = 5
set dyn_idle_min_rpm = 29
set simplified_dterm_filter_multiplier = 120

rateprofile 1

# rateprofile 1
set rates_type = BETAFLIGHT
set roll_rc_rate = 105
set pitch_rc_rate = 105
set yaw_rc_rate = 100
set roll_srate = 73
set pitch_srate = 73
set yaw_srate = 70

save
# 
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: CAMERA_CONTROL
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI_EXPRESSLRS_RESET
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_EXPRESSLRS_BUSY
A14: SWD
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : DSHOT_BITBANG 2
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1 : CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

CRAZYBEEF4SX1280

### Other components

- BetaFligt 4.3.0-RC3 
- SPI RX
- EdgeTX2.6 
- ExpressLRS TX v2.1

CMartin249, 2022-03-07:
Confirmed this continues to be an issue on current nightly build and betaflight_4.3.0_CRAZYBEEF4SX1280_667847d.hex in pull #11446. Using the VTX Admin function in ELRS LUA also does not appear to save on SPI ELRS targets. This causes a delay after plugging in before connection is established and TX changes the channel.

ss8688, 2022-03-27:
frsky spi seems to have same issue https://github.com/betaflight/betaflight/issues/11489

haslinghuis, 2022-04-20:
Please check with latest development build

bike797, 2022-04-20:
I think. This bug may related with below code.
please check it.

https://github.com/betaflight/betaflight/blob/d08ab1c6e8dd40e390b4a38e5d1dddfe7e24a022/src/main/rx/expresslrs_telemetry.c#L293-L297

klutvott123, 2022-04-20:
Yes, that's the code that prevents the the eeprom write command from being handled. I'm currently working on a possible solution for it.

klutvott123, 2022-04-30:
FIx here if anyone wants to try it https://github.com/betaflight/betaflight/pull/11573

closed 2022-05-27 by haslinghuis