betaflight/betaflight #11451: ELRS code broken on some MCUs.

hydra, 2022-03-04:
### Describe the bug

Since #11380 was merged, without testing on anything else than an F411, the ELRS code is currently broken.

Development was done on F4 and H7 by @phobos- @AlessandroAU and myself.

An H7 board was send to @SteveCEvans but it was not used for testing, likely because it's not in `master` yet, see #11110.

Anyway, after extensive investigation into the issue, involving scope probing, code and register inspection the likely cause of the failure is the lack of BUSY pin initialisation in the latest ELRS code from #11380.

On some MCUs the default state of GPIO pins is analog mode (e.g. H7), on others it's input mode (F4).

Here's the manuals for the H730 and F411, H7 on the left, F4 on the right.  Default difference highlighted.
[image]

What this means is that the `sx1280PollBusy` function fails, because the bit for the BUSY pin in the IDR register on the H7 is *always* `0`, so the code assumed the SX1280 isn't busy and then continues to communicate it with it when it shouldn't.

### To Reproduce

Flash https://github.com/spracing/betaflight/commits/bf-h7rf-wip-13 to H7RF, observe unable to bind.
Flash https://github.com/spracing/betaflight/commits/bf-h7rf-wip-11 to H7RF, observe able to bind.

The main difference between the two branches is #11380.

### Expected behavior

It works on H7 and F4.

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```

### Flight controller

SPRacingH7RF

SteveCEvans, 2022-03-04:
@hydra Focus for 4.3 is to support the majority of users and for SX1280 support that most certainly aligns with those using F411. And yes, the fact that H7RF support isn't in master doesn't help, nor the fact that it's awkward to debug due to a JLink not having access to its FLASH. Nonetheless, thank you for the hardware; in the past the SPRACINGH7ZERO was used to develop USB MSB support and the new SPI API supporting DMA. And the H7RF will be getting some attention now that the hard work of creating the F411 support is done.

For such FCs, waiting in the wings, it is entirely appropriate to discuss and resolve issues after master has been updated with new features, as we have done this evening. To delay a release candidate to support something that the release won't support doesn't make sense.

A call in `sx1280Init()` to `sx1280EnableBusy()` followed by one to `sx1280EnableExti()` will fix this.

hydra, 2022-03-04:
Example of where busy goes high and is ignored:

[image]

Here's what it looks like with a work-in-progress fix, the code now waits for busy to go low:

[image]

```
$ git show
commit 7f881efab8d2002fa20bff11c4a0358085a8a8d7 (HEAD -> bf-h7rf-wip-13)
Author: Dominic Clifton <dominic.clifton@cleanflight.com>
Date:   Fri Mar 4 20:22:52 2022 +0100

    Fix broken ELRS code on H7.

    The reason it's broken on the H7 is that the default gpio mode is
    ANALOG, not INPUT like it is on other MCUs, and that the busy pin is
    used before communication to the SX1280 and the busy wait function
    returns to early because the GPIO IDR bit is always 0 on the H7.  i.e.
    not busy, when it is.

    It may also be broken on other MCUs prior to this.

    Fixes #11451

diff --git a/src/main/drivers/rx/rx_sx1280.c b/src/main/drivers/rx/rx_sx1280.c
index f8550d70ee..bf9a5cc5a4 100644
--- a/src/main/drivers/rx/rx_sx1280.c
+++ b/src/main/drivers/rx/rx_sx1280.c
@@ -208,6 +208,7 @@ bool sx1280Init(IO_t resetPin, IO_t busyPin)

     if (busyPin) {
         IOInit(busyPin, OWNER_RX_SPI_EXPRESSLRS_BUSY, 0);
+        IOConfigGPIO(busyPin, IOCFG_IN_FLOATING);
     } else {
         busyPin = IO_NONE;
     }
```

hydra, 2022-03-04:
@haslinghuis yes I understand the current focus. I'm just a little disappointed.  Yes, boards with external flash are harder to work with, not just compile and hit go.  Still, no debugger was needed to know that it just didn't work.

Anyway...

With the above fix it still doesn't bind, so there's more to fix.

I'm also noting a mismatch with the names of the entries in the NVIC.h table.

Steve also mentioned that PR #11380 wasn't tested on SX1270.

See NVIC_PRIO_RX_INT_EXTI vs NVIC_PRIO_RX_SPI_INT_EXTI.  That needs cleanup.

hydra, 2022-03-04:
previous notes in text form, for those following along with scope traces:

```
         NEW CODE                      OLD CODE     

         GPIO      
         MODER   PUPDR  IDLE ACTIVITY  MODER  PUPDR   IDLE ACTIVITY    SCOPE CH/SIGNAL COLOR/WIRE TAG
PC7 BUSY 0x3  A  0x0 N  LOW  YES       0x0 I  0x1 PU  LOW  YES         CH2/CYAN/BLUE
PC6 DIO1 0x0  I  0x2 PD LOW  NO        0x0 I  0x2 PD  LOW  YES         CH4/BLUE/GREEN
PD4 DIO2 0x0  I  0x1 PU N/A  N/A       0x0 I  0x1 PU  N/A  N/A
```

[image]

current state is that the DIO line (CH4, BLUE) goes high once and only once, no amount of cli `bind_rx` or TX poking - turning off/on/binding - makes it show any activity.

hydra, 2022-03-04:
work-in-progress branch for H7RF based on a recent `master` is here: https://github.com/spracing/betaflight/commits/bf-h7rf-wip-14

hydra, 2022-03-05:
The current state is that the EXTI handler for BUSY appears to be working as intended, as demonstrated by the 'after' scope trace where the busy signal is no-longer ignored.  The  EXTI handler for DIO is hit once, and only once.

hydra, 2022-03-05:
The callback handler for `sx1280IrqCmdComplete` is never called, this usually occurs when an SPI transaction completes.

https://github.com/betaflight/betaflight/blob/master/src/main/drivers/rx/rx_sx1280.c#L642

however, the exti handler is pre-empting the spi irq handler, as seen in this stacktrace.

[image]

the line that was pre-empted was this line:

https://github.com/betaflight/betaflight/blob/master/src/main/drivers/bus_spi.c#L426
```
bus->curSegment = (busSegment_t *)BUS_SPI_FREE;
```

When the EXTI handler is called, the bus still has a valid curSegment so the call to spiSequence adds the segments to the current sequence, then when the exti isr returns, the current segment is erased by the spiIrqHandler code.

[image]

as observed:

[image]

So probably the exti nvic priority is wrong.

hydra, 2022-03-05:
first call to EXTIConfig results in this

[image]

second call to EXTIConfig results in this

[image]

I pushed a commit to the WIP branch that might fix it, testing tomorrow...

https://github.com/spracing/betaflight/commit/f933f20d8225124165aa29dafbfa50efc44bf853

hydra, 2022-03-06:
It still doesn't work, now 0x00 gets written one to IPR5/N3, only once because when they are corrected to be the same priority no change is made due to the `if (extiGroupPriority[group] > irqPriority)` check before the call to HAL_NVIC_SetPriority.

[image]

And then a bad value, 0x00, is written, as below:

[image]

Snippet of the macros and code, which you can see compiled here: https://godbolt.org/z/Ye4z5bcMa
```
#include<stdint.h>

#define NVIC_PRIORITYGROUP_2         ((uint32_t)0x00000005) /*!< 2 bits for pre-emption priority */

#define NVIC_PRIO_MAX                      NVIC_BUILD_PRIORITY(0, 1)
#define NVIC_PRIO_OFFSET                   1
#define NVIC_PRIO_LOWEST                   0x0f

#define NVIC_PRIO_TIMER                    NVIC_BUILD_PRIORITY(NVIC_PRIO_OFFSET + 1, 1)
#define NVIC_PRIO_RX_INT_EXTI_OLD          NVIC_BUILD_PRIORITY(NVIC_PRIO_LOWEST, 0x0f)

#define NVIC_PRIO_RX_SPI_INT_EXTI          NVIC_BUILD_PRIORITY(NVIC_PRIO_OFFSET + 3, 0x0f)
#define NVIC_PRIO_RX_INT_EXTI              NVIC_BUILD_PRIORITY(NVIC_PRIO_OFFSET + 3, 0x0f)
#define NVIC_PRIO_RX_BUSY_EXTI             NVIC_BUILD_PRIORITY(NVIC_PRIO_OFFSET + 3, 0x0f)

#define NVIC_PRIO_SPI_DMA                  NVIC_BUILD_PRIORITY(NVIC_PRIO_OFFSET + 0, 0)

#define NVIC_PRIORITY_GROUPING NVIC_PRIORITYGROUP_2
#define NVIC_BUILD_PRIORITY(base,sub) (((((base)<<(4-(7-(NVIC_PRIORITY_GROUPING))))|((sub)&(0x0f>>(7-(NVIC_PRIORITY_GROUPING)))))<<4)&0xf0)

void main() {
    unsigned int priorities[8] = {
        NVIC_PRIO_MAX,             //   16 / 0x10
        NVIC_PRIO_RX_INT_EXTI_OLD, //  240 / 0xf0

        NVIC_PRIO_RX_SPI_INT_EXTI, //   48 / 0x30
        NVIC_PRIO_RX_INT_EXTI,     //   48 / 0x30
        NVIC_PRIO_RX_BUSY_EXTI,    //   48 / 0x30

        NVIC_PRIO_TIMER,           //  144 / 0x90
        NVIC_PRIO_SPI_DMA,         //   64 / 0x40
        NVIC_PRIO_LOWEST,          //   15 / 0x0f
    };
}
```

Seems like there's an issue with the NVIC macros or something.

@ledvinap As the original author of the nvic.h macros, can you spot anything wrong with the nvic.h file in my branch and master?  My branch just has a commit that bumps some priorities by one at compile-time.

hydra, 2022-03-06:
ok, i think i see the issue, the code is trying to use 3 bits for priority but the define is for 2.

hydra, 2022-03-06:
Thinking that it might be a good idea to have a unit test that checks the NVIC priority levels are calculated correctly.

hydra, 2022-03-06:
@ledvinap Do you remember why only 2 bits were used for NVIC priority?  Any reason *not* to change it to use 3?

hydra, 2022-03-06:
When I bumped the bits to 3 it appears to work, I get a link established so the spi callback handlers work.

To summarize, there were these causes/issues:
* Incorrect GPIO configuration.
* Multiple EXTI lines sharing the same interrupt.
* EXTI PR bits not being cleared when an EXTI line is disabled.
* SPI RX NVIC configuration was incorrect causing SPI ISR's to be pre-empted.
* SPRacingPixelOSD needed 1 extra priority bit.  It might be possible to use fewer bits and use different group priorities, but not going down that route at this time as it's more complicated.

hydra, 2022-03-06:
I cut a release for SPRacingH7RF users to test.

https://github.com/spracing/betaflight/releases/tag/spracingh7-20220306-1100

@haslinghuis @SteveCEvans @AlessandroAU @phobos- I suggest we add ELRS to a NucleoH7 target and provide instructions on how to wire up an SX1280 module so that developers can test on a target in `master`.

hydra, 2022-03-06:
PR's to fix:

https://github.com/betaflight/betaflight/pull/11456 - EXTI fixes
https://github.com/betaflight/betaflight/pull/11457 - ELRS fixes

The change to the extra priority bit required for the H7RF (#11110) will go in #10702

SantiCO19, 2022-03-06:
> I cut a release for SPRacingH7RF users to test.
>
> https://github.com/spracing/betaflight/releases/tag/spracingh7-20220306-1100
>
> @haslinghuis @SteveCEvans @AlessandroAU @phobos- I suggest we add ELRS [...]

Link to my H7RF established immediately. I can't tell any difference from any previous working firmware, though Im only able to hover around in the house right now. I'll take it out for more testing in a few hours when I have some daylight to work with.

hydra, 2022-03-11:
@haslinghuis No, this is not fixed by #11456 as https://github.com/betaflight/betaflight/pull/11457 is also needed.

haslinghuis, 2022-03-11:
@hydra yes it was closed automatically as one of the PR's was merged.

closed 2022-03-15 by haslinghuis