betaflight/betaflight #11452: ESC TLM HalfDuplex on G4 via LPUART1

x4FF3, 2022-03-04:
### Describe the bug

Hi,

i try to get ESC TLM via halfduplex working on a G4 proto board on pin PB11, which is TX of LPUART1. When defining `esc_tlm_halfduplex = ON` i wont get any data. Using this pin as RX3, data is shown correctly.

KR
David

### To Reproduce

- define PB11 as TX of LPUART1
- set sensor ESC on LPUART1
- activate ESC sensor in motor tab
- `set esc_tlm_halfduplex = ON`

### Expected behavior

receiving data via LPUART1 TX halfduplex

### Flight controller configuration

none

### Flight controller

proto board

haslinghuis, 2022-03-04:
If using `STM32G47X unified target` it should work.

x4FF3, 2022-03-08:
So, i looked at it again and it seems the port isn't configured properly.
In the `target.h` i do `#define LPUART1_TX_PIN PB11`. After flashing the firmware, when i list the resources, it is listed as `resource SERIAL_RX 10 B11`. When i free the resource and do a `resource SERIAL_TX 10 B11` LPUART1 is gone in the ports tab.

haslinghuis, 2022-03-29:
Thanks for the update.

closed 2022-03-29 by haslinghuis