betaflight/betaflight #11453: GPS working in BF4.2 but not in BF4.3 RC3

hugonaneix, 2022-03-05:
### Describe the bug

Flywoo GOKU HD745 EVO 40A
Nameless Tiny GPS

When in BF4.2.x (tested 4.2.8 and 4.2.11) the GPS is correctly recognized and works as expected
When updated to BF4.3 RC3, no GPS is recognized by the FC/Betaflight Configurator
Connected to recommended UART 6 in documentation
No Telemetry output to the radio either (just in case it could have been a GUI flaw)
Wiring is ok
BF configuration is ok (worked in 4.2.x, and replicated in 4.3RC3)
Target board: FLYWOOF745
board_name FLYWOOF722EVO

### To Reproduce

Flash board to BF4.3 RC3 
Configure GPS correctly
Observe the GPS is not recognized
gpspassthrough makes BF Configurator freeze

### Expected behavior

GPS should work as expected

### Flight controller configuration

Below are the 2 dumps

[dump_4.2.11.txt](https://github.com/betaflight/betaflight/files/8189369/dump_4.2.11.txt)
[dump_4.3.txt](https://github.com/betaflight/betaflight/files/8189370/dump_4.3.txt)

### Flight controller

Flywoo GOKU HD745 EVO 40A

### Other components

Nameless Tiny GPS

### How are the different components wired up

Soldered on FC, plug on GPS

jsk2084, 2022-03-06:
I have the same GPS on a Flywoo F745 Versatile. I recognize the GPS not working on latest development build (#2646) with similar config settings (Ublox). I changed protocol to NMEA and the GPS is not recognized on the Betaflight GUI. 

Have ya tried NMEA protocol?

hugonaneix, 2022-03-06:
@jsk2084 so can I assume this is an issue that will be adressed in future builds? Just want to help and not really a pro when it comes to dev collaboration ;)

hugonaneix, 2022-03-06:
@jsk2084 I played with uCenter to try to change the GPS configuration, but forgot to backup its configuration...
if, by any chance you could attach your config in that thread that would be very appreciated
Here's how to do it: https://www.youtube.com/watch?v=zxMdOtspUGk&ab_channel=XR-4Drone

jsk2084, 2022-03-06:
I imagine someone will look into the changes that broke it and come up with a fix eventually! I know nothing of coding, just a grateful user of Betaflight. I will try to get a default config of the gps device, though earlier I was having issue connecting it to the ucenter software.

jsk2084, 2022-03-19:
After additional testing in latest development builds, I've determined that enabling auto-config with this NamelessRC TinyGPS in ublox protocol prevents it from working. Flashing build #2579 from Dec 20th 2021 and enabling auto-config there fixes the issue.

Leaving auto-config disabled seems to be required, otherwise the gps module settings become incompatible. I'm not sure if any other GPS module would be affected by this...

haslinghuis, 2022-03-19:
@TonyBlit any idea? There seems to be an issue with https://github.com/betaflight/betaflight/pull/10921

TonyBlit, 2022-03-31:
I've been looking for some info about this gps without success, all I found is the barest user manual I've ever seen. I'd need to know what ublox module this is using. However, I fear this may not be using a module manufactured by ublox.
Anyway, I've ordered one from fullspeed and it will take one or two weeks to arrive.
In the meantime, if anyone with this gps is willing to help, you can reach me on Discord (TonyBlitter on the BF server) and I can provide further instructions on how to debug what is happening.

miromir237, 2022-04-11:
My Matek M8Q-5883 is also not working on Zeus HGLRC F722 board.

UglyBabyGaming, 2022-04-20:
Target board: FLYWOOF745
board_name FLYWOOF722EVO???
are you sure you got the right target?

hugonaneix, 2022-04-20:
that's a typo, the board name is FLYWOOF745AIO

ea-fpv, 2022-06-10:
Similar issue with RC7 here as reported in https://github.com/betaflight/betaflight/issues/11508#issuecomment-1144772269 with HGLRC M80 wired to Betafpv405AIO

haslinghuis, 2022-06-10:
@TonyBlit did you get the GPS and had the time to investigate?
As a cheap GPS module is it worth the buy?

TonyBlit, 2022-06-11:
Yes I did bought one TinyGPS, but I'm still figuring out why some commands refuse to work.

What I can say for sure is these modules work totally fine:
- Beitian BN-220
- HGLRC Neo-M8N
- Matek SAM-M8Q
- Matek M9N-5883
- Flywoo GM8 Mini

For TinyGPS, I see a few commands are returning a NACK, and debugging this is very painful (compile and flash each try/change, unplugging the GPS every time) so it will take me a bit of time getting a proper fix.

However, in the meantime there is a quick workaround for that module: just disable auto-config. In practice there is little difference as I think the commands sent by BF 4.2 are failing as well so the module is ignoring them. 4.2 code was less strict and it ignored the NACKs, so it's not that it correctly worked in 4.2, it's that there was a nonsensical interchange of commands that was just ignored on both ends. 4.3 code is more strict, so meanwhile I find a clean fix disabling auto-config is quite equivalent to what happened in 4.2.

haslinghuis, 2022-06-11:
Thanks for the list @TonyBlit and workaround.

andytroy, 2022-06-13:
GPS OSD and GPS rescue are not working for me on RC7, using TBS M8 and Matek M8Q. In BF configurator the satellites are visible. It worked fine in RC6. The work around doesn’t work for me.

TonyBlit, 2022-06-17:
@Kriseye @andytroy can you provide a clear explanation? if BF configurator shows satellites, it means the module is working.

andytroy, 2022-06-19:
> @Kriseye @andytroy can you provide a clear explanation? if BF configurator shows satellites, it means the module is working.

The DJI OSD is not showing any satellites, home direction etc. So the whole GPS thing is kinda useless right now for long range flights.

vahdat-ab, 2022-06-30:
I have been dealing with this GPS issue for a long time #11508. I flashed the latest release (4.3) and the issue is still there. Now, it detects the GPS but it becomes so unstable that every 10-24 seconds it loses the GPS (zero satellites suddenly) and gains them again. Buzzer beeps show it as well during the flight. This instability started to happen with the NMEA protocol. It seems the ublox protocol is doing ok so far.  I disabled auto-config and it didn't help at all. Today, I lost a motor and frame while testing the rescue mode :-(
I use HAKRC F722 with a new HGLRC GPS (NMEA protocol).

dukevah, 2022-06-30:
I have been dealing with this GPS issue for a long time #11508. I flashed the latest release (4.3) and the issue is still there. Now, it detects the GPS but it becomes so unstable that every 10-24 seconds it loses the GPS (zero satellites suddenly) and gains them again. Buzzer beeps show it as well during the flight. This instability started to happen with the NMEA protocol. It seems the ublox protocol is doing ok so far.  I disabled auto-config and it didn't help at all. Today, I lost a motor and frame while testing the rescue mode :-(
I use HAKRC F722 with a new HGLRC GPS (NMEA protocol).

cwchapma, 2022-08-10:
@cronosrj  also having trouble with HGLRC M80PRO.  Do you just get a single green flash when the gps powers up and then nothing?

cwchapma, 2022-08-10:
Confirmed HGLRC M80PRO works on 4.2.11
[image]

 but not 4.3.1
 
[image]

TonyBlit, 2022-08-10:
@cwchapma if you use ublox protocol with autoconfig disabled, does it work? have you tried nmea?

cwchapma, 2022-08-11:
Ublox with auto config disabled does not work.  Also tried nmea and ublox at 9600 baud with no success.

TonyBlit, 2022-08-11:
if nmea and ublox without autoconfig doesn't work, the problem is clearly unrelated with recent GPS ublox changes, and that would be better addressed in a new issue report. One thing I've noticed though is that all FC's reported so far are F7's. @cwchapma is it your case as well?

cwchapma, 2022-08-11:
Yes, it's a speedybee F7

haslinghuis, 2022-08-14:
@TonyBlit FYI:

[image]

LeoniePhiline, 2022-08-14:
If it helps finding the root cause:
 
I have an iFlight BLITZ mini f722 v1.0 working with both GNSS and mag on an iFlight M8Q-5883.
Using Betaflight 4.3.1.

This configuration works for me:
[image]

cronosrj, 2022-08-19:
> ‎também ter problemas com o HGLRC M80PRO. Você só tem um único flash verde quando o GPS aumenta e depois nada?‎

I'm about to change the gps module to another one compatible with BF4.3... apparently this will not be resolved.

cronosrj, 2022-08-19:
O M80 funciona normalmente.

cronosrj, 2022-11-25:
Update?

Ls4life, 2023-07-04:
I just like to add to this discussion that I have two FC, "skystars Mini F722 HD pro and the skystar F722HD pro3" and both FC gps stop working after I upgraded to Betaflight above 4.2. I have used the following GPS model: TBS M8-2, Beitian BN-160, BN-220, and Mateksys M8Q-5883. I tried different UART and verify the connectivity. I see some telemetry on my Radio but it will not show any Satellite info, I get not data on the Betafight GPS tap. It seem the info is not making it to the fc. Also I have to initiate the skystars fc on bootloader mode, because it will freeze if I try to upgrade when connected to betafight.

ctzsnooze, 2023-08-14:
We have recently introduced significantly improved Betaflight to GPS hardware module connection and configuration code in Betaflight 4.5 zulu.  We expect that most modern GPS units should now connect and work reliably.  Please give it a try.

These changes will not be back-ported to 4.4.x

Prajwal-Koirala, 2023-12-29:
This is an article about a GPS issue in Betaflight 4.3. It discusses that the GPS module is not recognized in Betaflight 4.3, but it works in 4.2. The issue is with the Nameless Tiny GPS module. The workaround is to disable auto-config. However, this workaround does not work for everyone.

In Betaflight 4.5, the GPS connection and configuration code has been improved, and most modern GPS units should now work reliably. So, I would recommend updating to Betaflight 4.5 for the latest and most reliable GPS experience.

I hope this helps!

closed 2024-05-12 by haslinghuis

Ls4life, 2024-05-13:
Thanks for all the help.