betaflight/betaflight #11455: Lumenier H7 ultimate Receiver Glitching Out

StrawHatSam92, 2022-03-05:
### Describe the bug

Receiver does not work with Lumenier H7 Ultimate Flight controller. The first time when configurating the Rx to the FC, everything seems fine and the receiver tab shows all channels working smoothly and correctly.

But as soon as you unplug and re-plug the USB in again to reconnect the FC to Betaflight and go to the receiver tab, it's all messed up. All channels have a delayed and buggy response that jumps all over the place.

I used both BF 4.3 RC2 and RC3 both have the same issue. I cannot use BF 4.2 of any version because all of them can't get motor 7 to work. This is an octo X8 and I need 8 working motors.

### To Reproduce

To reproduce:

1. Flash Lumenier H7 Ultimate Flight controller with the BF 4.3 RC3 firmware target **betaflight_4.3.0_MATEKH743**
2. Install Betaflight configurator 10.8.0 RC3
3. Wire crossfire diversity nano Rx to any UART and 5v source in the correct way (Rx to Tx & Tx to Rx)
4. Configure Radio with correct crossfire settings
5. In betaflight set ports tab to match the UART the receiver is soldered to.
6. Go to Receiver tab and set to serial protocol and CRSF
7. You will see first that nothing is wrong and the receiver tab is responding correctly and smoothly to radio stick inputs
8. Disconnect the FC and unplug the USB
9. Re-plug the USB and go to the receiver tab again
10. The stick inputs with now be jumping all over the place and not smooth or correct. The effect goes in and out.

### Expected behavior

Receiver tab should how smooth correct response to stick inputs and the drone should arm.

### Flight controller configuration

```
# diff all

# version
# Betaflight / MATEKH743 (M743) 4.3.0 Feb  1 2022 / 20:07:06 (3267f0417) MSP API: 1.44

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKH743
mcu_id 003c001f3130511933323335
signature 

# serial
serial 5 0 115200 57600 0 115200
serial 6 64 115200 57600 0 115200

# master
set gyro_to_use = SECOND
set rssi_channel = 12
set serialrx_provider = CRSF

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```
```
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: FREE
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: BEEPER
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: ADC_CURR
C02: FREE
C03: FREE
C04: ADC_EXT
C05: FREE
C06: FREE
C07: FREE
C08: SDIO_D0
C09: SDIO_D1
C10: SDIO_D2
C11: SDIO_D3
C12: SDIO_CK
C13: FREE
C14: FREE
C15: PREINIT 1
D00: FREE
D01: FREE
D02: SDIO_CMD
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: SPI_MOSI 1
D08: FREE
D09: FREE
D10: PINIO 1
D11: PINIO 2
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: USB_DETECT
E03: LED 1
E04: LED 2
E05: FREE
E06: FREE
E07: SERIAL_RX 7
E08: SERIAL_TX 7
E09: FREE
E10: FREE
E11: GYRO_CS 2
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: GYRO_EXTI
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1 : BEEPER
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MOSI 2
DMA1 Stream 1: SPI_MISO 2
DMA1 Stream 2: SPI_MOSI 4
DMA1 Stream 3: SPI_MISO 4
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

Lumenier H7 Ultimate

### Other components

TBS Crossfire diversity nano receiver

### How are the different components wired up

Wired XF to all UARTS (I tried them all) and every time is the same bug.

klutvott123, 2022-04-11:
Could be related to the CRSF V3 baud negotiation. We have disabled it in RC4 while we're working on fixing it with TBS. Could you try it and report back please?

ctzsnooze, 2022-04-15:
Please test with current master.

haslinghuis, 2022-04-25:
Fixed in [4.3 RC6](https://github.com/betaflight/betaflight/releases/tag/4.3.0-RC6).

closed 2022-04-25 by haslinghuis