betaflight/betaflight #11473: BF4.3 RC3 communication with the equipment is constantly lost Flysky IBUS

tn2009, 2022-03-15:
### Describe the bug

Impossible arming drone. Communication with the receiver is constantly lost. TX16s and mobula6 crazybeef4s. When the connection is lost, you have to turn off and turn on the whoop.

The same is true on the IA6B receiver with matek f405ctr.

### To Reproduce

turn on the equipment and the quadcopter.

### Expected behavior

It works fine on BF4.2.11. Communication with the receiver is not lost.

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```

### Flight controller

Crazybeef4s and Matek F405CTR

haslinghuis, 2022-04-02:
Please test with latest development firmware and report back.

tn2009, 2022-04-03:
Test latest development firmware and same. Communication with the receiver is constantly falling off.

tn2009, 2022-04-04:
This my dump:

# 

# dump

# version
# Betaflight / STM32F405 (S405) 4.3.0 Feb  1 2022 / 20:13:01 (3267f0417) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF405CTR, version: b020e572, date: 2022-01-16T20:04:31Z

# start the command batch
batch start

board_name MATEKF405CTR
manufacturer_id MTKS

# name: -

# resources
resource BEEPER 1 C13
resource MOTOR 1 C06
resource MOTOR 2 C07
resource MOTOR 3 C08
resource MOTOR 4 C09
resource MOTOR 5 NONE
resource MOTOR 6 A08
resource MOTOR 7 B08
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 A02
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 A15
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
[... 1102 lines cut ...]
set quickrates_rc_expo = OFF
set roll_rc_rate = 110
set pitch_rc_rate = 110
set yaw_rc_rate = 170
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 5
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 75
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# end the command batch
batch end

#

haslinghuis, 2022-04-04:
Please replace dump with diff as it's TL:'DR

tn2009, 2022-04-05:
i update comment

tn2009, 2022-04-09:
On BF 4.3 RC4 same issue!!!

haslinghuis, 2022-04-09:
Please test https://github.com/betaflight/betaflight/pull/11509

You can find the firmware here: https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=6235&view=artifacts&pathAsName=false&type=publishedArtifacts

Select or detect your board in configurator in firmware flasher tab first (wait for it to load) and use the Load firmware local to select the downloaded STM32F405.hex

haslinghuis, 2022-04-09:
Please test with: 
https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=6235&view=artifacts&pathAsName=false&type=publishedArtifacts

Select or detect your board in firmware flasher tab and use load firmware local button to select STM32F405 from the download.

tn2009, 2022-04-10:
Installed about a minute later, the quadcopter began to beep. On OSD it is written "Rescue_SW"

Turning off and on the equipment does not help. Only disconnecting from the power supply of the quadcopter.

haslinghuis, 2022-04-10:
`Rescue_SW` indicates the rescue switch is in an unsafe position.
Please see this article for explanation of the beeping pattern: https://oscarliang.com/quad-arming-issue-fix/

tn2009, 2022-04-11:
The switch is as it should be :) I repeat on 4.2.11 there are no problems. The problem is in firmware 4.3. I have a return home configured in case of disconnection with the receiver. I don't start the quadrocopter engines. I turn on the equipment, turn on the quadcopter. The equipment is connected to the quadrocopter receiver. I wait for about a minute and the signal breaks.

ctzsnooze, 2022-04-15:
Please re-test with current master, if you have time.

tn2009, 2022-04-15:
I'll check the new firmware. 
Just two days ago, I checked that GPS was disabled, and the time of returning home was set to attack in case of signal loss. I just turned on the equipment and connected the battery to the quadcopter. I didn't start the engines. And literally after 2 minutes, the quadcopter lost contact and somehow started the engines itself!!!! And if I were testing on the street with propellers, then my quadcopter would just take off at great speed!!!!  Accordingly, the drone would have been lost.

tn2009, 2022-04-15:
I tested latest dev version 15.04.2022. The same thing. The quadcopter is constantly losing communication. Only disconnecting and connecting the battery helps.

tn2009, 2022-04-15:
So I tested the latest firmware again. On mobula 6 crazybeef4fs, the connection doesn't seem to be lost anymore. Nice!

As for the Matek F405CTR, what is the problem I found. I have an AI6B receiver and IBUS telemetry is enabled on the first port. Because of this, the connection is constantly lost. If I turn off IBUS telemetry, then everything seems to be fine with the connection and the problem does not repeat. On BF 4.2.11, the IBUS telemetry worked fine. Why then does IBUS telemetry not work on BF 4.3?

klutvott123, 2022-04-15:
I see. Looks like telemetry for IBUS is handled in the rx task. when telemetry is sent this would probably cause the task to take longer than expected to run

klutvott123, 2022-04-15:
@tn2009 Could you try this, please?
[betaflight_4.3.0_STM32F405_norevision (2).zip](https://github.com/betaflight/betaflight/files/8496947/betaflight_4.3.0_STM32F405_norevision.2.zip)

klutvott123, 2022-04-15:
Looking at your diffs, it looks like you're using 1 wire communication. The port config looks a little strange.
I believe it should be 
```
feature -TELEMETRY
serial 0 4160 115200 57600 115200 115200`
```
The difference when disabling telemetry output in the ports tab as you described is that it will not be using a shared port anymore for rx and telemetry. I bet if you look at the configuration tab you'll find that TELEMERTY is enabled. Try disabling it.
Can you provide the wiring information that was requested when you opened this issue?

tn2009, 2022-04-15:
> @tn2009 Could you try this, please? [betaflight_4.3.0_STM32F405_norevision (2).zip](https://github.com/betaflight/betaflight/files/8496947/betaflight_4.3.0_STM32F405_norevision.2.zip)

Same

tn2009, 2022-04-15:
> 
receiver ia6b one IBUS wire is connected to RX2 the second wire is connected to TX1. 
it only works when telemetry is enabled in the configuration and there is an IBUS on the first port.

klutvott123, 2022-04-15:
Reproduced. sort of. I'm using a flight controller with the rx/telemetry config just as yours with no receiver connected and it freezes on boot. In RC1 it works and RC2 freezes, so something went wrong there. I'll try narrowing it down further

klutvott123, 2022-04-15:
I have done some bisecting between RC1 and RC2 and this is the first bad commit it says https://github.com/betaflight/betaflight/commit/9bdf9c11e9cfc3c14c1efe3409c829f6a538b700

I have further narrowed it down to a call to `rescheduleTask` here https://github.com/betaflight/betaflight/blob/9360ab1276c8b8b4307e4baebfbe8014032819c1/src/main/telemetry/ibus.c#L139

klutvott123, 2022-04-15:
The issue after the commit I referred to is that  `tasks` is now uninitialized until `tasksInit()` runs at the end of `init()`. Everything else is initialized before that, including telemetry. In the check state function for IBUS there's the call to `rescheduleTask` which goes wrong since the tasks array hasn't been initialized yet.

klutvott123, 2022-04-15:
@tn2009 Hopefully this will work. Could you try it, please?
[betaflight_4.3.0_STM32F405_norevision (3).zip](https://github.com/betaflight/betaflight/files/8498382/betaflight_4.3.0_STM32F405_norevision.3.zip)

tn2009, 2022-04-16:
> @tn2009 Hopefully this will work. Could you try it, please? [betaflight_4.3.0_STM32F405_norevision (3).zip](https://github.com/betaflight/betaflight/files/8498382/betaflight_4.3.0_STM32F405_norevision.3.zip)

Finally!!! Fixed!
Within 10 minutes, there was not a single break in communication. Please add this fix to the main firmware of BF 4.3 RC5. Thank you very much!

klutvott123, 2022-04-16:
@tn2009 Great, thanks for testing!
I'll get a PR up to address this soon

klutvott123, 2022-04-16:
PR to fix this is up https://github.com/betaflight/betaflight/pull/11533

closed 2022-04-16 by tn2009