betaflight/betaflight #11479: Assign AnyLeaf manufacturer id

David-OConnor, 2022-03-18:
### Is your feature request related to a problem? Please describe

Add AnyLeaf as a manufacturer; assign ID. Per [instructions here, section 2.3](https://github.com/betaflight/betaflight/blob/master/docs/TargetMaintenance/CreatingAUnifiedTarget.md).

Name of the company selling the board and supporting customers of it: **AnyLeaf**

A website URL for this company: **https://www.anyleaf.org/mercury-g4**

Please let me know if you need any more information. Planned launch is in ~2 months.

### Describe the solution you'd like

Assign ID.

### Describe alternatives you've considered

N/A

David-OConnor, 2022-04-18:
Not stale

blckmn, 2022-04-18:
Merged.

closed 2022-04-18 by blckmn