betaflight/betaflight #11482: RC Smoothing should be separate for each Rateprofile

Combinacijus, 2022-03-21:
### Is your feature request related to a problem? Please describe

Same RC Smoothing is used for every Rateprofile. If we have Rateprofile for racing, freestyle and cinematic flying same RC Smoothing is not optimal and can't be easily switch using transmitter

### Describe the solution you'd like

Move RC Smoothing setting to PID Tuning -> Rateprofile Settings tab

From:
[image]

To:
[image]

### Describe alternatives you've considered

No alternatives

SupaflyFPV, 2022-03-21:
I like the idea if its technically not too complicated.

Quick-Flash, 2022-03-22:
> I like the idea if its technically not too complicated.

Wouldn't be "easy", but it also shouldn't be "hard".

PaulFPV, 2022-03-22:
for me it's a must.. this is why I proposed it here: [](https://github.com/betaflight/betaflight/issues/11152)

;)

ctzsnooze, 2022-04-04:
We need to think about what should be in rateProfiles, what should be in Profiles, and what doesn't need to be in either.

This is a discussion for 4.4, not 4.3.

It's really hard to get any clear consensus, we have tried, so in the end someone will need to make a PR and propose, in it, whatever they want and why.

RateProfiles cost more memory than Profiles which cost more than not in either.  So in practice we need to move some out if we move some in.

PaulFPV, 2022-04-07:
Thanks ctzsnooze for telling us about this problem.

To help me understand better:

By memory cost do you mean the eeprom because Rate profiles use more memory because they are 6, while PID profiles are just 3? .....Or RAM usage because rate profiles are always ready to be selected while another PID profile is just loaded with an fc reboot?

closed 2026-09-26 by blckmn