betaflight/betaflight #11484: ESC passthrough configuration behaviour change (non-BL Heli)

Vanja-APD, 2022-03-23:
### Describe the bug

Connection to passthrough ESCs using BF4.3+ fails with non-BL Heli ESCs due to a change in connection string format. The issue appears within RC1 of 4.3, appears to be a compile-time define (**USE_SERIAL_4WAY_SK_BOOTLOADER**) which affects the connection string format is now guarded by a flash_size check. Imagine below shows the changes. 

[image]

[https://github.com/betaflight/betaflight/commit/82b5e386f39465905c6dbdecbe2e1ee744efe000](url)

Is it possible to reverse this change, or is there a reason for this protocol change? 

Thanks!

### To Reproduce

Install BF4.3 onto FC, attempt ESC passthrough configurator connection to non BLHeli, non-SK ESC. No issues with prior versions of BF (4.2x).

### Expected behavior

USE_SERIAL_4WAY_SK_BOOTLOADER macro remains defined as per previous versions of BF releases. 

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```

### Flight controller

FC Invariant (reproduced on multiple)

DusKing1, 2022-03-23:
Removed due to firmware size

haslinghuis, 2022-04-02:
@daleckystepan @SteveCEvans I think this is important for the use case. How much memory did this save? Do we have room to get it back?

daleckystepan, 2022-04-02:
Enabling `USE_SERIAL_4WAY_SK_BOOTLOADER` adds more than 1.5kB FLASH on F722.

haslinghuis, 2022-04-02:
Yeah this really shows the end of 512K targets. Do we have still have some space left to support this method a little longer otherwise we can close it as `unsupported target` or `too old`? Is this not needed for AM32 etc?

daleckystepan, 2022-04-02:
```
Linking STM32F7X2 
Memory region         Used Size  Region Size  %age Used
        ITCM_RAM:       16256 B        16 KB     99.22%
     AXIM_FLASH1:      445563 B       480 KB     90.65%
```
and
```
Linking STM32F411 
Memory region         Used Size  Region Size  %age Used
          FLASH1:      488183 B       480 KB     99.32%
```

It will probably fit but it is very tough - we will not be able to add something more (and maybe more important). Ont the other hand it's regression. GCC11 will help us slightly.

haslinghuis, 2022-04-02:
🤞

klutvott123, 2022-04-17:
If we were to include this again, we would only have to remove something else. I'm sure manufacturers and firmware developers can include it and compile on their own

closed 2022-05-25 by github-actions[bot]