betaflight/betaflight #11485: Unable to acheive 2k logging on h7 fc without stutters

Quick-Flash, 2022-03-24:
### Describe the bug

While using the latest master, I cannot get 2k logging rate without stutters while using an 8k pidloop rate. I'm using an h7 fc that has plenty of capability to log at 2k.

### To Reproduce

To reproduce this behaviour flash the latest nightly and try logging at a 2k update rate while running 8k. 

### Expected behavior

No stuttering of blackbox logging.

### Flight controller configuration

```
# version
# Betaflight / STM32H743 (SH74) 4.3.0 Mar 23 2022 / 17:58:34 (norevision) MSP API: 1.44
# config: manufacturer_id: DIAT, board_name: MAMBAH743, version: e4b51680, date: 2021-10-04T00:52:01Z

# start the command batch
batch start

board_name MAMBAH743
manufacturer_id DIAT

# resources
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 C08
resource MOTOR 4 C09
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE

# feature
feature -RX_PARALLEL_PWM

# serial
serial 2 8192 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 1 1700 2100 0 0
aux 2 35 3 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set gyro_lpf1_static_hz = 400
set gyro_lpf2_static_hz = 800
set gyro_to_use = BOTH
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 400
set gyro_lpf1_dyn_max_hz = 800
set acc_hardware = NONE
set acc_calibration = -173,-49,67,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1000
set max_check = 2000
set rc_smoothing_auto_factor = 40
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 3
set yaw_deadband = 3
set simplified_gyro_filter_multiplier = 160
set osd_tim1 = 0
set osd_tim2 = 1
set osd_vbat_pos = 2433
set osd_tim_2_pos = 2455
set osd_warnings_pos = 14601
set osd_avg_cell_voltage_pos = 2444
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_menu_background = BLACK
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 3
set vtx_freq = 5917
set vcd_video_system = NTSC
set rpm_filter_harmonics = 2

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 97
set dterm_lpf1_dyn_max_hz = 195
set dterm_lpf1_static_hz = 97
set dterm_lpf2_static_hz = 195
set iterm_rotation = ON
set iterm_relax = RP_INC
set iterm_relax_cutoff = 10
set yaw_lowpass_hz = 0
set p_pitch = 72
set i_pitch = 130
set d_pitch = 63
set f_pitch = 193
set p_roll = 69
set i_roll = 124
set d_roll = 55
set f_roll = 186
set p_yaw = 69
set i_yaw = 124
set f_yaw = 186
set d_min_roll = 55
set d_min_pitch = 63
set simplified_master_multiplier = 155
set simplified_d_gain = 120
set simplified_dmax_gain = 0
set simplified_dterm_filter_multiplier = 130

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 217
set pitch_rc_rate = 217
set yaw_rc_rate = 217
set roll_expo = 71
set pitch_expo = 71
set yaw_expo = 71
set roll_srate = 4
set pitch_srate = 4
set yaw_srate = 4

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: MOTOR 1
B01: MOTOR 2
B02: SPI_MOSI 3
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: PINIO 1
C06: FREE
C07: FREE
C08: MOTOR 3
C09: MOTOR 4
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: SERIAL_RX 4
D01: SERIAL_TX 4
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: SERIAL_TX 3
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: BEEPER
E04: LED 2
E05: LED 1
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: GYRO_CS 2
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: GYRO_EXTI
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
    CH2 : DSHOT_BITBANG 3
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: DSHOT_BITBANG 2
DMA1 Stream 1: DSHOT_BITBANG 3
DMA1 Stream 2: SPI_MOSI 1
DMA1 Stream 3: SPI_MISO 1
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MISO 2
DMA1 Stream 6: SPI_MOSI 3
DMA1 Stream 7: SPI_MISO 3
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: SPI_MOSI 4
DMA2 Stream 3: SPI_MISO 4
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

Mamba h743 dual mpu6000

### Other components

Prydrone f3 45a esc, expresslrs 250hz

### How are the different components wired up

The normal way.

### Add any other context about the problem that you think might be relevant here

[image]

jan4uk, 2022-03-25:
Same problem with my Mamba h743 dual mpu6000, ESC Mamba F60Pro 60A, crossfire
[image]

Something strange here:

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)     10       1       0    0.0%    0.0%         5      0      7       0
01 - (         SYSTEM)    675       1       0    0.0%    0.0%       389      0    478       1
02 - (           GYRO)   8019       2       0    1.6%    0.6%     11706      0   5627       0
03 - (         FILTER)   8019       8       3    6.4%    3.0%     58651      0   5627       0
04 - (            PID)   8019      12      10    9.6%    8.2%    160060      0   5627       0
05 - (            ACC)    565       3       0    0.1%    0.0%       747      0    381       0
06 - (       ATTITUDE)    250       5       3    0.1%    0.0%      1412      0    176       4
07 - (             RX)     15       3       0    0.0%    0.0%    285391      0 180543       1
08 - (         SERIAL)     80    6561      77   52.4%    0.6%      3765      0     56       0
09 - (       DISPATCH)    652       2       0    0.1%    0.0%       284      0    498       1
10 - (BATTERY_VOLTAGE)     47       1       0    0.0%    0.0%        50      0     33       0
11 - (BATTERY_CURRENT)     47       1       0    0.0%    0.0%        22      0     34       0
12 - ( BATTERY_ALERTS)      5       1       0    0.0%    0.0%         4      0      3       0
13 - (         BEEPER)     33       4       1    0.0%    0.0%        93      0     24       3
14 - (            GPS)     50       7       1    0.0%    0.0%       171      0     36      10
18 - (       ALTITUDE)     36       1       0    0.0%    0.0%        24      0     27       0
20 - (      TELEMETRY)    299       1       0    0.0%    0.0%       689      0    210       0
23 - (            OSD)      1       4       7    0.0%    0.0%       118      0      4       5
25 - (            CMS)     18       1       0    0.0%    0.0%        13      0     12       0
26 - (        VTXCTRL)      5       1       0    0.0%    0.0%         2      0      4       0
27 - (        CAMCTRL)      5       0       0    0.0%    0.0%         1      0      3       0
29 - (    ADCINTERNAL)      1       1       0    0.0%    0.0%         1      0      1       1
30 - (       PINIOBOX)     18       1       0    0.0%    0.0%        13      0     13       0
31 - (SPEED_NEGOTIATION)     71       1       0    0.0%    0.0%        19      0     48       0
RX Check Function                   5       0                    119295
Total (excluding SERIAL)                                11.8%

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)     10       1       0    0.0%    0.0%         5      0      7       0
01 - (         SYSTEM)    646       3       0    0.1%    0.0%       389      0    444       0
02 - (           GYRO)    231       2       0    0.0%    0.0%     11711      0   5815       0
03 - (         FILTER)    231       6       4    0.1%    0.0%     58674      0   5815       0
04 - (            PID)    231      12      10    0.2%    0.2%    160119      0   5815       0
05 - (            ACC)    567       3       0    0.1%    0.0%       748      0    394       1
06 - (       ATTITUDE)    249       5       3    0.1%    0.0%      1413      0    182       4
07 - (             RX)     15       5       0    0.0%    0.0%    285509      2 186440       2
08 - (         SERIAL)     81   13558     116  109.8%    0.9%      3778      0     59       1
09 - (       DISPATCH)    743       1       0    0.0%    0.0%       284      0    506       0
10 - (BATTERY_VOLTAGE)     46       1       0    0.0%    0.0%        50      0     33       0
11 - (BATTERY_CURRENT)     47       1       0    0.0%    0.0%        22      0     34       1
12 - ( BATTERY_ALERTS)      5       1       0    0.0%    0.0%         4      0      4       0
13 - (         BEEPER)     33       4       1    0.0%    0.0%        93      0     24       3
14 - (            GPS)     50       9       1    0.0%    0.0%       171      0     36       9
18 - (       ALTITUDE)     36       1       0    0.0%    0.0%        24      0     26       1
20 - (      TELEMETRY)    299       1       0    0.0%    0.0%       689      0    225       1
23 - (            OSD)      1      21       8    0.0%    0.0%       118      0      5      21
25 - (            CMS)     18       1       0    0.0%    0.0%        13      0     14       0
26 - (        VTXCTRL)      5       1       0    0.0%    0.0%         2      0      3       1
27 - (        CAMCTRL)      5       1       0    0.0%    0.0%         1      0      4       0
29 - (    ADCINTERNAL)      1       1       0    0.0%    0.0%         1      0      1       1
30 - (       PINIOBOX)     18       1       0    0.0%    0.0%        13      0     13       0
31 - (SPEED_NEGOTIATION)     69       1       0    0.0%    0.0%        19      0     51       0
RX Check Function                   5       0                    119346
Total (excluding SERIAL)                                 0.2%
#

mituritsyn, 2022-03-30:
is it same as #11458 ?

haslinghuis, 2022-04-02:
Another MCU so like's a more general issue, Please continue here

Duplicate of https://github.com/betaflight/betaflight/issues/11458

closed 2022-04-02 by haslinghuis

hydra, 2022-07-05:
This seems like a BUG to me as a) that CPU is stupid fast, and b) so is the flash chip on it.

Note that the page program time of the W25N01GV is at worst 4.2 times FASTER than the NOR flash chips.

Flash chip on the Mamba:

hydra, 2022-07-05:
@Quick-Flash try a build with https://github.com/betaflight/betaflight/pull/11697