betaflight/betaflight #11486: Osd freezes while saving

Quick-Flash, 2022-03-24:
### Describe the bug

k

### To Reproduce

k

### Expected behavior

k

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `resource show all`
```

### Flight controller

k

haslinghuis, 2022-03-24:
@Quick-Flash Can you please test with https://github.com/betaflight/betaflight-configurator/pull/2846#issuecomment-1071618750. If issue persist we can look at the console.log to have a clue where to start looking.

Quick-Flash, 2022-03-24:
After further testing it seems to be an issue with the save command. Both save, and save and reboot both run into this issue. Funny thing is that the saves always seem to work as expected, I just can't always get back into the main loop after a save and have to power cycle.

haslinghuis, 2022-03-24:
Can you take a look at the console when it happens? F12 on Mac you need another combo to open chrome devtools

Quick-Flash, 2022-03-24:
> Can you take a look at the console when it happens? F12 on Mac you need another combo to open chrome devtools

[image]

Then if I wait long enough this happens
[image]

haslinghuis, 2022-03-24:
Lot of MSP Code 105 is MSP_RC, 
Timeout is dynamic between 250 and 2000 (before it as fixed at 1000ms).
Surprised this even ends up in a DFU connection (this is after save and reboot I assume)

Quick-Flash, 2022-03-24:
> (this is after save and reboot I assume)

Nope, I just went between the OSD tab, press save, go to pid tab, press save, repeat until it stops responding

haslinghuis, 2022-03-24:
There is no spamming protection yet ;-)

Quick-Flash, 2022-03-24:
> There is no spamming protection yet ;-)

happens when saving the osd sometimes, and sometimes after the first save. So unfortunately spamming protection won't help.

haslinghuis, 2022-03-24:
Could not reproduce on Holybro Katute H7 Mini

haslinghuis, 2022-03-24:
@kc10kevin happy to test some samples here together with @SteveCEvans

ctzsnooze, 2022-04-04:
Can't reproduce with Configurator on my Mac, able to save as many times as I like without anything crashing.  F722 latest firmware, nightly Configurator.

closed 2022-04-04 by haslinghuis

mrRobot62, 2022-04-04:
will check this in collaboration with haslinghuis on my mac with some different FCs too (all F4 ;-) )

Quick-Flash, 2022-06-17:
> @Quick-Flash could you solve this issue or find the root cause? 

No, some builds work, others dont. I think it is an uninitialized bug. I think that some part of code isn't initialized so some builds work and others dont. I cant seem to explain the inconsistency otherwise.

daleckystepan, 2022-06-17:
@Quick-Flash I have experienced CLI freezes after one of my PR that haven't touch anything near CLI. It seems to be caused because of some compiler optimization - which is definitely a consequence not the reason. It happens to me only on F405. No idea if it is connected but at least I let you know :)

Quick-Flash, 2022-06-17:
> @Quick-Flash I have experienced CLI freezes after one of my PR that haven't touched anything near CLI. It seems to be caused because of some compiler optimization - which is definitely a consequence not the reason. It happens to me only on F405. No idea if it is connected but at least I let you know :)

Yeah my guess is that there are some unitialised values that are set to different things based on compiler optimisations and such. Which makes it hard to nail down. Unintialised variables arent great practice so it might be worth trying to find a lot of them and get rid of them.

VMaximI, 2022-07-12:
I also have flight controller freezes after switching from CLI to any tab. This is followed by turning off the gyro. With the status command I get this message:

Entering CLI Mode, type 'exit' to return, or 'help'

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.28V, Core temp=42degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4100, max available: 16384
Devices detected: SPI:0, I2C:1
Gyros detected: gyro 1 locked
, BARO=DPS310
OSD: MAX7456
System Uptime: 28 seconds, Current Time: 2022-07-12T15:59:50.921+03:00
CPU:1%, cycle time: 0, GYRO rate: 0, RX rate: 0, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Manufacturer 0x27, 7561216kB, 05/2012, v3.0, 'SD08G'
Filesystem: Ready
Arming disable flags: NOGYRO BOOTGRACE CLI MSP